schreibfaul1/ESP32-audioI2S #306: ESP32S3 compile error

modi12jin, 2022-05-16:
```
#include "Arduino.h"
#include "WiFi.h"
#include "Audio.h"
#include "SD_MMC.h"
#include "FS.h"

#define SDMMC_D2 8
#define SDMMC_D3 14
#define SDMMC_CMD 13
#define SDMMC_CLK 11
#define SDMMC_D0 4
#define SDMMC_D1 3

#define I2S_DOUT      42
#define I2S_BCLK      45
#define I2S_LRC       46

Audio audio;

void setup() {
    Serial.begin(115200);
    SD_MMC.setPins(SDMMC_CLK,SDMMC_CMD,SDMMC_D0,SDMMC_D1,SDMMC_D2,SDMMC_D3);
    SD_MMC.begin("/cdcard",false,false,20000);   
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(21); // 0...21

    audio.connecttoFS(SD, "/320k_test.mp3");
}

void loop()
{
    audio.loop();
}
```

> error message

```
In file included from /home/jin/桌面/audioI2S-SDIO/audioI2S-SDIO.ino:3:
/home/jin/Arduino/libraries/ESP32-audioI2S/src/Audio.h:158:37: error: 'i2s_dac_mode_t' has not been declared
     Audio(bool internalDAC = false, i2s_dac_mode_t channelEnabled = I2S_DAC_CHANNEL_LEFT_EN); // #99
                                     ^~~~~~~~~~~~~~
/home/jin/Arduino/libraries/ESP32-audioI2S/src/Audio.h:196:64: error: 'i2s_dac_mode_t' has not been declared
     [[deprecated]]void setInternalDAC(bool internalDAC = true, i2s_dac_mode_t channelEnabled = I2S_DAC_CHANNEL_LEFT_EN);
                                                                ^~~~~~~~~~~~~~
/home/jin/Arduino/libraries/ESP32-audioI2S/src/Audio.h:475:5: error: 'i2s_dac_mode_t' does not name a type; did you mean 'i2s_mode_t'?
     i2s_dac_mode_t  m_f_channelEnabled = I2S_DAC_CHANNEL_LEFT_EN;  // internal DAC on GPIO26 for M5StickC/Plus
     ^~~~~~~~~~~~~~
     i2s_mode_t
/home/jin/Arduino/libraries/ESP32-audioI2S/src/Audio.h:158:69: error: 'I2S_DAC_CHANNEL_LEFT_EN' was not declared in this scope
     Audio(bool internalDAC = false, i2s_dac_mode_t channelEnabled = I2S_DAC_CHANNEL_LEFT_EN); // #99
                                                                     ^~~~~~~~~~~~~~~~~~~~~~~
/home/jin/Arduino/libraries/ESP32-audioI2S/src/Audio.h:158:69: note: suggested alternative: 'I2S_CHANNEL_STEREO'
     Audio(bool internalDAC = false, i2s_dac_mode_t channelEnabled = I2S_DAC_CHANNEL_LEFT_EN); // #99
                                                                     ^~~~~~~~~~~~~~~~~~~~~~~
                                                                     I2S_CHANNEL_STEREO
/home/jin/Arduino/libraries/ESP32-audioI2S/src/Audio.h:196:96: error: 'I2S_DAC_CHANNEL_LEFT_EN' was not declared in this scope
     [[deprecated]]void setInternalDAC(bool internalDAC = true, i2s_dac_mode_t channelEnabled = I2S_DAC_CHANNEL_LEFT_EN);
                                                                                                ^~~~~~~~~~~~~~~~~~~~~~~
/home/jin/Arduino/libraries/ESP32-audioI2S/src/Audio.h:196:96: note: suggested alternative: 'I2S_CHANNEL_STEREO'
     [[deprecated]]void setInternalDAC(bool internalDAC = true, i2s_dac_mode_t channelEnabled = I2S_DAC_CHANNEL_LEFT_EN);
                                                                                                ^~~~~~~~~~~~~~~~~~~~~~~
                                                                                                I2S_CHANNEL_STEREO
为SD.h找到了多个库
  使用：/home/jin/.arduino15/packages/esp32/hardware/esp32/2.0.3-RC1/libraries/SD
  未使用：/home/jin/Arduino/libraries/SD
Compilation error: exit status 1
```

schreibfaul1, 2022-05-16:
Hello modi12jin,
you are ahead of the curve I ordered two S3 boards in China, the delivery time is about 6 weeks.
According to my information, the S3 does not have an internal DAC. So there is no longer an 'i2s_dac_mode_t'. You could delete everything that refers to it, then the compiler won't complain.

modi12jin, 2022-05-17:
```
info        PSRAM found, inputBufferSize: 283615 bytes
info        buffers freed, free Heap: 319332 bytes
info        Reading file: "/320k_test.mp3"
info        Failed to open file for reading
```

@schreibfaul1  thank you very much! Following your advice I successfully compiled.
But there is no way to play after burning, I don't know if it is because I used SDIO 4bit.

modi12jin, 2022-05-17:
@schreibfaul1 already at work

```
audio.connecttoFS(SD_MMC, "320k_test.mp3");
```

https://user-images.githubusercontent.com/40233017/168774464-cfa1c907-5285-4cca-8bd5-aaba440de27a.mp4

closed 2022-05-17 by modi12jin

schreibfaul1, 2022-05-17:
very nice! It's also good to know that the ESP32S3 works with the library (except the DAC error).

best regards
Wolle