schreibfaul1/ESP32-audioI2S #307: Set pin failure

podaen, 2022-05-19:
Hi,

Thanks for share this library. I m trying to implement you library on my esp32 and have two diffrent programs running on my esp that can be select on boot. On of my other programs uses also I2S and I can't config it, even after deconstructing yours. I am able to install the drivers. Still I have a few error. 

E (12499) I2S: i2s_check_set_mclk(251): ESP32 only support to set GPIO0/GPIO1/GPIO3 as mclk signal, error GPIO number:26
E (12508) I2S: i2s_set_pin(312): mclk config failed

The i2s set pin failed too. Don't know what still to do...

Kind regards
Davy

podaen, 2022-05-20:
Found how I can make it work. If I don't set the pinconfig of the other it works for the other program.

```
i2s_pin_config_t pinconfig = {
     pinconfig .bck_io_num = I2S_BCLK,
     pinconfig .ws_io_num = I2S_LRC,
     pinconfig .data_out_num = I2S_DOUT,
     pinconfig .data_in_num = I2S_PIN_NO_CHANGE
};
```

schreibfaul1, 2022-05-20:
With pinconfig{...} you can assign the I2S lines to the GPIOs permanently. Changes are not possible afterwards. i2s_driver_uninstall() does not remove the assignment either. Then only a reboot helps.

podaen, 2022-05-20:
Idd. Still wondering if the datain of the I2S need to be inused of where it is used for?

podaen, 2022-05-20:
I have tested dozen of url's and dozen sd files and they all work well. There was some tikking background noice when standing close, but none of these fils ever stoped or cause a crash. Well done!

closed 2022-05-21 by podaen