schreibfaul1/ESP32-audioI2S #308: support esp32s3 dlna render

modi12jin, 2022-05-20:
@schreibfaul1 Hi! Are you planning to support the DLNA media renderer?(Since the S3 doesn't have classic bluetooth, I thought of playing the music from the phone via the DLNA media renderer)

schreibfaul1, 2022-05-20:
I can't tell yet. The DLNA media protocol would be possible and I believe that is implemented in the Espressif Audio Development Framework.

modi12jin, 2022-05-21:
@schreibfaul1 I don't know if you've seen this

[image]

https://github.com/yellobyte/SoapESP32

schreibfaul1, 2022-06-09:
DLNA is a good thing. Unfortunately, the ESP32 DLNA renderer cannot find my media server in the home network. I can see the multicast and also the replies to it, but the media server is ignored. I'll figure out why later, thanks for sharing.
Soap32 finds the media server immediately and also displays the directories.

modi12jin, 2022-12-11:
@schreibfaul1  Someone ported the ADF framework to the Arduino platform

【我成功将ESP-ADF移植到了Arduino上，并封装成ArduinoADFSimplify库，ESP32音频开发不再有门槛-哔哩哔哩】 https://b23.tv/h1f7luY

schreibfaul1, 2022-12-16:
Hello [modi12jin](https://github.com/modi12jin),
DLNA is complex and I can't integrate the function into the library. You will find a DLNA complete project in the 'Examples' folder. I created this in a hurry, I'm sure there are still many errors in it. But it works for me.

closed 2023-04-14 by modi12jin