schreibfaul1/ESP32-audioI2S #309: Decoder error

podaen, 2022-05-21:
I am testing my url-streams, and having some trouble with one.

> http://icecast-servers.vrtcdn.be/stubru_tijdloze-high.mp3

and giving me the folowing info...

> info        MP3 decode error -9 : INVALID_HUFFCODES
> info        MP3 decode error -6 : INVALID_FRAMEHEADER
> info        MP3 decode error -8 : INVALID_SCALEFACT
> info        MP3 decode error -2 : MAINDATA_UNDERFLOW

it crashes mostly after 

> info        Metadata block to long! Skipping all Metadata from now on.

> Exception:Error:
> /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:321:::0x40092224:memcpy
> 0x40035476: ?? ??:0
> /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:272:::0x400921cd:memcpy
> /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:290:::0x400921ef:memcpy
> /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:318:::0x40092221:memcpy
> G:\Documents\Visual Studio 2019\Arduino\hardware\espressif\esp32\libraries\WiFi\src/WiFiClient.cpp(121): error 0x400d9dee:WiFiClientRxBuffer
> G:\Documents\Visual Studio 2019\Arduino\hardware\espressif\esp32\libraries\WiFi\src/WiFiClient.cpp:465:::(inlined by) WiFiClient:
> G:\Documents\Visual Studio 2019\Arduino\libraries\ESP32-audioI2S\src/Audio.cpp(3119): error 0x400e5629:Audio
> G:\Documents\Visual Studio 2019\Arduino\libraries\ESP32-audioI2S\src/Audio.cpp:2222 (discriminator 1):::0x400e641c:Audio
> G:\Documents\Visual Studio 2019\Arduino\Player-URL-V2\Player-URL-V2/Player-URL-V2.ino(509): error 0x400d5167:TaskURL(void*)
> 0x40078000: ?? ??:0
> ??:?:::0x40080400:_init
> 0x400805f0: ?? ??:0

schreibfaul1, 2022-05-21:
Sieht nach einem Speicherproblem aus. Am besten funktioniert die Arduino Version 2.0.1-RC1

````
13:21:36: Connect to serial port /dev/ttyUSB0 at 115200
13:21:50: 
13:21:50: >>Send to /dev/ttyUSB0: "http://icecast-servers.vrtcdn.be/stubru_tijdloze-high.mp3"<<
13:21:51: audio_info   : Closing audio file
13:21:51: audio_info   : buffers freed, free Heap: 162928 bytes 
13:21:51: audio_info   : Connect to new host: "http://icecast-servers.vrtcdn.be/stubru_tijdloze-high.mp3"
13:21:51: audio_info   : Connect to "icecast-servers.vrtcdn.be" on port 80, extension "/stubru_tijdloze-high.mp3"
13:21:51: audio_info   : Connection has been established in 288 ms, free Heap: 159948 bytes
13:21:51: audio_info   : HTTP/1.0 200 OK
13:21:51: audio_info   : server: Icecast 2.4.3
13:21:51: audio_info   : date: Sat, 21 May 2022 11:21:51 GMT
13:21:51: audio_info   : content-type: audio/mpeg, format is mp3
13:21:51: audio_info   : MP3Decoder has been initialized, free Heap: 137352 bytes
13:21:51: audio_info   : access-control-allow-origin: *
13:21:51: bitrate      : 128000
13:21:51: audio_info   : ice-audio-info: bitrate=128
13:21:51: bitrate      : 128000
13:21:51: audio_info   : icy-name: StuBru De Tijdloze |Kwaliteit***
13:21:51: audio_info   : Switch to DATA, metaint is 16000
13:21:51: lasthost  : http://icecast-servers.vrtcdn.be/stubru_tijdloze-high.mp3
13:21:52: audio_info   : stream ready
13:21:52: audio_info   : buffer filled in 70 ms
13:21:52: audio_info   : syncword found at pos 84
13:21:52: audio_info   : syncword found at pos 0
13:21:52: audio_info   : MP3 decode error -2 : MAINDATA_UNDERFLOW
13:21:52: audio_info   : syncword found at pos 0
13:21:52: audio_info   : MP3 decode error -2 : MAINDATA_UNDERFLOW
13:21:52: audio_info   : syncword found at pos 0
13:21:52: audio_info   : Channels: 2
13:21:52: audio_info   : SampleRate: 48000
13:21:52: audio_info   : BitsPerSample: 16
13:21:52: audio_info   : BitRate: 128000 
13:21:52: audio_info   : StreamTitle='DE MENS - En in Gent'
13:21:52: [ 32162][I][ESP32_A1S_Radio.ino:608] audio_showstreamtitle(): DE MENS - En in Gent
13:22:08: audio_info   : StreamTitle='StuBru De Tijdloze'
13:22:08: [ 48167][I][ESP32_A1S_Radio.ino:608] audio_showstreamtitle(): StuBru De Tijdloze
13:22:29: audio_info   : StreamTitle='DE MENS - En in Gent'
````

podaen, 2022-05-21:
I have the latest 2.0.3. I have it solved now by decreasing the psram. 

I'm going to testing arround 200 streams, I hope you are willing to help please. I am allready on number 13.

> https://25433.live.streamtheworld.com:443/TOP_RADIO.mp3
> 
> write eeprom UrlSelector: 4
> info        buffers freed, free Heap: 84216 bytes
> info        Connect to new host: "https://25433.live.streamtheworld.com:443/TOP_RADIO.mp3"
> info        Connect to "25433.live.streamtheworld.com" on port 443, extension "/TOP_RADIO.mp3"
> [160293][D][ssl_client.cpp:174] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
> [161106][E][WiFiClient.cpp:336] setOption(): fail on fd -1, errno: 9, "Bad file number"
> info        SSL has been established in 960 ms, free Heap: 36668 bytes
> info        HTTP/1.0 403 Forbidden
> info        access-control-allow-headers: *
> info        access-control-expose-headers: *
> info        access-control-allow-methods: GET, HEAD
> info        access-control-allow-origin: *
> info        content-length: 103
> info        x-stw-error: 430 Invalid Mount
> [161336][E][Audio.cpp:3268] processAudioHeaderData(): can't see content in audioHeaderData

closed 2022-05-21 by podaen

schreibfaul1, 2022-05-21:
I often test with an ESP32 Developer module and 2.0.2 is the last version where PSRAM works properly.