schreibfaul1/ESP32-audioI2S #310: can't see content in audioHeaderData

podaen, 2022-05-21:
I am testing my url-streams, and having some trouble with one.

> https://25433.live.streamtheworld.com:443/TOP_RADIO.mp3
> 
> write eeprom UrlSelector: 4
> info buffers freed, free Heap: 84216 bytes
> info Connect to new host: "https://25433.live.streamtheworld.com:443/TOP_RADIO.mp3"
> info Connect to "25433.live.streamtheworld.com" on port 443, extension "/TOP_RADIO.mp3"
> [160293][D][ssl_client.cpp:174] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
> [161106][E][WiFiClient.cpp:336] setOption(): fail on fd -1, errno: 9, "Bad file number"
> info SSL has been established in 960 ms, free Heap: 36668 bytes
> info HTTP/1.0 403 Forbidden
> info access-control-allow-headers: *
> info access-control-expose-headers: *
> info access-control-allow-methods: GET, HEAD
> info access-control-allow-origin: *
> info content-length: 103
> info x-stw-error: 430 Invalid Mount
> [161336][E][Audio.cpp:3268] processAudioHeaderData(): can't see content in audioHeaderData

> http://21253.live.streamtheworld.com/RADIO1_128.mp3
> 
> write eeprom UrlSelector: 18
> info buffers freed, free Heap: 84252 bytes
> info Connect to new host: "http://21253.live.streamtheworld.com/RADIO1_128.mp3"
> info Connect to "21253.live.streamtheworld.com" on port 80, extension "/RADIO1_128.mp3"
> info Connection has been established in 132 ms, free Heap: 81652 bytes
> info HTTP/1.0 403 Forbidden
> info access-control-allow-headers: *
> info access-control-expose-headers: *
> info access-control-allow-methods: GET, HEAD
> info access-control-allow-origin: *
> info content-length: 103
> info x-stw-error: 430 Invalid Mount
> [720557][E][Audio.cpp:3268] processAudioHeaderData(): can't see content in audioHeaderData

schreibfaul1, 2022-05-21:
https://25433.live.streamtheworld.com:443/TOP_RADIO.mp3 also gives me the HTTP error 403

podaen, 2022-05-21:
I did a look for 403, it means that the request is for something forbidden. Authorization will not help.

podaen, 2022-05-21:
Having a new 403 code but now I got the message Station config not found (Redis). I have reached url stream no 100!

> http://stream.radio.co/s5709ea2c0/listen
> 
> write eeprom UrlSelector: 100
> info        buffers freed, free Heap: 84264 bytes
> info        Connect to new host: "http://stream.radio.co/s5709ea2c0/listen"
> info        Connect to "stream.radio.co" on port 80, extension "/s5709ea2c0/listen"
> info        Connection has been established in 170 ms, free Heap: 81700 bytes
> info        HTTP/1.0 403 Station config not found (Redis)
> [4001524][E][Audio.cpp:3268] processAudioHeaderData(): can't see content in audioHeaderData

podaen, 2022-05-21:
I have found an other link for qmusic with an other codec "http://25703.live.streamtheworld.com:80/QMUSICAAC.aac"
And it work great!

closed 2022-05-21 by podaen