schreibfaul1/ESP32-audioI2S #311: Request failed!

podaen, 2022-05-21:
I am testing my url-streams, and having some trouble with one.

> https://goeiedag.radiostudio.be/oilsjt
> 
> write eeprom UrlSelector: 99
> info        buffers freed, free Heap: 84240 bytes
> info        Connect to new host: "https://goeiedag.radiostudio.be/oilsjt"
> info        Connect to "goeiedag.radiostudio.be" on port 443, extension "/oilsjt"
> [3961249][D][ssl_client.cpp:174] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
> [3961285][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():271]: (-29184) SSL - An invalid SSL record was received
> [3961285][E][WiFiClientSecure.cpp:135] connect(): start_ssl_client: -29184
> info        Request https://goeiedag.radiostudio.be/oilsjt failed!

podaen, 2022-05-21:
After hours of working the all fall

> Auto Next
> http://stream.rlvw.be:8480/stream
> 
> write eeprom UrlSelector: 80
> info        buffers freed, free Heap: 88840 bytes
> info        Connect to new host: "http://stream.rlvw.be:8480/stream"
> info        Connect to "stream.rlvw.be" on port 8480, extension "/stream"
> [293544][E][WiFiGeneric.cpp:1384] hostByName(): DNS Failed for stream.rlvw.be
> info        Request http://stream.rlvw.be:8480/stream failed!
> 
> Auto Next
> http://stream.radiomaria.be:8000/RadioMaria-96
> 
> write eeprom UrlSelector: 81
> info        buffers freed, free Heap: 88836 bytes
> info        Connect to new host: "http://stream.radiomaria.be:8000/RadioMaria-96"
> info        Connect to "stream.radiomaria.be" on port 8000, extension "/RadioMaria-96"
> [333544][E][WiFiGeneric.cpp:1384] hostByName(): DNS Failed for stream.radiomaria.be
> info        Request http://stream.radiomaria.be:8000/RadioMaria-96 failed!
> 
> Auto Next
> http://91.121.219.80/minerva
> 
> write eeprom UrlSelector: 82
> info        buffers freed, free Heap: 88844 bytes
> info        Connect to new host: "http://91.121.219.80/minerva"
> info        Connect to "91.121.219.80" on port 80, extension "/minerva"
> [362189][I][WiFiClient.cpp:254] connect(): select returned due to timeout 2700 ms for fd 48
> info        Request http://91.121.219.80/minerva failed!
> 
> Auto Next
> http://radiomol.ice.infomaniak.ch/radiomol-192.mp3
> 
> write eeprom UrlSelector: 83
> info        buffers freed, free Heap: 88832 bytes
> info        Connect to new host: "http://radiomol.ice.infomaniak.ch/radiomol-192.mp3"
> info        Connect to "radiomol.ice.infomaniak.ch" on port 80, extension "/radiomol-192.mp3"
> [413544][E][WiFiGeneric.cpp:1384] hostByName(): DNS Failed for radiomol.ice.infomaniak.ch
> info        Request http://radiomol.ice.infomaniak.ch/radiomol-192.mp3 failed!
> 
> Auto Next
> http://s39.myradiostream.com:14494/listen.mp3
> 
> write eeprom UrlSelector: 84
> info        buffers freed, free Heap: 88836 bytes
> info        Connect to new host: "http://s39.myradiostream.com:14494/listen.mp3"
> info        Connect to "s39.myradiostream.com" on port 14494, extension "/listen.mp3"
> [453544][E][WiFiGeneric.cpp:1384] hostByName(): DNS Failed for s39.myradiostream.com
> info        Request http://s39.myradiostream.com:14494/listen.mp3 failed!
> 
> Auto Next
> http://live.radiostudio.be/plus
> 
> write eeprom UrlSelector: 85
> info        buffers freed, free Heap: 88848 bytes
> info        Connect to new host: "http://live.radiostudio.be/plus"
> info        Connect to "live.radiostudio.be" on port 80, extension "/plus"
> [493544][E][WiFiGeneric.cpp:1384] hostByName(): DNS Failed for live.radiostudio.be
> info        Request http://live.radiostudio.be/plus failed!
> 
> Auto Next
> http://audiostreamen.nl:8006/
> 
> write eeprom UrlSelector: 86
> info        buffers freed, free Heap: 88852 bytes
> info        Connect to new host: "http://audiostreamen.nl:8006/"
> info        Connect to "audiostreamen.nl" on port 8006, extension "/"
> [533544][E][WiFiGeneric.cpp:1384] hostByName(): DNS Failed for audiostreamen.nl
[... 573 lines cut ...]
> info        Connect to "stream.radioscorpio.be" on port 443, extension "/stream"
> [2813544][E][WiFiGeneric.cpp:1384] hostByName(): DNS Failed for stream.radioscorpio.be
> [2813544][E][WiFiClientSecure.cpp:135] connect(): start_ssl_client: -1
> info        Request https://stream.radioscorpio.be/stream failed!
> 
> Auto Next
> https://eu2-centova.serverse.com/proxy/wiupfvnu?mp=/TradCan
> 
> write eeprom UrlSelector: 35
> info        buffers freed, free Heap: 89068 bytes
> info        Connect to new host: "https://eu2-centova.serverse.com/proxy/wiupfvnu?mp=/TradCan"
> info        Connect to "eu2-centova.serverse.com" on port 443, extension "/proxy/wiupfvnu?mp=/TradCan"
> [2853544][E][WiFiGeneric.cpp:1384] hostByName(): DNS Failed for eu2-centova.serverse.com
> [2853544][E][WiFiClientSecure.cpp:135] connect(): start_ssl_client: -1
> info        Request https://eu2-centova.serverse.com/proxy/wiupfvnu?mp=/TradCan failed!
> 
> Auto Next
> http://stream.trendyfm.be/
> 
> write eeprom UrlSelector: 36
> info        buffers freed, free Heap: 89244 bytes
> info        Connect to new host: "http://stream.trendyfm.be/"
> info        Connect to "stream.trendyfm.be" on port 80, extension "/"
> [2893544][E][WiFiGeneric.cpp:1384] hostByName(): DNS Failed for stream.trendyfm.be
> info        Request http://stream.trendyfm.be/ failed!

podaen, 2022-05-21:
The did work before so I think it is a kind of a bug. They all have in comen that DNS Failed.

podaen, 2022-05-21:
If you like I want to share you my array and my auto next function...

schreibfaul1, 2022-05-21:
If the domain name service is not working, the destination cannot be determined. The DNS determines the valid IP address from the domain name.

If the information info 'slow stream, dropouts are possible' appears, more data is being decoded than is being received. The audio buffer is almost empty

podaen, 2022-05-22:
The DNS servers are all right. Maybe I have to use te WiFi event to check the status of the internet connect. I will see what I can do.

There are a few slow steam in my list, but wasn't a problem before. I could check for an other a URL-stream for that specific station.

podaen, 2022-05-23:
hmmm, its

podaen, 2022-05-23:
Now I received a reason

> https://ruby.torontocast.com:1095/stream failed!
> [480355][D][WiFiGeneric.cpp:876] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
> [480355][W][WiFiGeneric.cpp:897] _eventCallback(): Reason: 8 - ASSOC_LEAVE
> [WiFi-event] event: 5
> Disconnected from WiFi access point
> Disconnected from WiFi access point
> WiFi lost connection. Reason: 8
> Trying to Reconnect
> [480381][D][WiFiGeneric.cpp:876] _eventCallback(): Arduino Event: 3 - STA_STOP
> [WiFi-eventets Jul 29 2019 12:21:46

schreibfaul1, 2022-05-24:
this was maybe a temporary problem 'Disconnected from WiFi access point'

17:29:21: >>Send to /dev/ttyUSB0: "https://ruby.torontocast.com:1095/stream"<<
17:29:22: audio_info   : Closing audio file
17:29:22: audio_info   : buffers freed, free Heap: 162960 bytes
17:29:22: audio_info   : Connect to new host: "https://ruby.torontocast.com:1095/stream"
17:29:22: audio_info   : Connect to "ruby.torontocast.com" on port 1095, extension "/stream"
17:29:23: [ 12710][E][WiFiClient.cpp:318] setOption(): fail on fd -1, errno: 9, "Bad file number"
17:29:23: audio_info   : SSL has been established in 1428 ms, free Heap: 117012 bytes
17:29:24: audio_info   : HTTP/1.0 200 OK 
17:29:24: audio_info   : content-type: audio/mpeg, format is mp3
17:29:24: audio_info   : MP3Decoder has been initialized, free Heap: 89368 bytes
17:29:24: audio_info   : date: Tue, 24 May 2022 15:29:23 GMT
17:29:24: bitrate      : 192000 
17:29:24: audio_info   : ice-audio-info: bitrate=192
17:29:24: audio_info   : icy-name: ALL4MUSIC A4M3
17:29:24: audio_info   : server: Icecast 2.4.0-kh15 
17:29:24: audio_info   : access-control-allow-origin: *
17:29:24: audio_info   : access-control-allow-headers: Origin, Accept, X-Requested-With, Content-Type, Icy-MetaData
17:29:24: audio_info   : access-control-allow-methods: GET, OPTIONS, SOURCE, PUT, HEAD, STATS
17:29:24: audio_info   : Switch to DATA, metaint is 16000
17:29:24: lasthost  : https://ruby.torontocast.com:1095/stream
17:29:24: audio_info   : stream ready
17:29:24: audio_info   : buffer filled in 6 ms
17:29:24: audio_info   : syncword found at pos 0
17:29:24: audio_info   : MP3 decode error -2 : MAINDATA_UNDERFLOW
17:29:24: audio_info   : syncword found at pos 0
17:29:24: audio_info   : Channels: 2
17:29:24: audio_info   : SampleRate: 44100
17:29:24: audio_info   : BitsPerSample: 16
17:29:24: audio_info   : BitRate: 192000
17:29:24: audio_info   : StreamTitle='UB40 - Here I Am, Small Axe (Come And Take Me)'

podaen, 2022-10-07:
It is a temporary problem. The best way to solve it for now is restarting the ESP. It think it is caused because the ESP is occupied with something on a specific moment. It's not only the audio player that is disconnecting... Every client disconnects and then the WiFi disconnects with reason `WIFI_REASON_ASSOC_LEAVE`.

podaen, 2022-10-07:
when I leave audio.loop(); out it doesn't happen anymore. So I think the problem is in there. I can easily reproduce it by pausing it and resume it when an other client is called in a callback.

podaen, 2022-10-07:
maybe solving these will solve the issue...

> Audio.cpp: In member function size_t Audio::process_m3u8_ID3_Header(uint8_t*)
>  
> Audio.cpp: 2199:21: error: variable 'm_f_unsync' set but not used [-Werror=unused-but-set-variable]
>    bool            m_f_unsync = false, m_f_exthdr = false
>    ^~~~~~~~~~
>  
> Audio.cpp: 2200:21: error: unused variable 'last_timestamp' [-Werror=unused-variable]
>    static uint64_t last_timestamp;  \\ remember the last timestamp
>    ^~~~~~~~~~~~~~
>  
> Audio.cpp: 2201:21: error: unused variable 'lastSampleRate' [-Werror=unused-variable]
>    static uint32_t lastSampleRate
>    ^~~~~~~~~~~~~~
>  
> Audio.cpp: 2202:21: error: variable 'current_timestamp' set but not used [-Werror=unused-but-set-variable]
>    uint64_t        current_timestamp = 0
>    ^~~~~~~~~~~~~~~~~
>  
> Audio.cpp: 2203:21: error: unused variable 'newSampleRate' [-Werror=unused-variable]
>    uint32_t        newSampleRate = 0
>    ^~~~~~~~~~~~~
> Audio.cpp: In member function void Audio::processWebStreamTS()
>  
> Audio.cpp: 3148:21: error: variable 'bytesDecoded' set but not used [-Werror=unused-but-set-variable]
>    static int      bytesDecoded
>    ^~~~~~~~~~~~
> Audio.cpp: In member function void Audio::processWebStreamHLS()
>  
> Audio.cpp: 3282:21: error: variable 'bytesDecoded' set but not used [-Werror=unused-but-set-variable]
>    static int      bytesDecoded
>    ^~~~~~~~~~~~
> Audio.cpp: In member function bool Audio::playSample(int16_t*)
>  
> Audio.cpp: 4371:14: error: 'IIR_filterChain1' was not declared in this scope
>    sample = IIR_filterChain1(sample)
>    ^~~~~~~~~~~~~~~~
> Audio.cpp:4371: note  suggested alternative  IIR_filterChain0
>    sample = IIR_filterChain1(sample)
>    ^~~~~~~~~~~~~~~~
>    IIR_filterChain0
>  
> Audio.cpp: 4372:14: error: 'IIR_filterChain2' was not declared in this scope
>    sample = IIR_filterChain2(sample)
>    ^~~~~~~~~~~~~~~~
> Audio.cpp:4372: note  suggested alternative  IIR_filterChain0
>    sample = IIR_filterChain2(sample)
>    ^~~~~~~~~~~~~~~~
>    IIR_filterChain0
> Audio.cpp: At global scope
>  
> Audio.cpp: 4629:10: error: no declaration matches 'int16_t* Audio::IIR_filterChain1(int16_t*, bool)
>    IIR_filterChain1(int16_t iir_in[2], bool clear){  \\ Infinite Impulse Response (IIR) filters
>    ^~~~~
> Audio.cpp:4629: note  no functions named int16_t* Audio  IIR_filterChain1(int16_t*, bool)
>  
> Audio.cpp:11: In file included from
> Audio.h:183: note  class Audio defined here
>  class Audio *: private AudioBuffer{
>    ^~~~~
>  
> Audio.cpp: 4676:10: error: no declaration matches 'int16_t* Audio::IIR_filterChain2(int16_t*, bool)
>    IIR_filterChain2(int16_t iir_in[2], bool clear){  \\ Infinite Impulse Response (IIR) filters
>    ^~~~~
> Audio.cpp:4676: note  no functions named int16_t* Audio  IIR_filterChain2(int16_t*, bool)
>  
> Audio.cpp:11: In file included from
> Audio.h:183: note  class Audio defined here
>  class Audio *: private AudioBuffer{
>    ^~~~~
> Audio.cpp: In member function bool Audio::ts_parsePacket(uint8_t*, uint8_t*, uint8_t*)
>  
> Audio.cpp: 4728:19: error: unused variable 'PAYLOAD_SIZE' [-Werror=unused-variable]
>    const uint8_t PAYLOAD_SIZE = 184
>    ^~~~~~~~~~~~
> Audio.cpp: In member function size_t Audio::chunkedDataTransfer(uint8_t*)
>  
> Audio.cpp: 4957:9: error: value computed is not used [-Werror=unused-value]
>    *bytes++
>    ^~~~~~~~
> Audio.cpp: At global scope
>  
> Audio.cpp: 2201:21: error: 'lastSampleRate' defined but not used [-Werror=unused-variable]
>    static uint32_t lastSampleRate
>    ^~~~~~~~~~~~~~
> Error compiling libraries
>  
> Audio.cpp: 2200:21: error: 'last_timestamp' defined but not used [-Werror=unused-variable]
>    static uint64_t last_timestamp;  \\ remember the last timestamp

podaen, 2022-10-07:
I found it! I set the WiFi power management to maximum power `WiFi.setSleep(false);`. I don't have any failure anymore or a slow connection. It surely improved working behavior and I am glad that I finally found some improvement. I search a long time for this and don't understand this is a default setting in the SDK of ESP Arduino. A battery is optional... and it is not that I am overclocking.

FreeLANMan, 2023-03-20:
Set WiFi.setSleep(false); help me too! Thanks!

closed 2023-06-09 by podaen