schreibfaul1/ESP32-audioI2S #314: Sdfat & audio separately

podaen, 2022-05-24:
What do I need to do if I want to use sdfat in to my shetch. Do I need to close something? I tried songstop, songresumepauze, nothing helpt.

> Exception:Error:
> /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:402:::0x40083e01:panic_abort
> /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:128:::0x40094b89:esp_system_abort
> /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:85:::0x4009a4c1:__assert_func
> /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:821 (discriminator 2):::0x40095626:xQueueGenericSend
> G:\Documents\Visual Studio 2019\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-spi.c(1153): error 0x400f6649:spiEndTransaction
> G:\Documents\Visual Studio 2019\Arduino\hardware\espressif\esp32\libraries\SPI\src/SPI.cpp(193): error 0x400de354:SPIClass
> G:\Documents\Visual Studio 2019\Arduino\libraries\SdFat-master\src\SpiDriver/SdSpiESP.cpp(48): error 0x400de08d:SdSpiArduinoDriver
> G:\Documents\Visual Studio 2019\Arduino\libraries\SdFat-master\src\SdCard/SdSpiCard.h(307): error 0x400dda01:SdSpiCard
> G:\Documents\Visual Studio 2019\Arduino\libraries\SdFat-master\src\SdCard/SdSpiCard.cpp:661:::(inlined by) SdSpiCard:
> G:\Documents\Visual Studio 2019\Arduino\libraries\SdFat-master\src\SdCard/SdSpiCard.cpp:656:::(inlined by) SdSpiCard:
> G:\Documents\Visual Studio 2019\Arduino\libraries\SdFat-master\src\SdCard/SdSpiCard.cpp(633): error 0x400ddedb:SdSpiCard
> G:\Documents\Visual Studio 2019\Arduino\libraries\SdFat-master\src\SdCard/SdSpiCard.cpp(675): error 0x400ddbe3:SdSpiCard
> G:\Documents\Visual Studio 2019\Arduino\libraries\SdFat-master\src\SdCard/SdSpiCard.cpp:334 (discriminator 1):::0x400ddc74:SdSpiCard
> G:\Documents\Visual Studio 2019\Arduino\libraries\SdFat-master\src\SdCard/SdSpiCard.cpp(569): error 0x400dde94:SdSpiCard
> G:\Documents\Visual Studio 2019\Arduino\libraries\SdFat-master\src\SdCard/SdSpiCard.cpp(600): error 0x400ddf7c:SdSpiCard
> G:\Documents\Visual Studio 2019\Arduino\libraries\SdFat-master\src/SdCard/SdSpiCard.h(123): error 0x400dc285:SdSpiCard
> G:\Documents\Visual Studio 2019\Arduino\libraries\SdFat-master\src\common/FsCache.cpp:40:::(inlined by) FsCache:
> G:\Documents\Visual Studio 2019\Arduino\libraries\SdFat-master\src\FatLib/FatPartition.h:266 (discriminator 1):::0x400dc7e8:FatPartition
> G:\Documents\Visual Studio 2019\Arduino\libraries\SdFat-master\src\FatLib/FatFile.cpp:801 (discriminator 1):::(inlined by) FatFile:
> G:\Documents\Visual Studio 2019\Arduino\libraries\SdFat-master\src\FatLib/FatFile.cpp(876): error 0x400dc8a6:FatFile
> G:\Documents\Visual Studio 2019\Arduino\libraries\SdFat-master\src\FatLib/FatFileLFN.cpp(316): error 0x400dd16c:FatFile
> G:\Documents\Visual Studio 2019\Arduino\libraries\SdFat-master\src\FatLib/FatFile.cpp(460): error 0x400dcc58:FatFile
> G:\Documents\Visual Studio 2019\Arduino\libraries\SdFat-master\src\FatLib/FatFile.cpp(427): error 0x400dcc9f:FatFile
> G:\Documents\Visual Studio 2019\Arduino\libraries\SdFat-master\src\FatLib/FatFile.cpp:426:::(inlined by) FatFile:
> G:\Documents\Visual Studio 2019\Arduino\libraries\SdFat-master\src/FatLib/FatVolume.h(87): error 0x400e762d:FatVolume
> G:\Documents\Visual Studio 2019\Arduino\libraries\ESP32-audioI2S\src/Audio.cpp:642:::(inlined by) Audio:
> G:\Documents\Visual Studio 2019\Arduino\libraries\ESP32-audioI2S\src/Audio.cpp(627): error 0x400e7c63:Audio
> E:\AppData\Local\Temp\VMBuilds\Player-URL-V2\espressif_esp32\Debug/fnc_audio.h(431): error 0x400d5ab2:SelectSDpath(char*)
> (inlined by) PlaySong() at E:\AppData\Local\Temp\VMBuilds\Player-URL-V2\espressif_esp32\Debug/fnc_audio.h:241
> (inlined by) PlaySong() at E:\AppData\Local\Temp\VMBuilds\Player-URL-V2\espressif_esp32\Debug/fnc_audio.h:214
> (inlined by) TaskSD(void*) at G:\Documents\Visual Studio 2019\Arduino\Player-URL-V2\Player-URL-V2/Player-URL-V2.ino:232
> 0x40078000: ?? ??:0
> ??:?:::0x40080400:_init
> 0x400805f0: ?? ??:0

podaen, 2022-05-24:
I left out audio.connecttoFS(SD, path); after reading my playlist and now my backtrace is shorter.

I think that this audio.isRunning() returns true when I do some sdfat actions...

podaen, 2022-05-24:
it's not that. Double checked.

podaen, 2022-05-25:
The limitations of an SD card is that there can be only one file open... It seems that the buffer is stored at the same location where the audio is stored to. Is it? Could I use SPIFFS for the buffer and sdfat for the playlist and music, would that help?

schreibfaul1, 2022-05-25:
I believe SD can't open more than one file at a time, SD_MMC doesn't have this limitation

podaen, 2022-05-25:
Do you know if this card reader can read an SD_MMC?

[image]

schreibfaul1, 2022-05-25:
yes,
https://forum.espuino.de/t/welche-hardware-nutzen-develboard-rfid-sd/31

...
GPIOs (15, 14, 2) are fixed and cannot be changed.
One of the GPIOs is 2. If this GPIO is pulled HIGH when the ESP32 is started, the ESP32 cannot be put into flash mode. This means that any pull-up resistor must be removed from the SD board. See e.g. [here 19](https://github.com/biologist79/ESPuino/blob/master/pictures/Pullup-removal.jpg) at [this SD board 16](https://de.aliexpress.com/item/32886674838.html).

podaen, 2022-05-25:
hmmm, rfid. I just removed mine from my project because it was to far away from my mcu and the sd card reader didn't work anymore. But thats not where we are talking about.

It seems that that is pull-up of the miso. Than I have to delete it my code too. Thanks. I will try it.

podaen, 2022-05-25:
How to I connect them?

Pin Name | SPI2 | SPI3
CS0* | 15 | 5
SCLK | 14 | 18
MISO | 12 | 19
MOSI | 13 | 23
QUADWP | 2 | 22
QUADHD | 4 | 21

The available pins on the sd card reader are

3.3v
CS
MOSI
CLK ->14
MISO
GND

schreibfaul1, 2022-05-25:
Exactly, pin2 must not be high when booting, hence the resistor must be removed.

 SD Card | ESP32
     CMD| 15
     CLK | 14
      D0 | 2
````c++
#include "FS.h"
#include "SD_MMC.h"

void setup(){
    pinMode(2, INPUT_PULLUP);
    Serial.begin(115200);
    if(!SD_MMC.begin("/sdcard", true)){
        Serial.println("Card Mount Failed");
        return;
    }
    ...   
}
````

podaen, 2022-05-25:
I forgot I made a shematic for this. Could you check?

[image]

schreibfaul1, 2022-05-25:
The resistor at DAT0 is desoldered, CS remains unconnected, then it should work.

By the way, this is my adapter for SD_MMC without resistors and capacitors
[image]

podaen, 2022-05-25:
Great! I m going to do that too, I don't want to have possible failure issues. I have found some addition information about somebody that shared his expiences too. It seems those pins are also used for WiFi... still there is the possibility to end the mmc and than start wifi.

[a_breakdown_of_my_experience_trying_to_talk_to_an](https://www.reddit.com/r/esp32/comments/d71es9/a_breakdown_of_my_experience_trying_to_talk_to_an/)

I have found out that I was doing two commands on the spi at the same time. And then there was my keyboard that is running in a separate task that gave me also an spi error too, probebly because evrything was terminated, by crashing my task.

> assert failed: xQueueGenericSend queue.c:832 (pxQueue->pcHead != ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == xTaskGetCurrentTaskHandle())

I add a small delay before my vTaskDelayUntil and now evrything works fine. Thanks for helping me out. It was verry learning full and I will do that when I building my own board.

closed 2022-05-28 by podaen

podaen, 2022-10-27:
The mmc isn't easy to find. I both me an sdio, but it isn't fast for an esp32. The only thing I could do is VSPI or HSPI...

podaen, 2022-10-27:
I don't like to mod the boards, so I just orderd for my esp this on. It less sophisticated, I think this is the way to go to understand how it actually works.

[image]