schreibfaul1/ESP32-audioI2S #319: Mute to zero.

podaen, 2022-05-30:
I have tested it this weekend and evryting works great, just a small detail about the volume controle. When I set the volume to 0, there is still knacking sound coming out and sure when there is music playing. It's like mute isn't mute be just set to zero.

schreibfaul1, 2022-05-30:
that looks correct, in playSample() before i2s_write(...)
````c++
    #include assert.h
    if(m_vol == 0){
        assert(s32 == 0);
    }
````
there is only NULL on the I2S
I suspect the noise comes from the operating voltage

podaen, 2022-05-30:
Hmmm, I pulled the decoder out and now I have absolute silence. Sp it is not coming from my amp or the transformer. Maybe I need to use a mute pin on the decoder if there is...

podaen, 2022-05-30:
XSMT seems to be such a pin.

schreibfaul1, 2022-05-30:
I only connect SCK to GND, XSMT is open

podaen, 2022-05-30:
Wait, what! I have a black PCM5102a

I have...

VCC ->3V
3.3V -> 3.3V
GND -> GND
FLT 
DMP
SCL
BLK ->26
DIN -> 27
LCK -> 25
FMT -> GND
XMT -> 3.3V

[image]

According to my datasheet xmt neets to be high for unmute...

schreibfaul1, 2022-05-30:
I have this DAC, SCK could be the same as SCL
[image]

podaen, 2022-05-30:
I will connected to ground. having that board too, stored one of my boxes. For now I will try to do this one work, first try to mute it... if that goos well, than I will do a ESP sleep over night. And solder things up like I sad.

podaen, 2022-05-30:
I have done what you sad and it is much bether now, also the soldering helpt a lot. The mut pin wasn't an improvement like I was especting. I will close this issue now.

closed 2022-05-30 by podaen