schreibfaul1/ESP32-audioI2S #322: ESP32 only support to set GPIO0/GPIO1/GPIO3 as mclk signal, error GPIO number:-373225877

I-Connect, 2022-06-01:
Hi,

I have been running this lib on espressif version 3.1.2 for over a year now, worked perfectly.

I just upped my espressif version to  ^4.0.0 (platform = espressif32@^4.0.0) and pulled the latest version of this lib.

Now I get these errors before initializing:
```
E (363) I2S: i2s_calculate_common_clock(1149): sample rate is too large
E (364) I2S: i2s_calculate_clock(1198): Common clock calculate failed 
```

And after running this during initialization:
`audio.setPinout(GPIO_NUM26, GPIO_NUM25, GPIO_NUM13)`

These errors:
```
E (47891) I2S: i2s_check_set_mclk(251): ESP32 only support to set GPIO0/GPIO1/GPIO3 as mclk signal, error GPIO number:-373225877
E (47892) I2S: i2s_set_pin(312): mclk config failed
```

And later on multiple times:
`[D] [ 56891][E][Audio.cpp:5212] playSample(): Can't stuff any more in I2S...`

I did not pull the ESP32-audioI2S for a while so I am not sure what is now the cause of this.

I hope I do not need to connect anything to GPIO0/GPIO1/GPIO3 as I am already using these pins for other purposes on a custom made PCB...

Any help is appreciated, thx

schreibfaul1, 2022-06-01:
I think there was a version that didn't like assigning -1 to a pin.
https://github.com/espressif/arduino-esp32/issues/6737
Please search in the Audio.cpp for ````m_i2s_config.fixed_mclk = I2S_PIN_NO_CHANGE;````
Maybe if you assign a 0 the error will go away

I-Connect, 2022-06-02:
I changed the define

`#define I2S_PIN_NO_CHANGE (0)`

Errors on creation of the Audio object:

```
E (363) I2S: i2s_calculate_common_clock(1149): sample rate is too large
E (364) I2S: i2s_calculate_clock(1198): Common clock calculate failed
```

errors on playing audio:

```
E (16072) I2S: i2s_set_pin(312): mclk config failed
...
E (18154) I2S: i2s_check_set_mclk(251): ESP32 only support to set GPIO0/GPIO1/GPIO3 as mclk signal, error GPIO number:-459393656
E (18155) I2S: i2s_set_pin(312): mclk config failed
```

I-Connect, 2022-06-02:
Current setup:

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (4.3.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20003.0 (2.0.3)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
```

Changed it using
```
platform = https://github.com/platformio/platform-espressif32.git#v4.4.0
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.3-RC1
```

same errors:
```
E (363) I2S: i2s_calculate_common_clock(1149): sample rate is too large
E (364) I2S: i2s_calculate_clock(1198): Common clock calculate failed
```

```
E (15907) I2S: i2s_check_set_mclk(251): ESP32 only support to set GPIO0/GPIO1/GPIO3 as mclk signal, error GPIO number:626341219
E (15908) I2S: i2s_set_pin(312): mclk config failed
```

I-Connect, 2022-06-02:
I just tried to run I2Saudio.ino and this runs a stream via wifi without issues.

So the issue has to be caused by other code in my project :-(
I will investigate further, if you have any directions I could look let me know

I-Connect, 2022-06-02:
I have it working again

I had the declaration of the Audio object in the .h file within the project class that used it, this gave the error when running the constructor of the Audio object.

I now moved the declaration outside the class at the top of the .cpp file and now it works.

But I have no clue why...

schreibfaul1, 2022-06-02:
very good, creating one object in another should work just fine. I think it's a memory problem, maybe increasing the stack size will help?

I-Connect, 2022-06-02:
I do have a several tasks running (including one that uses the Audio object. 
But these tasks are not running yet during construction of the Audio object and declaration/construction runs in loop() I guess so which stack size or you referring to?

trevorsb, 2022-06-02:
Hey @schreibfaul1 I also have the same issue, where after updating to most recent espresso platform, I get the same errors as @I-Connect on code that worked just a day ago.

 If I run the code with
```
platform = espressif32@3.5.0
```
my audio will play just fine. But using he new update, I get the same exact errors as @I-Connect, namely
```
E (14012) I2S: i2s_calculate_common_clock(1149): sample rate is too large
E (14013) I2S: i2s_calculate_clock(1198): Common clock calculate failed
E (14015) I2S: i2s_check_set_mclk(251): ESP32 only support to set GPIO0/GPIO1/GPIO3 as mclk signal, error GPIO number:-1071136654
E (14026) I2S: i2s_set_pin(312): mclk config failed
[ 15035][E][Audio.cpp:4128] playSample(): Can't stuff any more in I2S...
[ 16036][E][Audio.cpp:4128] playSample(): Can't stuff any more in I2S...
[ 17036][E][Audio.cpp:4128] playSample(): Can't stuff any more in I2S...
[ 18036][E][Audio.cpp:4128] playSample(): Can't stuff any more in I2S...
```

I have had memory issues with the audio in the past and solved them, which makes me think this is something different. Would appreciate any thoughts. Is there an earlier release of your library that I can try rolling back to?

Thank you

I-Connect, 2022-06-02:
could something have changed in the core wrt I2S that is used in the constructor of Audio which causes this?

@trevorsb 
`most recent espresso platform`
Coffee any good with your new machine :-)

trevorsb, 2022-06-02:
@I-Connect You know they are always changing the versions on those coffee machines ;)

trevorsb, 2022-06-15:
On my side, I have narrowed down the error:
```
E (10009) I2S: i2s_calculate_common_clock(1149): sample rate is too large
E (10014) I2S: i2s_calculate_clock(1198): Common clock calculate failed
```
From this line:
```
i2s_driver_install((i2s_port_t)m_i2s_num, &m_i2s_config, 0, NULL);
```
in the Audio constructor `Audio::Audio(bool internalDAC /* = false */, uint8_t channelEnabled /* = I2S_DAC_CHANNEL_BOTH_EN */, uint8_t i2sPort)` at `Audio.cpp:214`.

And then the errors:
```
E (10031) I2S: i2s_check_set_mclk(251): ESP32 only support to set GPIO0/GPIO1/GPIO3 as mclk signal, error GPIO number:806951047
E (10034) I2S: i2s_set_pin(312): mclk config failed
```
From this line:
```
const esp_err_t result = i2s_set_pin((i2s_port_t) m_i2s_num, &m_pin_config);
```
in the `bool Audio::setPinout(uint8_t BCLK, uint8_t LRC, uint8_t DOUT, int8_t DIN)` in `Audio.cpp:4018`.

Those two functions seem to be esp32 native code, which gets pretty dense for me. Anyone have insights into the issue here?

Thank you!

klausahrenberg, 2022-06-19:
Hi, I was able to fix the issue
```
E (14015) I2S: i2s_check_set_mclk(251): ESP32 only support to set GPIO0/GPIO1/GPIO3 as mclk signal, error GPIO number:-1071136654
E (14026) I2S: i2s_set_pin(312): mclk config failed
```
It seems, that setting the i2s pinout and mclk pin setting is changed ad v4.4 of ESP IDF.

Function setPinout in Audio.h/.cpp must be changed. In short, MCK pin assignment must be added there:
```
bool Audio::setPinout(uint8_t BCLK, uint8_t LRC, uint8_t DOUT, int8_t MCK, int8_t DIN) {

    m_pin_config.bck_io_num   = BCLK;
    m_pin_config.ws_io_num    = LRC; //  wclk
    m_pin_config.data_out_num = DOUT;
    m_pin_config.mck_io_num = MCK;
    m_pin_config.data_in_num  = DIN;

    const esp_err_t result = i2s_set_pin((i2s_port_t) m_i2s_num, &m_pin_config);
    return (result == ESP_OK);
}
```
The additional line `m_pin_config.mck_io_num = MCK;` is needed for v4.4, the pin must be defined already there. The additional call of function `Audio::i2s_mclk_pin_select` for setting the mclk pin for your DAC is not longer needed. If you never called this function before, apply `I2S_PIN_NO_CHANGE` for `MCK`.

@schreibfaul1 : Would it be possible to implement this handling for ESP v4.4 with a compiler switch `#ifdef ESP_IDF_4`?

schreibfaul1, 2022-06-20:
Hello everyone,
I took Klaus' suggestion and tested it with different Arduino versions (V1.0.6, V2.0.1-RC1 and V2.0.3), whereby V2.0.3 works with IDF 4.4

klausahrenberg, 2022-06-21:
> Hello everyone, I took Klaus' suggestion and tested it with different Arduino versions (V1.0.6, V2.0.1-RC1 and V2.0.3), whereby V2.0.3 works with IDF 4.4

Thanks for fast implementation, for me the library works now with v4.4 (tested with DAC WM8978).

trevorsb, 2022-06-21:
@schreibfaul1  @klausahrenberg

Thank you for the help and quick implementation. The new fix solved the errors that were thrown in the creation of the object: 
```
E (14015) I2S: i2s_check_set_mclk(251): ESP32 only support to set GPIO0/GPIO1/GPIO3 as mclk signal, error GPIO number:-1071136654
E (14026) I2S: i2s_set_pin(312): mclk config failed
```
But the errors that are thrown when calling `audio->connecttoFS(SPIFFS,file_name.c_str());` are still being produced:
```
E (41406) I2S: i2s_calculate_common_clock(1149): sample rate is too large
E (41407) I2S: i2s_calculate_clock(1198): Common clock calculate failed
on ESP version 4.4
[ 40964][E][Audio.cpp:4259] playSample(): Can't stuff any more in I2S...
[ 41965][E][Audio.cpp:4259] playSample(): Can't stuff any more in I2S...
[ 42965][E][Audio.cpp:4259] playSample(): Can't stuff any more in I2S...
```

These errors also did not exist prior to the Espressif V4.4 . Any thoughts on fixing this?

trevorsb, 2022-06-21:
Here is how I have my code set up:

In the global variables, I declare a pointer for an Audio Object (I am doing this to save memory for other functions in my code):

```
Audio* audio;
```

Then, in my audio function (triggered by button push), I initialize a new Audio object pointed to by the original pointer, set some things, play the audio and delete the object (to free up memory):
```
audio = new Audio;
  
audio->setPinout(27, 26, 25);
audio->setVolume(7); // 0...21 Will need to add a volume setting in the app
audio->connecttoFS(SPIFFS,file_name.c_str());

int w = 0;     
while (w<=4000){
  audio->loop();
  w++;
}
delete audio;
```

I am not calling `audio.i2s_mclk_pin_select();` at any point explicitly, unless it is being called within one of the functions above.

schreibfaul1, 2022-06-22:
Hi trevorsb,
this is a difficult thing, it probably has nothing to do with V4.4. That's just coincidence. The difference is that one time the audio object is created immediately and the other time indirectly via a pointer in setup(). Probably? bits are then set randomly in the m_i2s_config structure, which are still present in the constructor after initialization. I changed the assignment in Audio.h
old: i2s_config_t m_i2s_config;
new: i2s_config_t m_i2s_config = {};
Now it works for me. I would appreciate feedback as to whether the error has actually been eliminated.

closed 2022-06-22 by schreibfaul1