schreibfaul1/ESP32-audioI2S #325: Redirecting to non-accessible URL

makuke1234, 2022-06-05:
When using streams from internet-radio.com the library redirects me from the original radio station URL to some non-accessible URL like http://?location:index.html?sid=1 and then fails to load the stream. Sometimes the redirection occurs to empty address. Also for some reason the whole program hangs sometimes but that may be because my own faulty code (I'm running multithreaded web server on the ESP32). Please note that the my program hangs only with faulty URLs. I tested those URLs with Winamp and they play fine.
Most URLs from [vTuner](https://vtuner.com/) work perfectly, problems occur only with URLs from [Internet Radio](https://www.internet-radio.com/) and almost with all of them.

Here's several logs from serial monitor:
[image]
[image]
[image]

schreibfaul1, 2022-06-05:
After the keyword 'Location:' the lib expects a StreamURL. Shoutcast redirects to a commercial page. Some media players search these for a valid StreamURL. The library cannot do that. As a workaround you could get the playlist, just append listen.pls.
So instead of http://us4.internet-radio.com:8258 rather http://us4.internet-radio.com:8258/listen.pls

makuke1234, 2022-06-05:
Thanks! This fixed my issue. I wouldn't have thought that the issue would be with the URL itself.

closed 2022-06-05 by makuke1234