schreibfaul1/ESP32-audioI2S #327: Mp3 with 48000Hz sampling rate playback failed, but 44100Hz is ok

eggfly, 2022-06-09:
As title mentioned

eggfly, 2022-06-09:
Let me double check it ..

schreibfaul1, 2022-06-09:
16KHz is just a default value, the actual sampling rate is set by the decoder.

xfce, 2022-11-30:
> 16KHz is just a default value, the actual sampling rate is set by the decoder.

@schreibfaul1 where can i find the decoder sample rates, if I want to play some wav files, which sample rates is 96kHz or 192kHz, does the project support? or do I have to change anything, Thanks

schreibfaul1, 2022-11-30:
Common sample rates are 44.1KHz or 48KHz. 192KHz would be four times that. I wonder if the ESP32 can do that.
wav is an uncompressed audio format. The RIFF header is at the beginning of every wav file. The RIFF header looks like this:
[image]
The RIFF header is read in the routine ``read_WAV_Header()`` in audio.cpp

schreibfaul1, 2022-12-13:
The audio files contain information on how to handle them, regardless of whether they are a4a, mp3, wav or flac. The playback of raw data (PAM, PCM) is not intended.
Why should the wav header contain incorrect information that has to be changed afterwards?
What you can do is play an audio file and halve the sample rate. This is like playing a vinyl record at half speed.