schreibfaul1/ESP32-audioI2S #328: RF 433 trouble when running together with Audio.h

jasonrepos, 2022-06-09:
I have been spending hours (days) trying to work this out but I'm honestly stuck.
What I am trying to achieve is create a MP3 player which can be controlled via a RF433Mhz controller (4 buttons, using RC-switch library) for my 2 year old nephew.
All sounds simple and I thought I had achieved this in a couple of hours until I stepped away from my desk and tried to use the remote from a few meters away.  To my suprise the remote would only work if it was held within 30 cm of the receiver!
Much head scratching later, I found that if I disabled the "audio.loop();" then the remote works from the other side of the house (~10 meters) but as soon as I enable the audio then the RF doesn't want to receive unless I hold the remote within 30cm.
I must say that the audio is working flawlessly, however I cannot get my head around why the RF range would be affected.

I have done many things to try and resolve / debug this.

1. After reading other issues in this project, I moved the audio and rf pieces of code to run as RTOS tasks.  Currently I have audio running on CPU 1 and rf on CPU 0.  This does not seem to make a difference to the RF issue.  If I put both tasks to run on CPU 1 then the audio starts to stutter and it too does not help with the RF.

2. I then thought maybe the problem is to do with power, ie: when the audio is playing (driving the MAX98357A) is it causing the voltage to drop on the RF receiver as I am powering via USB for my testing.  Measuring with a cheap scope showed roughly 4.7v.  I then got a bench power supply and hooked that up so a constant 5v were being provided to the RF receiver, but this too didn't change anything.

3. I then added decoupling caps (10uF and 0.1uF across the RF receiver) without any luck.

4. I have created many different antennas but these also do not make any difference.  - Remember without audio.loop the RF does work from a distance. 

I notice @schreibfaul1 mentions that the audio libs needs to be constantly "fed", however if I have these tasks running on different CPU's, shouldn't this constraint be eliminated, or is there another shared resource which both I2S and the RF would be contending over?

Here is a link to my "cobbled-together-code".  https://github.com/jasonrepos/rf-test
I would really appreciate if anyone could point me in the right direction on how to get these two modules to play nicely.

biologist79, 2022-06-09:
@jasonrepos What about using RFID-cards to direct the music-player? https://github.com/biologist79/ESPuino.
In case you don't want to: maybe there's stuff that's useful for your implementation.

schreibfaul1, 2022-06-09:
Hello jasonrepos,
I think the antennas influence each other. A shield might help.
A very good idea is the use of RFID-cards, as used by biologist79 in his ESPuino project.
I like to use IR remote controls, you don't need any further tasks, the normal loop() is enough, see https://github.com/schreibfaul1/ESP32-IR-Remote-Control

jasonrepos, 2022-06-09:
@biologist79 - that is a very cool project.  I would use it however my nephew is always stealing our keys to turn the car immobiliser on and off (bleep, bleep-bleep), so I am looking to replicate this for him with a remote control car which will play the same sounds as the real car along with some nursery rhymes when he presses different buttons.  So as you can see in this case the "user" is defining the requirements :)

@schreibfaul1 - you are spot on.  I have just put a buck converter and longer wires between my 5v power rail and the RF receiver and I can already take a few steps away and it still works.  It's not perfect or the final solution but I believe it proves that there is interference which is to blame, so I probably need to look more into how I should filter / shield the MAX98357A from the RF.

Thank you both for your quick replies.

SerjPr, 2022-12-12:
You are using a class D amplifier. It is quite possible that it is causing radio interference. It uses PWM. You need to use a class AB amplifier.

closed 2023-04-14 by stale[bot]