schreibfaul1/ESP32-audioI2S #329: Unreasonably small maximum buffer size, ~32kB without PSRAM

makuke1234, 2022-06-12:
When using the library without PSRAM, the maximum buffer size I am able to achieve bug-free is 32768 bytes. When using buffer sizes beyond 32768 and up to 48000 (I have tested only 48000) the ESP32 acts buggy, certain other functions start to fail like switching through radio stations. When using anything higher than 65535 (I have not tested anything between 48000 & 65535) the audio will simply not play. That's all strange because in the log the library reports that almost 80kB of heap RAM is free (even when using 128kB buffer).
The only reason I have bothered with higher bitrates is because some radios stutter - playing for half a second, silence for half a second. This doesn't depend so much on bitrate as some 320 kbps radios play fine, but even some 192 kbps radios stutter badly. My internet connection is out of question as my download speed exceeds 100 Mbps. I suspect that the radios are stuttering are because of long delays between packets sent from radio station servers. I think they are sending out large packets every second or so which unfortunately don't fit into my ESPs buffer.

So the only way out for me to ensure smooth playback is to increase the buffer size which turned out to be almost impossible without PSRAM. Any info regarding possible bugs/shortcomings in the library would be helpful.

makuke1234, 2022-06-18:
I figured out the problem. The heap wasn't really free, I was creating another thread that filled the heap. Now I am able to increase the buffer size to 64kB, although this didn't fix my stuttering problem with a few radios. :/

closed 2022-06-18 by makuke1234

schreibfaul1, 2022-06-18:
Very well solved! In fact, getFreeHeap() displays the entire range. The tasks have only a part of it assigned (with xTaskCreate()). And then, e.g. with malloc(), the free area must also be unfragmented.