schreibfaul1/ESP32-audioI2S #331: how to connect SPIFFS?

murarduino, 2022-06-13:
I tried to modify the I2Saduio.ino file for playing mp3 files in SPIFFS. And upload the pno-cs.mp3 file to the SPIFFS partition. But it seems that the program cannot connect to SPIFFS. How to solve?

terminal output
`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1284
load:0x40078000,len:12808
load:0x40080400,len:3032
entry 0x400805e4
info        PSRAM not found, inputBufferSize: 6399 bytes
info        buffers freed, free Heap: 252148 bytes
info        Reading file: "/pno-cs.mp3"
[   128][E][vfs_api.cpp:22] open(): File system is not mounted
[   128][E][vfs_api.cpp:22] open(): File system is not mounted
info        Failed to open file for reading`

sketch：

`#include "Arduino.h"
#include "Audio.h"

// Digital I/O used
#define I2S_DOUT      2
#define I2S_BCLK      15
#define I2S_LRC       13

Audio audio;

void setup() {
    Serial.begin(115200);
    WiFi.mode(WIFI_OFF);
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(12); // 0...21

    audio.connecttoFS(SPIFFS, "/pno-cs.mp3");
}

void loop()
{
    audio.loop();
    if(Serial.available()){ // put streamURL in serial monitor
        audio.stopSong();
        String r=Serial.readString(); r.trim();
        if(r.length()>5) audio.connecttohost(r.c_str());
        log_i("free heap=%i", ESP.getFreeHeap());
    }
}

// optional
void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
void audio_id3data(const char *info){  //id3 metadata
    Serial.print("id3data     ");Serial.println(info);
}
void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);
}
void audio_showstation(const char *info){
    Serial.print("station     ");Serial.println(info);
}
void audio_showstreamtitle(const char *info){
    Serial.print("streamtitle ");Serial.println(info);
}
void audio_bitrate(const char *info){
    Serial.print("bitrate     ");Serial.println(info);
}
void audio_commercial(const char *info){  //duration in sec
    Serial.print("commercial  ");Serial.println(info);
}
void audio_icyurl(const char *info){  //homepage
    Serial.print("icyurl      ");Serial.println(info);
}
void audio_lasthost(const char *info){  //stream URL played
    Serial.print("lasthost    ");Serial.println(info);
}
`

nodoubtman, 2022-06-14:
In your setup, you have to init SPIFFS filesystem.

bool spiffsOK = SPIFFS.begin()

Give me some news.
Marc.

murarduino, 2022-06-14:
> In your setup, you have to init SPIFFS filesystem.
> 
> bool spiffsOK = SPIFFS.begin()
> 
> Give me some news. Marc.

Thanks for the reply, the problem has been solved. The header files for the SPIFFS library need to be added to the sketch.

murarduino, 2022-06-14:
> In your setup, you have to init SPIFFS filesystem.
> 
> bool spiffsOK = SPIFFS.begin()
> 
> Give me some news. Marc.

Thanks for the reply, the problem has been solved. As you pointed out, add SPIFFS.begin(); in the setup().

code:
`void setup() {
    Serial.begin(115200);
    bool spiffsOK = SPIFFS.begin();
    Serial.printf("SPIFFS Status: %d\n",spiffsOK);
    WiFi.mode(WIFI_OFF);
.........`

monitor:
`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1284
load:0x40078000,len:12808
load:0x40080400,len:3032
entry 0x400805e4
SPIFFS Status: 1
info        PSRAM not found, inputBufferSize: 6399 bytes
info        buffers freed, free Heap: 247580 bytes
info        Reading file: "/pno-cs.mp3"
........`

closed 2022-06-14 by murarduino