schreibfaul1/ESP32-audioI2S #334: HLS streams

renatoianhez, 2022-06-18:
Hi! I would first like to congratulate you for the great library!
I'm setting up a Web radio and I noticed that several URL's don't work. Many of them stream over HLS (eg: "https://eldoradolive02.akamaized.net/hls/live/2043453/eldorado/master.m3u8"). Is this format not supported?
Thanks!

schreibfaul1, 2022-06-18:
m3u8 playlists are a special challenge. Some simple ones can be played like http://stream.revma.ihrhls.com/zc4435/hls.m3u8
In principle, individual streams (playing only a few seconds) are summarized in the playlist. Once the individual streams have been played, the playlist is often outdated and a new one has to be fetched. The only solution I have is to cut the existing connection  and establish a new one. Especially with encrypted connections, if it is too long (approx. 2 seconds) in the current stream, this leads to dropouts. If anyone is reading this and knows a solution, please post it here.

pavankarney, 2022-06-28:
https://github.com/Yokohama-Miyazawa/M5HLSPlayer

renatoianhez, 2022-07-02:
My project is here: https://github.com/renatoianhez/Lolin32-radio
In the code, I inform which radio stations have glitches with a "GL+"

renatoianhez, 2022-07-02:
> https://github.com/Yokohama-Miyazawa/M5HLSPlayer

It would be nice if this library here had this addendum!

schreibfaul1, 2022-07-02:
Yes, HLS integration would be a good thing. I had already started with it, very simple m3u8 streams are already running. This is only a small part of all possible streams. The linked project is extensive and it takes some time to understand exactly how it works. The TS part with the PAT/PMT tables is interesting.

pavankarney, 2022-07-04:
Thank you so much, 
In India, ALL INDIA RADIO (AIR) nowadays streams its channels with m3U8.

pavankarney, 2022-07-11:
.ts format links not working
https://air.pc.cdn.bitgravity.com/air/live/pbaudio001/playlist.m3u8
thankyou for your wonderful work..

https://mirchiplaylive.akamaized.net/hls/live/2036929/MUM/MEETHI_Auto.m3u8

info        Connect to new host: "https://mirchiplaylive.akamaized.net/hls/live/2036929/MUM/MEETHI_Auto.m3u8"
info        buffers freed, free Heap: 182260 bytes
info        connect to mirchiplaylive.akamaized.net on port 443 path /hls/live/2036929/MUM/MEETHI_Auto.m3u8
info        SSL has been established in 2263 ms, free Heap: 141672 bytes
info        httpResponseHeader: HTTP/1.0 200 OK
info        httpResponseHeader: Akamai-Path-Timestamp: i=1657505881.714;xi=1657505881.713;xo=1657505890.610;s=1657505890.624;
info        httpResponseHeader: Last-Modified: Mon, 11 Jul 2022 02:18:01 GMT
info        httpResponseHeader: X-Akamai-Live-Origin-QoS: d=10000;t=1657505881.715
info        httpResponseHeader: X-Akamai-Server: Akamai-SMT
info        httpResponseHeader: Akamai-Mon-Iucid-Ing: 2036929
info        httpResponseHeader: Vary: Accept-Encoding
info        httpResponseHeader: Expires: Mon, 11 Jul 2022 02:18:11 GMT
info        httpResponseHeader: Cache-Control: max-age=0, no-cache, no-store
info        httpResponseHeader: Pragma: no-cache
info        httpResponseHeader: Date: Mon, 11 Jul 2022 02:18:11 GMT
info        httpResponseHeader: Content-Length: 245
info        httpResponseHeader: Connection: keep-alive
info        httpResponseHeader: Akamai-Mon-Iucid-Del: 1277986
info        httpResponseHeader: Alt-Svc: h3-Q050=":443"; ma=93600,quic=":443"; ma=93600; v="46,43"
info        httpResponseHeader: Content-Type: application/x-mpegURL
info        content-type: application/x-mpegURL
info        httpResponseHeader: Access-Control-Max-Age: 86400
info        httpResponseHeader: Access-Control-Allow-Credentials: true
info        httpResponseHeader: Access-Control-Expose-Headers: Server,range,hdntl,hdnts,Akamai-Mon-Iucid-Ing,Akamai-Mon-Iucid-Del,Akamai-Request-BC
info        httpResponseHeader: Access-Control-Allow-Headers: origin,range,hdntl,hdnts
info        httpResponseHeader: Access-Control-Allow-Methods: GET,POST,OPTIONS
info        httpResponseHeader: Access-Control-Allow-Origin: *
info        now parse playlist
info        cth https://mirchiplaylive.akamaized.net/hls/live/2036929/MUM/MEETHI64kbps.m3u8
info        httpResponseHeader: HTTP/1.0 200 OK
info        httpResponseHeader: Akamai-Path-Timestamp: i=1657505881.704;xi=1657505881.707;xo=1657505885.723;s=1657505885.721;
info        httpResponseHeader: Last-Modified: Mon, 11 Jul 2022 02:18:01 GMT
info        httpResponseHeader: X-Akamai-Live-Origin-QoS: d=10000;t=1657505881.705
info        httpResponseHeader: X-Akamai-Server: Akamai-SMT
info        httpResponseHeader: Akamai-Mon-Iucid-Ing: 2036929
info        httpResponseHeader: Vary: Accept-Encoding
info        httpResponseHeader: Expires: Mon, 11 Jul 2022 02:18:12 GMT
info        httpResponseHeader: Cache-Control: max-age=0, no-cache, no-store
info        httpResponseHeader: Pragma: no-cache
info        httpResponseHeader: Date: Mon, 11 Jul 2022 02:18:12 GMT
info        httpResponseHeader: Content-Length: 325
info        httpResponseHeader: Connection: keep-alive
info        httpResponseHeader: Akamai-Mon-Iucid-Del: 1277986
info        httpResponseHeader: Alt-Svc: h3-Q050=":443"; ma=93600,quic=":443"; ma=93600; v="46,43"
info        httpResponseHeader: Content-Type: application/x-mpegURL
info        content-type: application/x-mpegURL
info        httpResponseHeader: Access-Control-Max-Age: 86400
info        httpResponseHeader: Access-Control-Allow-Credentials: true
info        httpResponseHeader: Access-Control-Expose-Headers: Server,range,hdntl,hdnts,Akamai-Mon-Iucid-Ing,Akamai-Mon-Iucid-Del,Akamai-Request-BC
info        httpResponseHeader: Access-Control-Allow-Headers: origin,range,hdntl,hdnts
info        httpResponseHeader: Access-Control-Allow-Methods: GET,POST,OPTIONS
info        httpResponseHeader: Access-Control-Allow-Origin: *
info        now parse playlist
info        StreamTitle=iHeartRadio.com - iHeart Country Classics
streamtitle iHeartRadio.com - iHeart Country Classics
info        cth https://mirchiplaylive.akamaized.net/hls/live/2036929/MUM/MEETHI64kbps5328000.ts
info        httpResponseHeader: HTTP/1.0 200 OK
info        httpResponseHeader: Akamai-Path-Timestamp: i=1657505832.948;xi=1657505832.948;xo=1657505838.443;s=1657505838.500;
info        httpResponseHeader: Content-Length: 120393
info        httpResponseHeader: Last-Modified: Mon, 11 Jul 2022 02:17:12 GMT
info        httpResponseHeader: X-Akamai-Live-Origin-QoS: d=10000;t=1657505832.949
info        httpResponseHeader: X-Akamai-Server: Akamai-SMT
info        httpResponseHeader: Pragma: no-cache
info        httpResponseHeader: Akamai-Mon-Iucid-Ing: 2036929
info        httpResponseHeader: Cache-Control: max-age=31535948
info        httpResponseHeader: Expires: Tue, 11 Jul 2023 02:17:20 GMT
info        httpResponseHeader: Date: Mon, 11 Jul 2022 02:18:12 GMT
info        httpResponseHeader: Connection: keep-alive
info        httpResponseHeader: Akamai-Mon-Iucid-Del: 1277986
info        httpResponseHeader: Alt-Svc: h3-Q050=":443"; ma=93600,quic=":443"; ma=93600; v="46,43"
info        httpResponseHeader: Content-Type: video/MP2T
info        content-type: video/MP2T
info        ContentType video/mp2t not supported
station     
icyurl

pavankarney, 2022-07-11:
info        Connect to new host: "https://air.pc.cdn.bitgravity.com/air/live/pbaudio001/playlist.m3u8"
info        buffers freed, free Heap: 182152 bytes
info        connect to air.pc.cdn.bitgravity.com on port 443 path /air/live/pbaudio001/playlist.m3u8
info        SSL has been established in 1239 ms, free Heap: 139308 bytes
info        httpResponseHeader: HTTP/1.1 200 OK
info        httpResponseHeader: Content-Type: application/vnd.apple.mpegurl
info        content-type: application/vnd.apple.mpegurl
info        httpResponseHeader: Content-Length: 105
info        httpResponseHeader: Connection: keep-alive
info        httpResponseHeader: Date: Mon, 11 Jul 2022 02:20:54 GMT
info        httpResponseHeader: Access-Control-Expose-Headers: Date, Server, Content-Type, Content-Length
info        httpResponseHeader: Cache-Control: max-age=1
info        httpResponseHeader: Access-Control-Allow-Credentials: true
info        httpResponseHeader: Access-Control-Allow-Methods: OPTIONS, GET, POST, HEAD
info        httpResponseHeader: Content-Encoding: gzip
info        can't extract gzip
station     
icyurl

schreibfaul1, 2022-07-11:
I still have some 'normal' m3u8 streams that aren't working yet, I'll do that first. TS is coming soon, I still have some testing to do.
I can't unpack gzip compressed responses from server.

pavankarney, 2022-07-12:
Thank you in advance

pavankarney, 2022-07-26:
thank you so much for bringing .ts format live.. its playing now, but, while changing the stream files there is small glitch of noise. and after streaming one or two files we can see "slow stream, dropouts are possible".

Some links are unable to play as below..
  "https://mirchiplaylive.akamaized.net/hls/live/2036929/MUM/MRETRO_Auto.m3u8"
  "https://mirchiplaylive.akamaized.net/hls/live/2036929/MUM/MNINTS_Auto.m3u8"
  "https://mirchiplaylive.akamaized.net/hls/live/2036929/MUM/MEETHI_Auto.m3u8"

this will be your roaring update.. thank you so much. I am happy to use your work.

renatoianhez, 2022-07-26:
In my case it's resulting in multiple compilation errors like:
```
Audio.cpp:4541:23: error: 'struct Audio::ts_parsePacket(uint8_t*, uint8_t*, uint8_t*)::pid_array' has no member named 'pids'
             pidsOfPMT.pids[indexOfPids++] = program_map_PID;
```

renatoianhez, 2022-07-26:
I'm facing this error:` text section exceeds available space in board. Sketch too big;`
I'll delete the comments and see if it works...

schreibfaul1, 2022-07-26:
There is still a lot to do, this is only a partial success, but now I know how to correctly unpack TS packets without the AAC decoder complaining. I have a list of URLs that don't work at all or get into an infinite loop and then quit.
To make better use of the flash memory I recommend the huge_app.cvs partition table, and to reduce dropouts the DMA can be set as high as possible, in the constructor `m_i2s_config.dma_buf_count` set  from 14 to 20 .

pavankarney, 2022-07-28:
line no 2568    return NULL;
when replaced by : return m_lastHost;
Solved my problem of drop outs please check .. now its running fine with 
"http://air.pc.cdn.bitgravity.com/air/live/pbaudio001/playlist.m3u8"

pavankarney, 2022-07-28:
thank you for updates now ,
"https://mirchiplaylive.akamaized.net/hls/live/2036929/MUM/MRETRO_Auto.m3u8"
"https://mirchiplaylive.akamaized.net/hls/live/2036929/MUM/MNINTS_Auto.m3u8"
"https://mirchiplaylive.akamaized.net/hls/live/2036929/MUM/MEETHI_Auto.m3u8"
all playing

renatoianhez, 2022-07-28:
I receive this errors wen compiling for WROVER:
```
C:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp: In member function 'bool Audio::ts_parsePacket(uint8_t*, uint8_t*, uint8_t*)':
C:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:4517:11: error: 'u_int8_t' does not name a type
     const u_int8_t TS_PACKET_SIZE = 188;
           ^
C:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:4518:11: error: 'u_int8_t' does not name a type
     const u_int8_t PAYLOAD_SIZE = 184;
           ^
C:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:4519:11: error: 'u_int8_t' does not name a type
     const u_int8_t PID_ARRAY_LEN = 4;
           ^
C:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:4523:18: error: 'PID_ARRAY_LEN' was not declared in this scope
         int pids[PID_ARRAY_LEN];
                  ^
C:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:4532:58: error: 'struct Audio::ts_parsePacket(uint8_t*, uint8_t*, uint8_t*)::pid_array' has no member named 'pids'
         for(int i = 0; i < PID_ARRAY_LEN; i++) pidsOfPMT.pids[i] = 0;
                                                          ^
C:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:4595:23: error: 'struct Audio::ts_parsePacket(uint8_t*, uint8_t*, uint8_t*)::pid_array' has no member named 'pids'
             pidsOfPMT.pids[indexOfPids++] = program_map_PID;
                       ^
C:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:4611:31: error: 'TS_PACKET_SIZE' was not declared in this scope
             *packetLength   = TS_PACKET_SIZE - posOfPacketStart;
                               ^
C:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:4636:33: error: 'TS_PACKET_SIZE' was not declared in this scope
                 *packetLength = TS_PACKET_SIZE - posOfPacketStart - startOfData;
                                 ^
C:\Users\reinh\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:4650:33: error: 'struct Audio::ts_parsePacket(uint8_t*, uint8_t*, uint8_t*)::pid_array' has no member named 'pids'
             if(PID == pidsOfPMT.pids[i]) {
                                 ^
```

schreibfaul1, 2022-07-28:
I renamed the old type u_int8_t to the more common uint8_t.

renatoianhez, 2022-07-28:
> Not work with [Eldorado FM: https://eldoradolive02.akamaized.net/hls/live/2043453/eldorado/master.m3u8](https://eldoradolive02.akamaized.net/hls/live/2043453/eldorado/master.m3u8)

Congratulations!!! You got it! This radio station now works perfectly!!!
Thank you very much!!

[This one](https://cbn.globoradio.globo.com/Player/playerAoVivoBH.htm) I don't understand how it works... I couldn't get the link to her stream.

schreibfaul1, 2022-07-31:
I've changed a few things, ELDORADO FM works for me now, can you please test it.

renatoianhez, 2022-08-01:
Perfect! Thank you very much!!
I could also notice that the connection with the radios was a little faster, or is it just me?

pavankarney, 2022-08-23:
I found some tags as below inbetween .ts format audio
"https://mirchiplaylive.akamaized.net/hls/live/2036929/MUM/MRETRO_Auto.m3u8",
"https://mirchiplaylive.akamaized.net/hls/live/2036929/MUM/MNINTS_Auto.m3u8",
"https://mirchiplaylive.akamaized.net/hls/live/2036929/MUM/MEETHI_Auto.m3u8",
"Service 01ATCLabs"
at this positions music is jumping, plese have alook.

schreibfaul1, 2022-08-23:
There are problems with the three URLs mentioned. The running time of the files is a little shorter than specified in the m3u8 as '#EXTINF:9.752381' with 9.75s. Have no idea why.
What do you mean by 'Service 01�ATCLabs', that might help me with troubleshooting.

pavankarney, 2022-08-25:
Dear Brother,
Please find attached .ts file
you can find in this file a tag "Service 01 ATCLabs€ü+
ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ"
which you can see more than one times in a file. at this point file is
skipping some music.
If we can find some correction for this, I think the audio jumping will be
eliminated.

schreibfaul1, 2022-08-25:
Yes, I found that in the ts file.
[image]
Every 211th data packet contains this strange one "Service 01".
I have no idea what that is, and I couldn't find anything about it on the internet. The data packet with the PID 0x011
HEX   47 4[0 11] 12 00 42 ... is discarded, so it does not reach the AAC decoder. With the fix from August 21st, the stream is running quite well for me.

pavankarney, 2022-08-25:
> Yes, I found that in the ts file.
> [image]
> Every 211th data packet contains this strange one "Service 01".
> I have no idea what that is, and I couldn't find anything about it on the internet. The data [...]

I think it’s company advertisement actlabs is the IT company which is developing streaming services. 
Problem still sustaining maybe you don’t know Hindi you can’t make out

closed 2023-04-14 by stale[bot]