schreibfaul1/ESP32-audioI2S #338: audio.loop() doesn't loop

witivx, 2022-07-11:
Hello,
I'm trying to run the follwing code but it doesn't run properly. Nothing is happening when I'am trying to read .mp3 file, and by passing .wav it is working but only once. I have to unpower and power the board to make it work again.
I'm running the code with an ESP32-Wroover-dev from FREENOVE, the [MAX98357A](https://www.adafruit.com/product/3006) amplifier and [Adafruit 254](https://www.adafruit.com/product/254) micro SD reader.
The code : 
`#include "Arduino.h"
#include "Audio.h"
#include "SD.h"
#include "FS.h"
 
// Digital I/O used
#define SD_CS          5
#define SPI_MOSI      23    // SD Card
#define SPI_MISO      19
#define SPI_SCK       18
 
#define I2S_DOUT      25
#define I2S_BCLK      27    // I2S
#define I2S_LRC       26
 
Audio audio;
 
void setup() {
    pinMode(SD_CS, OUTPUT);      
    digitalWrite(SD_CS, HIGH);
    SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
    Serial.begin(115200);
    if(!SD.begin(SD_CS))
    {
      Serial.println("Error talking to SD card!");
      while(true);  // end program
    }
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(15);
    audio.connecttoFS(SD,"/ready.wav");
}
 
void loop()
{
    audio.loop();    
}`

xuniuer, 2022-08-25:
You could add some aux functions to get more information, like the below
```
// optional
// void audio_info(const char *info){
//     Serial.print("info        "); Serial.println(info);
// }
// void audio_id3data(const char *info){  //id3 metadata
//     Serial.print("id3data     ");Serial.println(info);
// }
// void audio_eof_mp3(const char *info){  //end of file
//     Serial.print("eof_mp3     ");Serial.println(info);
// }
// void audio_showstation(const char *info){
//     Serial.print("station     ");Serial.println(info);
// }
// void audio_showstreamtitle(const char *info){
//     Serial.print("streamtitle ");Serial.println(info);
// }
// void audio_bitrate(const char *info){
//     Serial.print("bitrate     ");Serial.println(info);
// }
// void audio_commercial(const char *info){  //duration in sec
//     Serial.print("commercial  ");Serial.println(info);
// }
// void audio_icyurl(const char *info){  //homepage
//     Serial.print("icyurl      ");Serial.println(info);
// }
// void audio_lasthost(const char *info){  //stream URL played
//     Serial.print("lasthost    ");Serial.println(info);
// }

```

boguz, 2022-09-18:
i have the same problem.

I have the following code on the setup function (to debug and to try to get it working):
```
Serial.println("Setting up audio player...");
audioPlayer.setPinout(AUDIOPLAYER_BCLK, AUDIOPLAYER_LRC, AUDIOPLAYER_DOUT);
audioPlayer.i2s_mclk_pin_select(AUDIOPLAYER_BCLK);
audioPlayer.setVolume(100);

Serial.println("AFTER SETUP ========================");
Serial.println(audioPlayer.isRunning());
Serial.println(audioPlayer.getVolume());
Serial.println(audioPlayer.getBitsPerSample());
Serial.println(audioPlayer.getFileSize());

audioPlayer.connecttoSD("/amaze.mp3", 0);

Serial.println("AFTER CONNECT ========================");
Serial.println(audioPlayer.isRunning());
Serial.println(audioPlayer.getVolume());
Serial.println(audioPlayer.getBitsPerSample());
Serial.println(audioPlayer.getFileSize());
```

on the void loop i have `audioPlayer.loop();`

and now i also added the aux functions.
On the Serial monitor i get:
```
Setting up audio player...
AFTER SETUP ========================
0
21
16
0
info        PSRAM not found, inputBufferSize: 6399 bytes
info        buffers freed, free Heap: 149252 bytes
info        Reading file: "/amaze.mp3"
info        MP3Decoder has been initialized, free Heap: 121536 bytes
AFTER CONNECT ========================
1
21
16
3980786
info        Closing audio file
info        End of file "amaze.mp3"
eof_mp3     amaze.mp3
```

i think it is reading the file, because it gets the correct byte size (3980786) and we get `1` on the `.isRunning()`. But immediately after that it shows `Closing audio file`.

No idea on what the problem is...
If you want to take a look this is on the [emilBox github](https://github.com/boguz/emilBox)

schreibfaul1, 2022-09-19:
getFileSize() transfers the value of the file.size() function of the SD card. This only works as long as the file is open. Within an mp3 file there are one or more ID3 headers and sometimes images are embedded. The size of the file is bigger than the size of the audio data block.

luis-pato, 2022-09-19:
@schreibfaul1 Thank you for you answer.
Do you have any idea of what the problem is, how i can fix it or debug it any further?

schreibfaul1, 2022-09-19:
one possibility would be:
````c++
void audio_info(const char *info){
    Serial.print("audio_info: "); Serial.println(info);
    if(strcmp(info, "stream ready") == 0){
        Serial.printf("FileSize is %i bytes\n", audioPlayer.getFileSize());
    }
}
````

luis-pato, 2022-09-20:
@schreibfaul1 I have added this function to my code and i get
```
AFTER SETUP ========================
0
21
16
0
audio_info: PSRAM not found, inputBufferSize: 6399 bytes
audio_info: buffers freed, free Heap: 149260 bytes
audio_info: Reading file: "/amaze.mp3"
audio_info: MP3Decoder has been initialized, free Heap: 121544 bytes
1111111111
2
0
AFTER CONNECT ========================
1
21
16
3980786
audio_info: Content-Length: 3980786
audio_info: ID3 framesSize: 144
audio_info: ID3 version: 2.4
audio_info: ID3 normal frames
id3data     UserDefinedText: major_brandisom
id3data     UserDefinedText: minor_version512
id3data     UserDefinedText: compatible_brandsisomiso2mp41
id3data     SettingsForEncoding: Lavf58.45.100
audio_info: Audio-Length: 3980642
audio_info: stream ready
FileSize is 3980786 bytes
audio_info: syncword found at pos 0
audio_info: Channels: 2
audio_info: SampleRate: 44100
audio_info: BitsPerSample: 16
audio_info: BitRate: 128000
```

Unfortunately i have no idea what to make of these results

luis-pato, 2022-09-22:
@schreibfaul1 maybe you have an idea?

luis-pato, 2022-09-24:
Could the problem be my mp3 file. Maybe some problem with the metadata?
Here is the met data of the file i am using: https://www.metadata2go.com/result/7053de8f-2c28-4808-b40e-761d829c89b8

schreibfaul1, 2022-09-25:
I can't see that from the data. The ID3 header may be the cause. To convert an audio file to mp3 I recommend using VLC Player or Audacity.

luis-pato, 2022-09-25:
ok, thank you, i will try that