schreibfaul1/ESP32-audioI2S #339: change i2s port

alazka8, 2022-07-15:
Hi, its possible to change i2s_num_0 to i2s_num_1 , on port 0 works perfectly, but when i change port in .h file to 1 then not working ;/

schreibfaul1, 2022-07-15:
yes it is possible, the constructor must be called with parameters
```` Audio audio(false, 3, 1); ````

false: means external DAC, whereas the internal DAC is no longer supported with the newer Arduino versions
3 : means channelEnabled
```` c++
    //    possible values for channelEnabled are:
    //    I2S_DAC_CHANNEL_DISABLE  = 0,     Disable I2S built-in DAC signals
    //    I2S_DAC_CHANNEL_RIGHT_EN = 1,     Enable I2S built-in DAC right channel, maps to DAC channel 1 on GPIO25
    //    I2S_DAC_CHANNEL_LEFT_EN  = 2,     Enable I2S built-in DAC left  channel, maps to DAC channel 2 on GPIO26
    //    I2S_DAC_CHANNEL_BOTH_EN  = 0x3,   Enable both of the I2S built-in DAC channels.
    //    I2S_DAC_CHANNEL_MAX      = 0x4,   I2S built-in DAC mode max index
````
1: i2s_port_t can be 0 or 1 (=  i2s_num_0 or  i2s_num_1)

alazka8, 2022-07-19:
I set Audio audio(false,3,1) but not work ;// when i change it to Audio audio(false,3,0) it work :/

schreibfaul1, 2022-07-19:
it works for me

````c++
#include "Arduino.h"
#include "WiFiMulti.h"
#include "Audio.h"
//#include "SPIFFS.h"
//#include "FS.h"

Audio *audio;
WiFiMulti wifiMulti;
String ssid =     "*****";
String password = "*****";

void setup() {
    Serial.begin(115200);
//    uint32_t idf = ESP_IDF_VERSION_PATCH + ESP_IDF_VERSION_MINOR *10 + ESP_IDF_VERSION_MAJOR * 100;
//    uint32_t ard = ESP_ARDUINO_VERSION_PATCH + ESP_ARDUINO_VERSION_MINOR * 10 + ESP_ARDUINO_VERSION_MAJOR * 100;
//    log_i("idf %d  ard %d", idf, ard);
//    SPIFFS.begin();
    WiFi.mode(WIFI_STA);
    wifiMulti.addAP(ssid.c_str(), password.c_str());
    wifiMulti.run();
    if(WiFi.status() != WL_CONNECTED){
    WiFi.disconnect(true);
    wifiMulti.run();
    }
    audio = new Audio(false, 3, 1);
    vTaskDelay(100);
    audio->setPinout(27, 26, 25);
    audio->setVolume(7); // 0...21 Will need to add a volume setting in the app
 //   audio->connecttoFS(SPIFFS, "006.mp3");

    audio->connecttohost("http://www.wdr.de/wdrlive/media/einslive.m3u");
    Serial.printf("I2S_port is %d\n", audio->getI2sPort());
}

void loop(){
    audio->loop();
}
````

schreibfaul1, 2022-07-19:
The ES8388 doesn't care about the I2S port at all.
https://github.com/schreibfaul1/ESP32-audioI2S/tree/master/examples/ESP32-ES8388 it does not work?

alazka8, 2022-07-19:
I will check it tomorrow. I need this because on port 1 i must send music to ES but on port 0 i need to record by microphone on Internal ADC :)

alazka8, 2022-07-20:
I checked one more time on simple code:

es8388 lib - > https://github.com/vanbwodonk/es8388arduino
```
#include "Arduino.h"
#include "SD_MMC.h"
#include "FS.h"
#include "Wire.h"
#include "ES8388.h"  
#include "Audio.h"   

int Volume = 20;                            

ES8388 es8388(21, 22, 400000);
Audio audio;

void setup()
{
    Serial.begin(115200);
    SD_MMC.begin("/sdcard", true);
    es8388.init();
    es8388.inputSelect(IN1);
    es8388.setInputGain(0);
    es8388.outputSelect(OUT2);
    es8388.setOutputVolume(Volume);
    es8388.mixerSourceSelect(MIXIN1, MIXIN1);
    es8388.mixerSourceControl(true, false, 0, true, false, 0);
    audio.setPinout(32, 25, 26);
    audio.i2s_mclk_pin_select(0);
    audio.setVolume(21); // 0...21
    audio.connecttoFS(SD_MMC, "/m.mp3");
}

void loop()
{
    audio.loop();
}
```
It works, but this under ->

```

#include "Arduino.h"
#include "SD_MMC.h"
#include "FS.h"
#include "Wire.h"
#include "ES8388.h"  
#include "Audio.h"   

int Volume = 20;                            

ES8388 es8388(21, 22, 400000);
Audio audio(false,3,1);

void setup()
{
    Serial.begin(115200);
    SD_MMC.begin("/sdcard", true);
    es8388.init();
    es8388.inputSelect(IN1);
    es8388.setInputGain(0);
    es8388.outputSelect(OUT2);
    es8388.setOutputVolume(Volume);
    es8388.mixerSourceSelect(MIXIN1, MIXIN1);
    es8388.mixerSourceControl(true, false, 0, true, false, 0);
    audio.setPinout(32, 25, 26);
    audio.i2s_mclk_pin_select(0);
    audio.setVolume(21); // 0...21
    audio.connecttoFS(SD_MMC, "/m.mp3");
}

void loop()
{
    audio.loop();
}
```
Not working

alazka8, 2022-07-20:
Yee :D  i got this, thanks :) 🥇

closed 2022-11-30 by alazka8