schreibfaul1/ESP32-audioI2S #341: Radios with .TS extension extension do not work

BerranRemzi, 2022-07-16:
Some radios use m3u8playlist which contains TS files inside.
This file contains an empty video header and AAC encoded audio data.
e.g this radio have TS stream https://bss1.neterra.tv/magicfm/magicfm.m3u8

BerranRemzi, 2022-07-16:
I think this should be added to line 1835 for correct Sync word finding
`&& (buf[i+1] != 0xFFu))`

schreibfaul1, 2022-07-16:
Hello Berran,
thanks for your contribution, the syncword is FFFx which makes finding it difficult when a FFFFFF... block exists. If the tests are successful I will implement the correction.

TS files are difficult. There is probably always a sequence of 188 byte frames where the PID decides if the payload is part of the stream. But I don't really know.

BerranRemzi, 2022-07-17:
With this improvement, your library now has the ability to find the correct start offset, but still can't play TS files.
I spend several hours for find a solution but there are no success

BerranRemzi, 2022-07-18:
Thank you for your reply. I tried cloned code of Yokohama-Miyazawa, it's working with TS streams. But your code can decode almost all streams except TS.

podaen, 2022-10-09:
In the 2000's I had a dvb-c card to watch digital tv, that wasn't seen in my country at that time. I did :-) It was also in mpeg TS (transport stream). The problem is that you can not play it it properly, because it is not divided in to frames like a evry other mpeg is. You need to converted first to a valid format.

schreibfaul1, 2022-10-09:
Only AAC transport streams work, no *.ts as used with IPTV.

Example: https://air.pc.cdn.bitgravity.com/air/live/pbaudio001/playlist.m3u8
In the m3u8 playlist there is a link to https://air.pc.cdn.bitgravity.com/air/live/pbaudio001/chunklist.m3u8 and there you will find several ts files:
https://air.pc.cdn.bitgravity.com/air/live/pbaudio001/media_29493.ts
https://air.pc.cdn.bitgravity.com/air/live/pbaudio001/media_29494.ts
which are then played, after which the playlist is fetched again and contains the next *.ts files.
The *.ts files are only valid and available for a short time.

An valid AAC audio stream has the identifier #EXT-X-STREAM-INF:'mp4a.40, everything else is discarded

closed 2023-04-14 by stale[bot]