schreibfaul1/ESP32-audioI2S #352: Error Compiling when combined with https://github.com/prampec/IotWebConf

ahmadfathan, 2022-08-11:
I try to add some code from IotWebConf. Here is the code.

```
/**
   IotWebConf06MqttApp.ino -- IotWebConf is an ESP8266/ESP32
     non blocking WiFi/AP web configuration library for Arduino.
     https://github.com/prampec/IotWebConf

   Copyright (C) 2020 Balazs Kelemen <prampec+arduino@gmail.com>

   This software may be modified and distributed under the terms
   of the MIT license.  See the LICENSE file for details.
*/

/**
   Example: MQTT Demo Application
   Description:
     All IotWebConf specific aspects of this example are described in
     previous examples, so please get familiar with IotWebConf before
     starting this example. So nothing new will be explained here,
     but a complete demo application will be built.
     It is also expected from the reader to have a basic knowledge over
     MQTT to understand this code.

     This example starts an MQTT client with the configured
     connection settings.
     Will post the status changes of the D2 pin in channel "test/status".
     Receives messages appears in channel "test/action", and writes them to serial.
     This example also provides the firmware update option.
     (See previous examples for more details!)

   Software setup for this example:
     This example utilizes Joel Gaehwiler's MQTT library.
     https://github.com/256dpi/arduino-mqtt

   Hardware setup for this example:
     - An LED is attached to LED_BUILTIN pin with setup On=LOW.
     - [Optional] A push button is attached to pin D2, the other leg of the
       button should be attached to GND.
*/

#include "Arduino.h"
#include "WiFi.h"
#include "Audio.h"

#include <MQTT.h>
#include <IotWebConf.h>
#include <IotWebConfUsing.h> // This loads aliases for easier class names.

// -- Initial name of the Thing. Used e.g. as SSID of the own Access Point.
const char thingName[] = "testThing";

// -- Initial password to connect to the Thing, when it creates an own Access Point.
const char wifiInitialApPassword[] = "smrtTHNG8266";

#define STRING_LEN 128

// -- Configuration specific key. The value should be modified if config structure was changed.
#define CONFIG_VERSION "mqt1"

// -- When CONFIG_PIN is pulled to ground on startup, the Thing will use the initial
//      password to buld an AP. (E.g. in case of lost password)
#define CONFIG_PIN 5

// -- Status indicator pin.
//      First it will light up (kept LOW), on Wifi connection it will blink,
//      when connected to the Wifi it will turn off (kept HIGH).
#define STATUS_PIN 4

// -- Method declarations.
[... 212 lines cut ...]
  bool result;
  if (mqttUserPasswordValue[0] != '\0')
  {
    result = mqttClient.connect(iotWebConf.getThingName(), mqttUserNameValue, mqttUserPasswordValue);
  }
  else if (mqttUserNameValue[0] != '\0')
  {
    result = mqttClient.connect(iotWebConf.getThingName(), mqttUserNameValue);
  }
  else
  {
    result = mqttClient.connect(iotWebConf.getThingName());
  }
  return result;
}

void mqttMessageReceived(String &topic, String &payload)
{
  Serial.println("Incoming: " + topic + " - " + payload);
}
```

Then, I got this error while compiling. 

sketch\v2.ino.cpp.o:(.literal._ZN10iotwebconf17WebRequestWrapperC5Ev[iotwebconf::WebRequestWrapper::WebRequestWrapper()]+0x0): undefined reference to `vtable for iotwebconf::WebRequestWrapper'

schreibfaul1, 2022-08-11:
I've never done anything with MQTT. Have a look at the ESPUINO project,
https://forum.espuino.de/t/espuino-und-mqtt/43
https://github.com/biologist79/ESPuino

ahmadfathan, 2022-08-22:
I have solved this problem. I delete some line in i2s config.

closed 2022-08-22 by ahmadfathan