schreibfaul1/ESP32-audioI2S #354: audio.getAudioFileDuration() returns zero seconds

JTElectronicsNZ, 2022-08-16:
GitHub newbie so be gentle... Thanks :-)

I want to display the MP3 file approximate total play time (in minutes and seconds) so when I use the audio.getAudioFileDuration() function it always returns zero seconds. I can successfully calculate the play time in seconds using audio.getFileSize() and audio.getBitRate() to be 68 seconds but I don't know why audio.getAudioFileDuration() always returns zero seconds. I wait for "BitRate:" info to be displayed before I try to obtain the getAudioFileDuration information as in the code snippet below, and substitute me calculated playTime value if getAudioFileDuration is zero. Also at the bottom is the serial output (from Arduino IDE serial monitor) showing the various values with "playTime" being my calculated value. Not that it probably matters but board type is set to "ESP32 Dev Module" and the sounds play perfectly fine. I have downloaded the latest ESP32-audioI2s library code today. I get the same problem of zero file duration returned when playing MP3's from SD card or from SPIFFS, but I can always calculate them correctly. Please let me know if you need more code to replicate the problem.

```
void audio_info(const char *info) {
  Serial.print("Audio:info       "); Serial.println(info);
  const char* searchForBitRate = "BitRate:";
  if ( strstr(info, searchForBitRate) ) { //detect when "BitRate" info is displayed
    unsigned long audioFileDuration = audio.getAudioFileDuration();
    Serial.print("audioFileDuration: "); Serial.println(audioFileDuration);
    if (audioFileDuration == 0) {  //yeah, getAudioFileDuration ALWAYS returns zero...
      unsigned long fileSize = audio.getFileSize();
      Serial.print("fileSize: "); Serial.println(fileSize);
      unsigned long bitRate = audio.getBitRate();
      Serial.print("bitRate: "); Serial.println(bitRate);
      unsigned long playTime = audio.getFileSize()*8 / bitRate;
      Serial.print("playTime: "); Serial.println(playTime);
      audioFileDuration = playTime; //update to use my calculated audio play time
    }
    Serial.print("Sound will be playing for ");
    if (audioFileDuration > 59) {
      Serial.print(audioFileDuration / 60); //get whole minutes
      Serial.print(" minutes and ");
    }
    Serial.print(audioFileDuration % 60); //get remainder seconds
    Serial.print(" seconds at volume "); Serial.println(audio.getVolume() );
  }
}
```

> 21:32:37.048 -> Starting SD Card...
> 21:32:37.142 -> SD Card Type: SDHC
> 21:32:37.142 -> SD Card size: 3724MB
> 21:32:37.142 -> SD Card files:
> 21:32:37.142 -> Listing directory: /
> 21:32:37.142 ->   DIR : System Volume Information
> 21:32:37.142 ->   FILE: waiting.mp3		SIZE: 2735168
> 21:32:37.142 -> Audio:info       PSRAM found, inputBufferSize: 283615 bytes
> 21:32:37.142 -> Audio:info       buffers freed, free Heap: 270348 bytes
> 21:32:37.142 -> Audio:info       Reading file: "/waiting.mp3"
> 21:32:37.142 -> Audio:info       MP3Decoder has been initialized, free Heap: 242628 bytes
> 21:32:37.835 -> Audio:info       Content-Length: 2735168
> 21:32:37.835 -> Audio:info       ID3 framesSize: 10
> 21:32:37.835 -> Audio:info       ID3 version: 2.3
> 21:32:37.835 -> Audio:info       ID3 normal frames
> 21:32:37.882 -> Audio:info       Audio-Length: 2735158
> 21:32:37.882 -> Audio:info       stream ready
> 21:32:37.882 -> Audio:info       syncword found at pos 950
> 21:32:37.882 -> Audio:info       syncword found at pos 0
> 21:32:37.929 -> Audio:info       Channels: 2
> 21:32:37.929 -> Audio:info       SampleRate: 48000
> 21:32:37.929 -> Audio:info       BitsPerSample: 16
> 21:32:37.929 -> Audio:info       BitRate: 320000
> 21:32:37.929 -> audioFileDuration: 0
> 21:32:37.929 -> fileSize: 2735168
> 21:32:37.929 -> bitRate: 320000
> 21:32:37.929 -> playTime: 68
> 21:32:37.929 -> Sound will be playing for 1 minutes and 8 seconds at volume 10
> 21:33:46.245 -> Audio:info       Closing audio file
> 21:33:46.245 -> Audio:info       End of file "waiting.mp3"
> 21:33:46.245 -> Audio:eof_mp3    waiting.mp3
> 21:33:46.245 -> Sound has stopped playing.
>

JTElectronicsNZ, 2022-08-16:
It looks like m_avr_bitrate is zero so I added this line of code near the beginning of the Audio::getAudioFileDuration() function in file Audio.cpp and the audio.getAudioFileDuration() function now returns 64 seconds (instead of zero) which is close to correct.

`	m_avr_bitrate = m_bitRate;
`

schreibfaul1, 2022-08-16:
Hello JTElectronicsNZ,
the bit rate can be variable, I determine the bit rate several times at the beginning of the file and then form the mean value.
this is my test code, the timer is called every second
````++
ticker.attach(1, tcr1s);

void tcr1s(){
   uint32_t act=audio.getAudioCurrentTime();
   uint32_t afd=audio.getAudioFileDuration();
    if(audio.isRunning())log_i("audioCurrentTime: %i:%02d - duration: %i:%02d", (act/60), (act%60) , (afd/60), (afd%60));
}
````
[  5191][I][main.cpp:405] tcr1s(): audioCurrentTime: 0:00 - duration: 5:04
[  6191][I][main.cpp:405] tcr1s(): audioCurrentTime: 0:01 - duration: 5:12
[  7191][I][main.cpp:405] tcr1s(): audioCurrentTime: 0:02 - duration: 5:04
[  8191][I][main.cpp:405] tcr1s(): audioCurrentTime: 0:03 - duration: 5:04
[  9191][I][main.cpp:405] tcr1s(): audioCurrentTime: 0:04 - duration: 4:56
[ 10191][I][main.cpp:405] tcr1s(): audioCurrentTime: 0:05 - duration: 4:56
[ 11191][I][main.cpp:405] tcr1s(): audioCurrentTime: 0:06 - duration: 4:56
[ 12191][I][main.cpp:405] tcr1s(): audioCurrentTime: 0:07 - duration: 4:56
[ 13191][I][main.cpp:405] tcr1s(): audioCurrentTime: 0:08 - duration: 4:56
[ 14191][I][main.cpp:405] tcr1s(): audioCurrentTime: 0:09 - duration: 4:56

tested with this mp3 file
[seconds-VBR.mp3.txt](https://github.com/schreibfaul1/ESP32-audioI2S/files/9354368/seconds-VBR.mp3.txt)

closed 2023-04-14 by stale[bot]

ra0943-VK3ACH, 2024-11-13:
audio.loop has to be called before getting the file duration. I assume this is because it needs to read bytes before it can estimate the duration, but I haven't looked into this theory