schreibfaul1/ESP32-audioI2S #357: Cannot play MP3 file with zero time duration

xuniuer, 2022-08-25:
Hi there,

I guess I find a problem that the library cannot play any mp3 file with zero duration.

As follows, I have a short mp3 file with short time duration less than 1s (00:00:00), the library cannot play it successfully, at least I could not listen its sound. But it could be played well by a windows application.
```
Connect audio file: /0.mp3
info        Reading file: "/0.mp3"
info        MP3Decoder has been initialized, free Heap: 250828 bytes
info        Content-Length: 2160
info        file has no mp3 tag, skip metadata
info        Audio-Length: 2160
info        stream ready
info        syncword found at pos 0
info        Channels: 1
info        SampleRate: 16000
info        BitsPerSample: 16
info        BitRate: 32000
info        Closing audio file
```

And, when I play a file with time duration greater than 0s (00:00:01), it's OK.
```
Connect audio file: /b.mp3
info        Reading file: "/b.mp3"
info        MP3Decoder has been initialized, free Heap: 250828 bytes
info        Content-Length: 7776
info        file has no mp3 tag, skip metadata
info        Audio-Length: 7776
info        stream ready
info        syncword found at pos 0
info        Channels: 1
info        SampleRate: 16000
info        BitsPerSample: 16
info        BitRate: 32000
info        Closing audio file
info        End of file "b.mp3"
eof_mp3     b.mp3
```
How to fix it?

BR

schreibfaul1, 2022-08-25:
Please attach the mp3 file for testing, thanks

xuniuer, 2022-08-25:
Hi, thanks for your response. Attached are two test files. 
First is one with zero file duration, the other is with 1s file duration.

[1_chn.zip](https://github.com/schreibfaul1/ESP32-audioI2S/files/9424219/1_chn.zip)
[1.zip](https://github.com/schreibfaul1/ESP32-audioI2S/files/9424226/1.zip)

xuniuer, 2022-09-20:
@schreibfaul1 May I ask if or not you have tested the uploaded mp3 files? 

BTW: The files are generated by a TTS software, and they're the simplest or shortest MP3 file.

schreibfaul1, 2022-09-20:
Sorry, I had lost track of the topic, short files are now played correctly.
I tested it with this file, test tone 440Hz, duration 0.1s
[440Hz_44100Hz_16bit_0.1sec.mp3.zip](https://github.com/schreibfaul1/ESP32-audioI2S/files/9606129/440Hz_44100Hz_16bit_0.1sec.mp3.zip)

closed 2022-09-20 by schreibfaul1

xuniuer, 2022-09-21:
By the way, I search a same name library, whose version is 2.1.0 released 4 months ago, from PIO home. 
I notice that library is also with your name but with different location `esphome/ESP32-audioI2S`.  Something wrong?

xuniuer, 2022-09-21:
@schreibfaul1 I tested latest library to drive 44100Hz mp3 file, it's OK.
And I checked my uploaded test file with `mutgen`, its info is `MPEG 2 layer 3, 32000 bps (CBR?), 16000 Hz, 1 chn, 0.94 seconds`. The latest library cannot play well with it. But when I reformat it with 44100Hz, it's also OK.  The problem might be about different sample rate.

commit f69582a4 (2022-09-20): fix #357 Cannot play MP3 file with zero time duration

correct wrong dma buffersize in playI2Sremains()