schreibfaul1/ESP32-audioI2S #359: aac links doesn't work anymore.

podaen, 2022-08-27:
Any idees why the AAC links doesn't work anymore after updating? I get an exception decoder msg, see below.

> Exception:Error:
> 0x40078000: ?? ??:0
> ??:?:::0x40080400:_init
> 0x400805f0: ?? ??:0
> /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:84:::0x40092f05:strlen
> /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:96:::0x40092f15:strlen
> G:\Documents\Visual Studio 2019\Arduino\libraries\ESP32-audioI2S\src/Audio.cpp(4021): error 0x400e697e:Audio
> G:\Documents\Visual Studio 2019\Arduino\libraries\ESP32-audioI2S\src/Audio.cpp(3408): error 0x400e6abb:Audio
> G:\Documents\Visual Studio 2019\Arduino\libraries\ESP32-audioI2S\src/Audio.cpp(3034): error 0x400eb082:Audio
> G:\Documents\Visual Studio 2019\Arduino\libraries\ESP32-audioI2S\src/Audio.cpp(2396): error 0x400ebb0b:Audio
> E:\AppData\Local\Temp\VMBuilds\Player-URL-V2\espressif_esp32\Debug/fnc_audio.h(125): error 0x400d7268:PlaySong()
> (inlined by) TaskURL(void*) at G:\Documents\Visual Studio 2019\Arduino\Player-URL-V2\Player-URL-V2/Player-URL-V2.ino:343

podaen, 2022-08-27:
I had some problems before with strlen, the solution for my program was to replace it with this function. 

```
int constChrSize(const char* str)
{
	int Size = 0;
	while (str[Size] != '\0') Size++;
	return Size;
}
```

schreibfaul1, 2022-08-27:
strlen() is a classic c function that I use a lot. Where are the problems? maybe a nullpointer?

podaen, 2022-08-27:
I think it comes from Audio.cpp arround line 4021. See exception decoder message above. I only have it with AAC hyperlinks.

podaen, 2022-08-27:
If I follow my code, I think it start here.

audio.loop();

after selecting link 2 of my erray.

String urls[] = {
"http://22183.live.streamtheworld.com/JOE.mp3",//ok//OK
"http://23603.live.streamtheworld.com/JOE_80S.mp3",//nederlands//OK
"http://23553.live.streamtheworld.com/TOPRETROAAC_48_SC",//Tomorrowland
}

like this

audio.connecttohost(urls[UrlSelector].c_str());

podaen, 2022-08-27:
it stops on audio_info(chbuf); in void Audio::showCodecParams()

schreibfaul1, 2022-08-27:
I think you are looking in the wrong place, many aac streams run without problems, the link you mentioned http://23553.live.streamtheworld.com/TOPRETROAAC_48_SC has 20% packet loss, this is not acceptable
[image]

podaen, 2022-08-27:
I was so glad that my favorite radio station worked. Now non of my link do anymore.

//"http://23553.live.streamtheworld.com/TOPRETROAAC_48_SC",//Tomorrowland
//"http://22733.live.streamtheworld.com/TOPRETROAAC.aac",
//"http://21313.live.streamtheworld.com/TOPRETROAAC.aac",
//"http://21313.live.streamtheworld.com:3690/TOPRETROAAC.aac",
//"http://25503.live.streamtheworld.com:3690/TOPRETROAAC.aac",

If I play them in VLC player or http://fmstream.org/index.php?s=retro%20arena they work without a problem.

podaen, 2022-08-27:
VLC player is open source. Mybe you could have a peek?

[VLC](https://github.com/videolan/vlc)

schreibfaul1, 2022-08-27:
Yes, with the VLC player it works for me too. If you could tell the ESP to re-request a lost packet after ~100ms, it wouldn't be a problem. But the ESP32 seems to wait several seconds. Maybe that was better with a previous Arduino version.

podaen, 2022-08-27:
Maybe, They have made some improvements on past versions, and there are some changes coming.

> There is a change I've made to WiFi configuration that will be available for Arduino Core 2.0.5.
> This change is about the number of simultaneous sockets connections, that was just 1 and now it will be 4.
> It may be related to the MQTT performance issue... MQTT may need an extra check regarding WiFi IDF sdkconfig setup.

Commented by Su

schreibfaul1, 2022-08-27:
Only the link http://23553.live.streamtheworld.com/TOPRETROAAC_48_SC seems to have problems with lost packets.

//"http://22733.live.streamtheworld.com/TOPRETROAAC.aac",
//"http://21313.live.streamtheworld.com/TOPRETROAAC.aac",
//"http://21313.live.streamtheworld.com:3690/TOPRETROAAC.aac",
//"http://25503.live.streamtheworld.com:3690/TOPRETROAAC.aac",
work perfectly.

If different error messages occur frequently that cannot be assigned exactly, it may be that there is a memory problem. Maybe you can enlarge the partition?

podaen, 2022-08-27:
I once started an investigation on package lost. but It came out there wasn't. An important thing is that the buffer is emptied enough and that is what I like about your library because you use psram to maximize the input buffer. 

What I just don't understand is where the psram is located on the esp.

This is one of my partition tables.

# Name,   Type, SubType,   Offset,  Size, Flags
--
nvs, data,   nvs, 0x9000, 0x5000 |  
otadata,   data, ota, 0xe000, 0x2000
app0, app,   ota_0, 0x10000, 0x3E5000
eeprom,   data, 0x99, 0x3F5000, 0x1000
spiffs,   data, spiffs, 0x3F6000, 0xA000

Witch I don't use anymore, because I updating my ESP Arduino continuously... but it still works.

podaen, 2022-08-27:
Ok, the psram is a separate chip of 8MB.

If I am understand it right I need to decrease m_buffSizePSRAM that is currenttly...

size_t   m_buffSizePSRAM = 80000;// 300000;   // most webstreams limit the advance to 100...300Kbytes 

Currently I am using this amount of memory by my program.
#define URL_STACK_SIZE 30000
#define KEYPAD_STACK_SIZE 6000
and the defauld of 8K

podaen, 2022-08-28:
> http://22733.live.streamtheworld.com/TOPRETROAAC.aac
> 
> Station connecting done
> xPortGetFreeHeapSize: 4170795
> getFreeHeap: 69656
> getFreePsram: 4111847
> 
> keypad Task
> Setup keypad Task done
> 
> xPortGetFreeHeapSize: 4163851
> getFreeHeap: 62712
> getFreePsram: 4111847
> 
> 
> xPortGetFreeHeapSize: 4158295
> getFreeHeap: 57156
> getFreePsram: 4111847
> 
> setup TaskURL done
> 
> xPortGetFreeHeapSize: 4137911
> getFreeHeap: 36772
> getFreePsram: 4111847
> 
> 
> xPortGetFreeHeapSize: 4141407
> getFreeHeap: 40472
> getFreePsram: 4111847
> 
> 
> xPortGetFreeHeapSize: 4141643
> getFreeHeap: 40504
> getFreePsram: 4111847
> 
> 
> xPortGetFreeHeapSize: 4143339
> getFreeHeap: 42200
> getFreePsram: 4111847
> 
> 
> xPortGetFreeHeapSize: 4143339
> getFreeHeap: 42200
> getFreePsram: 4111847
> 
> 
> xPortGetFreeHeapSize: 4143339
> getFreeHeap: 42200
> getFreePsram: 4111847
> Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

I have 42200 free memory left after the loop. So I don't think it's a memory problem. 

What I don't understand is that I set
`size_t   m_buffSizePSRAM = 80000;// 300000;   // most web streams limit the advance to 100...300Kbytes`
But still the getFreePsram= 4111847  is bigger than the 80000 I set. How is this possible?

schreibfaul1, 2022-08-28:
The link works fine for me, I suspect the stack is not sufficient

podaen, 2022-09-03:
I increased the stack to 60000, That is double of the original size I used and having FreeHeap of 19448. Still having that same error exception.

> /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:84:::0x40092f2d:strlen

I tried strlen in my shetch and it works. 

I did create a new project with you [example ](https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/examples/Simple_WiFi_Radio/Simple_WiFi_Radio.ino) and I have that same error on my visual studio 2017-19. I tried different compilers but nothing changes. ("mp3.ffh.de/radioffh/hqlivestream.aac") All aac links have that problem.

What I am thinking is that the MP3 link is differently handled that for an AAC link, because it is working for MP3 but not for AAC in connecttohost and it has something to do with strlen. The way of MP3 handeling is more compatible with other machines.

podaen, 2022-09-04:
I found it! I have comment this function and causes a crash on the AAC links.

> Now I remember. You did the logging with a callback.
> 
> ```
> void audio_info(const char* info) {
> 	Serial.print("info        "); Serial.println(info);
> }
> ```

Now it's working.

closed 2022-09-04 by podaen