schreibfaul1/ESP32-audioI2S #360: WAV links doesn't work

sergi-reus-code, 2022-08-28:
I've tested this...

audio.connecttoFS(SD, "/320k_test.mp3");  ---> OK
audio.connecttoFS(SD, "/Pink-Panther.wav");   ---> DON'T PLAY      

audio.connecttohost("http://www.wdr.de/wdrlive/media/einslive.m3u");  ---> OK

audio.connecttohost("https://github.com/schreibfaul1/ESP32-audioI2S/raw/master/additional_info/Testfiles/Pink-Panther.wav");  ---> DON'T PLAY

audio.connecttohost("https://github.com/schreibfaul1/ESP32-audioI2S/raw/master/additional_info/Testfiles/320k_test.mp3");  ---> OK

audio.connecttospeech("Wenn die Hunde schlafen, kann der Wolf gut Schafe stehlen.", "de");  ---> OK

Please, can someone help me?¿

Thanks

podaen, 2022-08-28:
What error do you get in the exception decoder?

schreibfaul1, 2022-08-28:
Your system is probably running without PSRAM. Please look in the audio.h (line 467) and set the value for m_frameSizeWav from the current 1024*8 to a maximum of 1600. Does it work then?

sergi-reus-code, 2022-08-28:
GREAT.......!!!!

WORKS FINE!!!!

Thanks.....,

closed 2022-08-28 by sergi-reus-code

schreibfaul1, 2022-08-28:
thanks, I will adjust the value automatically if no PSRAM is available.