schreibfaul1/ESP32-audioI2S #363: Issue with User-Agent setting in HTTP-Request header

EFWob, 2022-08-31:
Some stations do seem to evaluate the User-Agent field in the HTTP-Request header and do not like the setting "Mozilla/5.0" and respond with Code 302 which will not result in playing any music. Issue is reproducible with the following examples:

```
http://www.memoryradio.de:4000/
http://www.memoryradio.de:5000/
http://protostar.shoutca.st:8370/
http://81.169.223.8/
http://192.99.16.17:3414/
```

Example debug output:
```
info        Connect to new host: "http://protostar.shoutca.st:8370/"
[V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.
[I][Audio.cpp:4733] ts_parsePacket(): parseTS reset
info        buffers freed, free Heap: 104936 bytes
info        connect to protostar.shoutca.st on port 8370 path /
info        Connection has been established in 84 ms, free Heap: 104224 bytes
[I][Audio.cpp:3468] parseHttpResponseHeader(): httpResponseHeader: HTTP/1.1 302 Found
[I][Audio.cpp:3468] parseHttpResponseHeader(): httpResponseHeader: Content-Type:text/html;charset=utf-8
[I][Audio.cpp:3468] parseHttpResponseHeader(): httpResponseHeader: Location:index.html?sid=1
[I][Audio.cpp:3468] parseHttpResponseHeader(): httpResponseHeader: Content-Length:118
info        content-length: 118
info        unknown content found at: http://protostar.shoutca.st:8370/
station     
icyurl      
```

### Solution
Simply remove the User-Agent line in the HTTP-request header...

schreibfaul1, 2022-09-01:
Many Thanks!

EFWob, 2022-09-01:
Danke für die prompte Erledigung!

EFWob, 2022-09-01:
Und close!

closed 2022-09-01 by EFWob