schreibfaul1/ESP32-audioI2S #369: connecttospeech

podaen, 2022-09-07:
Is it possible to add an input and an output lang? Because you are already using translate...

`bool Audio::connecttospeech(const char* speech, const char* InputLang, const char* OutputLang)`

podaen, 2022-09-20:
Is this list OK for mary marytts languages?

> /#English
> //cmu-bdl-hsmm :A male US English hidden semi-Markov model voice, built from recordings provided by Carnegie Mellon University
> //
> //cmu-slt :A female US English unit selection voice, built from recordings provided by Carnegie Mellon University
> //
> //cmu-slt-hsmm :A female US English hidden semi-Markov model voice, built from recordings provided by Carnegie Mellon University
> //
> //cmu-rms-hsmm :A male US English hidden semi-Markov model voice, built from recordings provided by Carnegie Mellon University
> //
> //dfki-obadiah :A male British English expressive unit selection voice: Gloomy Obadiah
> //
> //dfki-obadiah-hsmm :A male British English hidden semi-Markov model voice
> //
> //dfki-poppy :A female British English expressive unit selection voice: Cheerful Poppy
> //
> //dfki-poppy-hsmm :A female British English hidden semi-Markov model voice
> //
> //dfki-prudence :A female British English expressive unit selection voice: Pragmatic Prudence
> //
> //dfki-prudence-hsmm :A female British English hidden semi-Markov model voice
> //
> //dfki-spike :A male British English expressive unit selection voice: Aggressive Spike
> //
> //dfki-spike-hsmm :A male British English hidden semi-Markov model voice
> //
> //#French
> //enst-camille :A female French unit selection voice, built at Télécom ParisTech (ENST) using data recorded by Camille Dianoux
> //
> //enst-camille-hsmm :A female French hidden semi-Markov model voice, built at Télécom ParisTech (ENST) using data recorded by Camille Dianoux
> //
> //enst-dennys-hsmm :A male Québécois French hidden semi-Markov model voice, built at Télécom ParisTech (ENST)
> //
> //upmc-jessica :A female French unit selection voice, built at ISIR (UPMC) using data recorded by Jessica Durand
> //
> //upmc-jessica-hsmm :A female French hidden semi-Markov model voice, built at ISIR (UPMC) using data recorded by Jessica Durand
> //
> //upmc-pierre :A male French unit selection voice, built at ISIR (UPMC) using data recorded by Pierre Chauvin
> //
> //upmc-pierre-hsmm :A male French hidden semi-Markov model voice, built at ISIR (UPMC) using data recorded by Pierre Chauvin
> //
> //#German
> //dfki-pavoque-neutral :A male German unit selection voice
> //
> //dfki-pavoque-neutral-hsmm :A male German hidden semi-Markov model voice
> //
> //dfki-pavoque-styles :A male German unit selection voice with expressive styles "happy", "sad", "angry", and "poker"
> //
> //bits1-hsmm :A female German hidden semi-Markov model voice, built from voice recordings provided by the BITS project at the Bavarian Archive of Speech Signals
> //
> //bits3 :A male German unit selection voice, built from voice recordings provided by the BITS project at the Bavarian Archive of Speech Signals
> //
> //bits3-hsmm :A male German hidden semi-Markov model voice, built from voice recordings provided by the BITS project at the Bavarian Archive of Speech Signals
> //
> //#Italian
> //istc-lucia-hsmm :Italian female Hidden semi-Markov model voice kindly made available by Fabio Tesser
> // 
> //#Turkish
> //dfki-ot :A male Turkish unit selection voice
> //
> //dfki-ot-hsmm :A male Turkish hidden semi-Markov model voice
> //
> //#Russian
> //voxforge-ru-nsh :Russian male voice kindly made available by Nickolay V. Shmyrev
> // 
> //#Telugu
> //cmu-nk :A female Telugu unit selection voice built from voice recordings provided by IIIT Hyderabad and Carnegie Mellon University
> //
> //cmu-nk-hsmm :A female Telugu hidden semi-Markov model voice built from voice recordings provided by IIIT Hyderabad and Carnegie Mellon University

This doesn't work.

SelectVoice(msg.c_str(), "en", "cmu-bdl-hsmm");

podaen, 2022-09-20:
I tried the "en-US" too... it didn't work.

schreibfaul1, 2022-09-20:
Hello podaen,
Thanks for the tip, the MaryTTS service is not responding. The website 'mary.dfki.de' is still online. MaryTTS hasn't been developed for years, maybe the service will be shut down completely?

podaen, 2022-09-21:
I read this somewhere to make it work in english... 

> You need to understand that the languages and voices are different components. When you build MaryTTS from source, you build all language components, and one voice (cmu-slt-hsmm). All other voices are installed from the Web via the Installer GUI.
> 
> See https://github.com/marytts/marytts#downloading-and-installing-voices

Soo I tried a onther voice with the basic example and now it works.

` audio.connecttomarytts("This is a test. Nice to hear that it works.", "en-us", "cmu-slt-hsmm");`

podaen, 2022-09-24:
Google TTS doesn't stop in dutch when I use the example. Keep getting  "info        slow stream, dropouts are possible".

`audio.connecttospeech("Dit is een test. Blij the horen dat het werkt.", "nl");`

podaen, 2022-09-24:
> Google TTS doesn't stop in dutch when I use the example. Keep getting "info slow stream, dropouts are possible".
> 
> `audio.connecttospeech("Dit is een test. Blij the horen dat het werkt.", "nl");`

I downloaded the latest. It's working now!

closed 2022-09-24 by podaen

podaen, 2022-09-27:
> Is it possible to add an input and an output lang? Because you are already using translate...
> 
> `bool Audio::connecttospeech(const char* speech, const char* InputLang, const char* OutputLang)`

I see now that's there is no way to do that. Sorry about that.

stieneker, 2023-10-30:
Something lik this word perfikt in German:

audio.connecttospeech("Hallo Leute, wie geht es euch?", "de");

:) :)