schreibfaul1/ESP32-audioI2S #370: TTS

optimaxch, 2022-09-08:
wolle,
working fine till about 5 words -
than continues in " scratchy sound "

  danke
  rudi

optimaxch, 2022-09-08:
s o r r y,
i overstressed ESP32 pico with additional 
Serial.print  cmd
-- this caused the   scratch-sound

ALL WORKING FINE
ALSO STREAM

   thanks
   rudi

failure:
 while (connect ==true)
   {
       audio.loop();
       //Serial.print("audio loop sleep           ");
   }    
without Serial.print  WORKING!

schreibfaul1, 2022-09-08:
Hello,
Serial.print is too slow to be called continuously in a loop. 
Maybe it goes like this:
````c++
int i = 0;
while(connected){
    audio.loop();
    if(!i) Serial.Print("sleep");
    i++;
    if(i == 1000) i = 0;
}
````

closed 2023-05-15 by stale[bot]