schreibfaul1/ESP32-audioI2S #373: Does this library not support esp32 c3 chips?

llinzzi, 2022-09-23:
This library is working pretty good on my esp32. but seem not work on esp32 c3.
`
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.cpp: In function 'void UnpackQuads(int, int, int*)':
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.cpp:3289:106: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
         nCodeBits = DecodeHuffmanScalar(huffTabSpec, &huffTabSpecInfo[cb - HUFFTAB_SPEC_OFFSET], bitBuf, &val);
                                                                                                          ^~~~
In file included from d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.cpp:9:
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.h:490:115: note:   initializing argument 4 of 'int DecodeHuffmanScalar(const short int*, const HuffInfo_t*, unsigned int, int32_t*)'
 int DecodeHuffmanScalar(const signed short *huffTab, const HuffInfo_t *huffTabInfo, unsigned int bitBuf, int32_t *val);
                                                                                                          ~~~~~~~~~^~~
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.cpp: In function 'void UnpackPairsNoEsc(int, int, int*)':
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.cpp:3336:104: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
         nCodeBits = DecodeHuffmanScalar(huffTabSpec, &huffTabSpecInfo[cb-HUFFTAB_SPEC_OFFSET], bitBuf, &val);
                                                                                                        ^~~~
In file included from d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.cpp:9:
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.h:490:115: note:   initializing argument 4 of 'int DecodeHuffmanScalar(const short int*, const HuffInfo_t*, unsigned int, int32_t*)'
 int DecodeHuffmanScalar(const signed short *huffTab, const HuffInfo_t *huffTabInfo, unsigned int bitBuf, int32_t *val);
                                                                                                          ~~~~~~~~~^~~
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.cpp: In function 'void UnpackPairsEsc(int, int, int*)':
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.cpp:3378:104: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
         nCodeBits = DecodeHuffmanScalar(huffTabSpec, &huffTabSpecInfo[cb-HUFFTAB_SPEC_OFFSET], bitBuf, &val);
                                                                                                        ^~~~
In file included from d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.cpp:9:
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.h:490:115: note:   initializing argument 4 of 'int DecodeHuffmanScalar(const short int*, const HuffInfo_t*, unsigned int, int32_t*)'
 int DecodeHuffmanScalar(const signed short *huffTab, const HuffInfo_t *huffTabInfo, unsigned int bitBuf, int32_t *val);
                                                                                                          ~~~~~~~~~^~~
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.cpp: In function 'int DecodeOneScaleFactor()':
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.cpp:4179:82: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
     nBits = DecodeHuffmanScalar(huffTabScaleFact, &huffTabScaleFactInfo, bitBuf, &val);
                                                                                  ^~~~
In file included from d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.cpp:9:
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.h:490:115: note:   initializing argument 4 of 'int DecodeHuffmanScalar(const short int*, const HuffInfo_t*, unsigned int, int32_t*)'
 int DecodeHuffmanScalar(const signed short *huffTab, const HuffInfo_t *huffTabInfo, unsigned int bitBuf, int32_t *val);
                                                                                                          ~~~~~~~~~^~~
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.cpp: In function 'void GenerateNoiseVector(int*, int*, int)':
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.cpp:5313:41: error: invalid conversion from 'uint32_t*' {aka 'long unsigned int*'} to 'unsigned int*' [-fpermissive]
         coef[i] = ((int32_t)Get32BitVal((uint32_t *)last)) >> 16;
                                         ^~~~~~~~~~~~~~~~
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.cpp:5146:40: note:   initializing argument 1 of 'unsigned int Get32BitVal(unsigned int*)'
 unsigned int Get32BitVal(unsigned int *last)
                          ~~~~~~~~~~~~~~^~~~
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.cpp: In function 'int DecodeOneSymbol(int)':
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.cpp:8549:57: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
     nBits = DecodeHuffmanScalar(huffTabSBR, hi, bitBuf, &val);
                                                         ^~~~
In file included from d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.cpp:9:
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\aac_decoder\aac_decoder.h:490:115: note:   initializing argument 4 of 'int DecodeHuffmanScalar(const short int*, const HuffInfo_t*, unsigned int, int32_t*)'
 int DecodeHuffmanScalar(const signed short *huffTab, const HuffInfo_t *huffTabInfo, unsigned int bitBuf, int32_t *val);
                                                                                                          ~~~~~~~~~^~~
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\Audio.cpp: In member function 'void Audio::processWebStream()':
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\Audio.cpp:3005:65: error: no matching function for call to 'min(uint32_t&, size_t)'
         availableBytes = min(availableBytes, InBuff.writeSpace());
                                                                 ^
In file included from c:\users\zulin\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3\riscv32-esp-elf\include\c++\8.4.0\algorithm:62,
                 from C:\Users\zulin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5\cores\esp32/Arduino.h:162,
                 from d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\Audio.h:15,
                 from d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\Audio.cpp:11:
c:\users\zulin\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algo.h:3456:5: note: candidate: 'template<class _Tp, class _Compare> _Tp std::min(std::initializer_list<_Tp>, _Compare)'
     min(initializer_list<_Tp> __l, _Compare __comp)
     ^~~
c:\users\zulin\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algo.h:3456:5: note:   template argument deduction/substitution failed:
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\Audio.cpp:3005:65: note:   mismatched types 'std::initializer_list<_Tp>' and 'long unsigned int'
         availableBytes = min(availableBytes, InBuff.writeSpace());
                                                                 ^
In file included from c:\users\zulin\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3\riscv32-esp-elf\include\c++\8.4.0\algorithm:62,
                 from C:\Users\zulin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5\cores\esp32/Arduino.h:162,
                 from d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\Audio.h:15,
                 from d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\Audio.cpp:11:
c:\users\zulin\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algo.h:3450:5: note: candidate: 'template<class _Tp> _Tp std::min(std::initializer_list<_Tp>)'
     min(initializer_list<_Tp> __l)
[... 50 lines cut ...]
                                                             ^
[... cut ...]
c:\users\zulin\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algobase.h:195:5: note:   template argument deduction/substitution failed:
d:\dev\esp32\libraries\ESP32-audioI2S-2.0.5\src\Audio.cpp:3064:61: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned int' and 'uint32_t' {aka 'long unsigned int'})
     availableBytes = min(InBuff.writeSpace(), availableBytes);

exit status 1

Compilation error: exit status 1
`

podaen, 2022-10-09:
I am think that the ESP32-S3 will support other protocols than `WIFI_PROTOCOL_11B` like the old ESP32 and that's limited to 10Mbs, I tested it this weekend to set anything else but it didn't.

Are you pepaired to use my code, if you could set `WIFI_PROTOCOL_11G` or `WIFI_PROTOCOL_11N` with the ESP32-S3?

Here is my code

```
#include "esp_wifi.h"
#include "esp_log.h"

esp_err_t err;

wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	err = esp_wifi_init(&cfg);
	if (err != ESP_OK) {
		Serial.println(err);
		Serial.println("Could not set wifi power mode!");
		return false;
	}

	err = esp_wifi_set_mode(WIFI_MODE_STA);
	if (err != ESP_OK) {
		Serial.println(err);
		Serial.println("Could not set sta!");
		return false;
	}

	//const uint8_t protocol = WIFI_PROTOCOL_11G;
	//const uint8_t protocol = WIFI_PROTOCOL_11B | 0 | 0;
	const uint8_t protocol = WIFI_PROTOCOL_11B;

	err = esp_wifi_set_protocol(WIFI_IF_STA, protocol);
	if (err != ESP_OK) {
		//258 = ESP_ERR_INVALID_ARG
		Serial.println(err);
		Serial.println("Could not set protocol!");
		return false;
	}

	uint8_t getprotocol;
	//esp_err_t err;
	err = esp_wifi_get_protocol(WIFI_IF_STA, &getprotocol);
	if (err != ESP_OK) {
		Serial.println("Could not get protocol!");
		//log_e("Could not get protocol! %d", err);
		return false;
	}
	if (getprotocol & WIFI_PROTOCOL_11N) {
		Serial.println("WiFi_Protocol_11n");
	}
	if (getprotocol & WIFI_PROTOCOL_11G) {
		Serial.println("WiFi_Protocol_11g");
	}
	if (getprotocol & WIFI_PROTOCOL_11B) {//worst, but dificuld to set else
		Serial.println("WiFi_Protocol_11b");
	}
```

It will not log you in to the WiFi, but you can see if you you where able to set other with a higher transfer speed. It think it would help to make the ESP more stable, if there more internet tasks needed, like in my case. The default is (WIFI_PROTOCOL_11B|WIFI_PROTOCOL_11G|WIFI_PROTOCOL_11N).

podaen, 2022-10-17:
When I read your fault codes again. I see you have a problem with toolchains. On what software are you programming?

hmax42, 2022-11-24:
duplicate of https://github.com/schreibfaul1/ESP32-audioI2S/issues/387
esp32-c3 only has a single core

closed 2023-05-15 by stale[bot]