schreibfaul1/ESP32-audioI2S #377: SdFat and FS

podaen, 2022-10-04:
I want to implement a webserver that is using the FS lib and I had already the SdFat in my program. I didn't know what I was doing wrong... so I raised the question at Greiman and this what he says. I don't know the audio.h enough to understand what I have to do further, so I am asking you. 

@schreibfaul1 Please read this [SdFat](https://github.com/greiman/SdFat/issues/403)

In Audio.h I changed

```
#ifdef SDFATFS_USED
#include <SdFat.h>  // https://github.com/greiman/SdFat
#else
#include <SD.h>
#include <SD_MMC.h>
#include <SPIFFS.h>
#include <FS.h>
#include <FFat.h>
#endif // SDFATFS_USED
```
And

```
#ifdef SDFATFS_USED
//typedef File32 File;
typedef FsFile File;
```
These are the lines where I get stuck on.

https://github.com/schreibfaul1/ESP32-audioI2S/blob/e48c95805a9dd6e006860569d5b4b20d01a568d6/src/Audio.h#L73

https://github.com/schreibfaul1/ESP32-audioI2S/blob/e48c95805a9dd6e006860569d5b4b20d01a568d6/src/Audio.h#L458

https://github.com/schreibfaul1/ESP32-audioI2S/blob/e48c95805a9dd6e006860569d5b4b20d01a568d6/src/Audio.h#L54

https://github.com/schreibfaul1/ESP32-audioI2S/blob/e48c95805a9dd6e006860569d5b4b20d01a568d6/src/Audio.h#L57

schreibfaul1, 2022-10-04:
The reason I implemented the SDFat Lib is to use filenames in UTF-8 or UTF16 format. (SD in ESP32 only knows ASCII) Thus, Russian, Chinese, Japanese... file names can be used.
In older versions, SDFat was incompatible with the ESP32 libraries SD.h, SD_MMC.h. FS.h, FFat.h ... SPIFFS.h).
The SDFat developer writes "Just use FsFile in place of File and SdFs in place of SdFat. "
I'll look at that in the next few days.

podaen, 2022-10-04:
Hi, wolle. yes it was. I like the SdFat to because it supports long file names. I'm upgrading my software too.

If I place in this order...

```
#include <SdFat.h>
#include <FS.h>
#include <WebServer.h>
#include <Audio.h>
```

Now the class SDFATFS is not a problem anymore. Still I got these compiling errors I don't know how to solve. 

> Audio.cpp: In member function bool Audio::connecttoFS(fs::FS&, const char*, uint32_t)
>  
> Audio.cpp: 677:38: error: no match for 'operator=' (operand types are 'File' {aka 'FsFile'} and 'File32')
>    audiofile = fs.open(audioName); 
>  
> FsLib.h:32: In file included from
> SdFat.h:35: from
> Audio.h:26: from
> Audio.cpp:11: from
> FsFile.h:796: note  candidate  FsFile& FsFile  operator=(const FsFile&)
>  class FsFile *: public StreamFile<FsBaseFile, uint64_t> {
>    ^~~~~~
> FsFile.h:796: note    no known conversion for argument 1 from File32 to const FsFile&
> FsFile.h:796: note  candidate  FsFile& FsFile  operator=(FsFile&&)
> FsFile.h:796: note    no known conversion for argument 1 from File32 to FsFile&&
>  
> Audio.cpp: 683:42: error: no match for 'operator=' (operand types are 'File' {aka 'FsFile'} and 'File32')
>    audiofile = fs.open(audioName)
>  
> FsLib.h:32: In file included from
> SdFat.h:35: from
> Audio.h:26: from
> Audio.cpp:11: from
> FsFile.h:796: note  candidate  FsFile& FsFile  operator=(const FsFile&)
>  class FsFile *: public StreamFile<FsBaseFile, uint64_t> {
>    ^~~~~~
> FsFile.h:796: note    no known conversion for argument 1 from File32 to const FsFile&
> FsFile.h:796: note  candidate  FsFile& FsFile  operator=(FsFile&&)
> FsFile.h:796: note    no known conversion for argument 1 from File32 to FsFile&&
> Audio.cpp: In member function void Audio::processLocalFile()
>  
> Audio.cpp: 2859:72: error: no matching function for call to 'min(uint32_t&, long long unsigned int)
>    availableBytes = min(availableBytes, audiofile.size() - byteCounter)
>  
> algorithm:62: In file included from
> Arduino.h:162: from
> Audio.h:15: from
> Audio.cpp:11: from
> stl_algo.h:3456: note  candidate  template<class _Tp, class _Compare> _Tp std  min(std  initializer_list<_Tp>, _Compare)
>    min(initializer_list<_Tp> __l, _Compare __comp)
>    ^~~
> stl_algo.h:3456: note    template argument deduction\substitution failed
> Audio.cpp:2859: note    mismatched types std  initializer_list<_Tp> and unsigned int
>    availableBytes = min(availableBytes, audiofile.size() - byteCounter)
>  
> algorithm:62: In file included from
> Arduino.h:162: from
> Audio.h:15: from
> Audio.cpp:11: from
> stl_algo.h:3450: note  candidate  template<class _Tp> _Tp std  min(std  initializer_list<_Tp>)
>    min(initializer_list<_Tp> __l)
>    ^~~
> stl_algo.h:3450: note    template argument deduction\substitution failed
> Audio.cpp:2859: note    mismatched types std  initializer_list<_Tp> and unsigned int
>    availableBytes = min(availableBytes, audiofile.size() - byteCounter)
>  
> vector:60: In file included from
> Audio.h:14: from
> Audio.cpp:11: from
> stl_algobase.h:243: note  candidate  template<class _Tp, class _Compare> const _Tp& std  min(const _Tp&, const _Tp&, _Compare)
>    min(const _Tp& __a, const _Tp& __b, _Compare __comp)
>    ^~~
> stl_algobase.h:243: note    template argument deduction\substitution failed
> Audio.cpp:2859: note    deduced conflicting types for parameter const _Tp (unsigned int and long long unsigned int)
>    availableBytes = min(availableBytes, audiofile.size() - byteCounter)
>  
> vector:60: In file included from
> Audio.h:14: from
> Audio.cpp:11: from
> stl_algobase.h:195: note  candidate  template<class _Tp> const _Tp& std  min(const _Tp&, const _Tp&)
>    min(const _Tp& __a, const _Tp& __b)
>    ^~~
> stl_algobase.h:195: note    template argument deduction\substitution failed
> Audio.cpp:2859: note    deduced conflicting types for parameter const _Tp (unsigned int and long long unsigned int)
>    availableBytes = min(availableBytes, audiofile.size() - byteCounter)
> Error compiling libraries

Thanks for the acknowledge.

podaen, 2022-10-07:
I want to use the webserver at bootup for a dynamic WiFi login. After that it is not in use anymore.

podaen, 2022-10-09:
Dit it work?