schreibfaul1/ESP32-audioI2S #379: compile error on line m_pin_config.mck_io_num   = MCK;

elshnkhll, 2022-10-08:
Arduino IDE
compile error on line `m_pin_config.mck_io_num   = MCK;`

ESP32 core version: 2.0.0

Code:
`Audio audio = Audio(true, I2S_DAC_CHANNEL_RIGHT_EN);

Board:
LILYGO® TTGO T5 V2.2 ESP32 2.9" where speaker is attached to internal DAC channel on GPIO25

schreibfaul1, 2022-10-08:
Hello elshnkhall,
with the last versions the internal DAC did not work. From V2.0.5 the defect has been eliminated. I created an example for this:
https://github.com/schreibfaul1/ESP32-audioI2S/tree/master/examples/Internal%20DAC

elshnkhll, 2022-10-08:
Thanks! Clean compile with v2.0.5, but sound is terrible.
Any suggestions to improve it?
Her is my environment:

`Board: "ESP32 Dev Module"
Upload Speed: "921600"
CPU Frequency: "80MHz (WiFi/BT)"
Flash Frequency: "80MHz"
Flash Mode: "QIO"
Flash Size: "4MB (32Mb)"
Partition Scheme: "Huge APP (3MB No OTA/1MB SPIFFS)"
Core Debug Level: "None"
PSRAM: "Enabled"
Arduino Runs On: "Core 1"
Events Run On: "Core 1"
Erase All Flash Before Sketch Upload: "Enabled"`

closed 2022-10-08 by elshnkhll