schreibfaul1/ESP32-audioI2S #380: void getSource()

podaen, 2022-10-08:
Getting the source would be handy because of the length can be different. SD and speech will end, but the radio not.

schreibfaul1, 2022-10-08:
In the case of web files (podcasts, audio files), the advertising server transmits a content length. Then the bytes are counted until the end is reached. This is not the case with radio streams.

podaen, 2022-10-08:
It's the radio where I having problems with. I can not turn it automatic off, because it is maybe other audio file I don't know. If I had a function to check if possible...

podaen, 2022-10-09:
Ok. So there is some [documentation](https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/examples/separate_audiotask/separate_audiotask.ino).

podaen, 2022-10-09:
Are the Queue playing until finished or will the next Queue override the previous one?

podaen, 2022-10-09:
I have it working now, it was easier than I though. I have created a playlist for all file types.

> Are the Queue playing until finished or will the next Queue override the previous one?

I think I see now, it doesn't override until it reached the end like you told me before.