schreibfaul1/ESP32-audioI2S #381: Get Data for a Spectrum Analyzer

ProfZiebart, 2022-10-09:
Hi, first of all thank you for your excellent Work.

I built a Webradio based on your Library with the Arduino IDE and an ESP32. Now I thought about adding a (1 band) spectrometer.

in the [blotfi](https://github.com/blotfi) /
[ESP32-Radio-with-Spectrum-analyzer](https://github.com/blotfi/ESP32-Radio-with-Spectrum-analyzer) they added a spectrum analyzer to the Display. 

This is done by a routine displaying the bars ( I think about some LEDs, frankly)
and a snippet getting the maximum volume directly from the Codec:
`void VS1053::getBands()
{
    write_register(SCI_WRAMADDR, BASE+2);
    bands = read_register(SCI_WRAM);
    write_register(SCI_WRAMADDR, BASE+4);
    for (uint8_t i = 0; i < 14; i++) {
        uint8_t val = read_register(SCI_WRAM);
        /* current value in bits 5..0, normally 0..31
        peak value in bits 11..6, normally 0..31 */
        uint8_t cur1 = val & 31;
        uint8_t cur = (cur1*curvol)/30; // big bars
        if ( cur < 0 ) cur = 0;
        else if ( cur > (Spectrum_hy-4) ) cur = Spectrum_hy-4;
        spectrum[i][0] = cur;
    }
}`

I assume you wrote your own MP3 Codec, therefore I have no entry point where to get my beloved information about the current playing maximum Volume. 
Could you give me a hint where to look or how to realize this?

podaen, 2022-10-11:
The easiest way to do it with an analog signal. don't look for the output, because that is digital. All incoming data is also digital, so you need an ADC. In most cases the data will be sterio, so you need to collect one byte and skip the next one... and so on. That's what you want to do.

ProfZiebart, 2022-10-31:
This is trivial and no answer to my problem. I look for a digital solution like blotfi already HAS. Measuring analog signals costs energy I don't want to spend.

podaen, 2022-10-31:
That's how I allready did. But feel free to make your own code!

ProfZiebart, 2022-10-31:
Yeah, I want to but I need an entry point in the current MP3 Codec. Then I will. I can't find the peak value yet.

podaen, 2022-10-31:
Can you from a compressed format?? You probably want PCM.

ProfZiebart, 2022-11-24:
blotfi did, so I do think you can. Otherwise I would say you're right but somehow they managed it.

podaen, 2022-11-24:
What can?

schreibfaul1, 2022-11-24:
Well, there are two ways to decouple the I2S signal, as a block ``void audio_process_extern(int16_t* buff, uint16_t len, bool *continueI2S)``or individual samples ``audio_process_i2s(uint32_t* sample, bool *continueI2S)``
The ESPUino project https://github.com/biologist79/ESPuino uses this to transmit to a BT speaker. A Fast Fourier Transform should be possible with some skill.

podaen, 2022-11-24:
And I have a third way. 😀

schreibfaul1, 2022-11-24:
Do you use the analog signal after the DAC for this?

podaen, 2022-11-24:
No looked in to the code and found an other way. It reminds me that I have build after the dac a long time a go. I almost forgot... and that's the best way to do it, because more sources can be added and it will improve the audio by take all those algorithms out of the player.

ProfZiebart, 2023-05-08:
@podaen How did you do it?

podaen, 2023-06-09:
> Do you use the analog signal after the DAC for this?

Not after the DAC, but there are so many ways. After the decoder by split the signal. but you could if you know V0. You only need the positive site to make a spectrum analyzer. Everything below relative 0, you leave out.

closed 2024-03-21 by github-actions[bot]