schreibfaul1/ESP32-audioI2S #388: how can I get the mp3 list from the sd_mmc card, then I can paly it one by one?

justdomyself, 2022-10-18:
how can I get the mp3 list from the sd_mmc card, then I can paly it one by one?

the example give is already know the mp3 name, but the real work mode need you scan the sd card ,get the mp3 names ,then play it

biologist79, 2022-10-19:
Well, basically that's your job!
SDMMC: https://github.com/espressif/arduino-esp32/blob/master/libraries/SD_MMC/examples/SDMMC_Test/SDMMC_Test.ino
Playlist-Managment: https://github.com/biologist79/ESPuino/blob/master/src/SdCard.cpp#L213. With filename/directoryname as input it creates an array of strings (char**).

To make it short: can be pretty complex.

justdomyself, 2022-10-24:
> 

thanks ,it can work.  but i find a problem, this just support english filename, can not support chinese filename

closed 2022-10-24 by justdomyself

schreibfaul1, 2022-10-24:
https://github.com/espressif/arduino-esp32/issues/7246