schreibfaul1/ESP32-audioI2S #394: Does this lib surpport flac format?

justdomyself, 2022-10-24:
I use it play mp3 is ok, but i use it play flac format music, there is no sound。

I find  source files in the examples about flac, so I think maybe it support  flac format musics。

schreibfaul1, 2022-10-24:
This works as long as flac is not embedded in an ogg stream. PSRAM is required.
Example: audio.connecttohost("https://raw.githubusercontent.com/schreibfaul1/ESP32-audioI2S/master/additional_info/Testfiles/Santiano-Wellerman.flac");

justdomyself, 2022-10-25:
> 
I mean I play the music on the sd card, the mp3 format is ok ?but  flac can't work

justdomyself, 2022-11-20:
> 
i use esp32 wrover module, it has psram。 how can i use it with your code .I do not understand " not embedded in an ogg stream. ", what does it mean?

schreibfaul1, 2022-11-20:
ogg is a wrapper to make FLAG capable of streaming and is not currently supported

schreibfaul1, 2022-11-21:
Only 'native' FLAC works. For example, files from the SD card (or from the webserver)

````
audio.connecttohost("https://github.com/schreibfaul1/ESP32-audioI2S/raw/master/additional_info/Testfiles/Santiano-Wellerman.flac");
````

closed 2023-05-15 by stale[bot]