schreibfaul1/ESP32-audioI2S #408: Problem with GPIO0 on ESP32, seems used by the library

thierry7100, 2022-11-07:
Hi,

I have used your library to play mP3 files. It is working fine, but I have a strange issue. The GPIO0 pin is not available when using your library, the pin is used as an Output and I don't know for which purpose.
In the code, I have seen the function Audio::i2s_mclk_pin_select(const uint8_t pin) where pin could be 0, 1 or 3 but this function doesn't seem to be called... 

It's a problem in my case, because all remaining pins of my ESP are in use.

Thanks in advance

schreibfaul1, 2022-11-07:
If GPIO0 is not used as MCLK, it can be used.
The example works for me:
````c++
void setup() {
    ....
    pinMode(0, INPUT_PULLUP);
}

void loop(){
    audio.loop();
    ....
    if(digitalRead(0) == LOW){
        Serial.println("GPIO 0 is low");
    }
}    
````

dominicgermana, 2023-01-08:
I have also observed that GPIO0 is controlled by the Audio library.  Running the example below, GPIO0 measures 3.3 V after reboot, but once the serial port is connected it drops to 1.6 V.  I haven't looked at the signal on an oscilloscope, but I assume it's a clock output (presumably MCLK) because the average voltage is halved.

If the audio.setPinout() line is commented out, GPIO0 voltage remains at 3.3 V.

I found a workaround that allows the use of all GPIO pins.  I assign DOUT, LRC, and BCLK to pins 0, 1, and 3 (the three options for MCLK).  In this case, GPIO0 is no longer MCLK, but DOUT instead.  Audio playback works properly.

Test configuration:  PlatformIO in VS Code
```
[env:um_pros3]
platform = espressif32
board = um_pros3
framework = arduino
lib_deps = 
	esphome/ESP32-audioI2S@^2.0.6
```

Example code demonstrating the problem:

```#include <Arduino.h>
#include <WiFi.h>
#include "Audio.h"
#include "SD.h"
#include "FS.h"

// These are the hardware SPI pins for the ESP32 ProS3 board. Used for SD card access.
#define SPI_MOSI    (35)
#define SPI_MISO    (37)
#define SPI_SCK     (36)
#define SD_CS       (34)  // SD card chip select

// I2S (digital audio data transfer) pin definitions - these can be any GPIO pins
#define I2S_LRC       1
#define I2S_DOUT      2
#define I2S_BCLK      3

// Integer for default audio volume level
int volume = 2;

Audio audio;  // Audio playback object
 
void setup() {
  // Set GPIO0 as input pullup (but it will end up controlled as an MCLK output by the Audio library)
  pinMode(0, INPUT_PULLUP);

  // Set microSD Card CS as OUTPUT and set HIGH
  pinMode(SD_CS, OUTPUT);      
  digitalWrite(SD_CS, HIGH); 
  
  // Initialize SPI bus for microSD Card
  SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);

  Serial.begin(115200);
  while (!Serial) { 
      ; // Wait for serial to connect 
  }

  // Start microSD Card
  if(!SD.begin(SD_CS)) {
    Serial.println("Error accessing microSD card!");
    while (true); // halt 
  }
  
  // Setup I2S pins (this step causes GPIO0 to become the MCLK output)
  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);

  // Set default volume
  audio.setVolume(volume);

  Serial.println("Starting to play the sound file...");
  audio.connecttoFS(SD,"/gs-16b-2c-44100hz.mp3");
}

void loop(void) {
  audio.loop();
}

// Print audio library information
void audio_info(const char *info){
  Serial.print("info        "); Serial.println(info);
}

// Callback for end of mp3 file
void audio_eof_mp3(const char *info){  //end of file
  Serial.print("eof_mp3     ");Serial.println(info);
}
```

thierry7100, 2023-01-09:
Hi,

Thanks for the update.

To overcome the issue, I have a modified version of the library.

The only change is insetPinout, I initialize mck_io_num to 
I2S_PIN_NO_CHANGE instead of letting it unitialized, and it is working fine.

Thierry

//---------------------------------------------------------------------------------------------------------------------
bool Audio::setPinout(uint8_t BCLK, uint8_t LRC, uint8_t DOUT, int8_t DIN) {

     m_pin_config.mck_io_num  = I2S_PIN_NO_CHANGE;
     m_pin_config.bck_io_num   = BCLK;
     m_pin_config.ws_io_num    = LRC; //  wclk
     m_pin_config.data_out_num = DOUT;
     m_pin_config.data_in_num  = DIN;

     const esp_err_t result = i2s_set_pin((i2s_port_t) m_i2s_num, 
&m_pin_config);
     return (result == ESP_OK);
}