schreibfaul1/ESP32-audioI2S #412: Unsatisfied dependency WiFiClientSecure.h

cap60552, 2022-11-15:
I have been trying to add ESP32-audioI2S  into a project using Platform.IO / VScode and I have been unable to find the right combination of additional libraries to resolve the dependency on  WiFiClientSecure.h

schreibfaul1, 2022-11-15:
I use PlatformIO frequently. WiFiClientSecure is included in the Arduino ESP32 package. It is not necessary to include such header files.
Here is my platformio.ini:
````
PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#v5.2.0   ; Arduino V2.0.5 ESP-IDF v4.4.2

board = esp32dev ;chipmodel ESP32, 4M FLASH, USBtoTTL

board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
framework = arduino
monitor_speed = 115200
monitor_filters = direct, esp32_exception_decoder
board_build.partitions = huge_app.csv
upload_speed = 460800  ; 921600, 512000, 460800, 256000, 115200

lib_deps =
	https://github.com/schreibfaul1/ESP32-audioI2S.git

platform_packages =
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.5 ; run bootloader in dio mode
;    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#master ; newest release

board_upload.maximum_size = 3145728
board_upload.flash_size = 4MB
board_build.flash_mode = qio
board_build.bootloader = dio

;build_flags = -DCORE_DEBUG_LEVEL=0    ; None
;build_flags = -DCORE_DEBUG_LEVEL=1    ; Error
;build_flags = -DCORE_DEBUG_LEVEL=2    ; Warn
;build_flags = -DCORE_DEBUG_LEVEL=4    ; Debug
;build_flags = -DCORE_DEBUG_LEVEL=5    ; Verbose
build_flags =
        -Wdouble-promotion             ; double to float warnings
        -DCORE_DEBUG_LEVEL=3
        -DCONFIG_ARDUHAL_LOG_COLORS=1
        -DBOARD_HAS_PSRAM
;       -mfix-esp32-psram-cache-issue
        -DARDUINO_RUNNING_CORE=1       ; Arduino Runs On Core (setup, loop)
        -DARDUINO_EVENT_RUNNING_CORE=1 ; Events Run On Core

build_unflags =
;        -DARDUINO_USB_CDC_ON_BOOT=0    ; traditional log
;        -DBOARD_HAS_PSRAM
````

cap60552, 2022-11-15:
This is my platformio.ini, yet no matter what I do I cannot seem to get it to compile / locate WiFiClientSecure.h
`
[env:wemos_d1_mini32]
platform =  https://github.com/platformio/platform-espressif32.git#v5.2.0
;platform = espressif32
board = wemos_d1_mini32
framework = arduino
monitor_speed = 115200
lib_deps = 
	marvinroger/AsyncMqttClient@^0.9.0
	ottowinter/ESPAsyncWebServer-esphome@^3.0.0
	https://github.com/pololu/maestro-arduino#@^1.0.0
	https://github.com/schreibfaul1/ESP32-audioI2S.git
	fastled/FastLED@^3.5.0
platform_packages = 
	framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.5
`

Here is the log output:
`
Compiling .pio\build\wemos_d1_mini32\FrameworkArduino\HWCDC.cpp.o
Compiling .pio\build\wemos_d1_mini32\FrameworkArduino\HardwareSerial.cpp.o
In file included from .pio/libdeps/wemos_d1_mini32/ESP32-audioI2S-master/src/Audio.cpp:11:
.pio/libdeps/wemos_d1_mini32/ESP32-audioI2S-master/src/Audio.h:21:10: fatal error: WiFiClientSecure.h: No such file or directory

**************************************************************************
* Looking for WiFiClientSecure.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiClientSecure.h"
* Web  > https://registry.platformio.org/search?q=header:WiFiClientSecure.h
*
**************************************************************************

 #include <WiFiClientSecure.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\wemos_d1_mini32\liba34\ESP32-audioI2S-master\Audio.cpp.o] Error 1
================================================================= [FAILED] Took 127.33 seconds =================================================================
`

Any idea as to what is going on?  I'm at a loss...  If I pull the "https://github.com/schreibfaul1/ESP32-audioI2S.git" lib_deps  everything else compiles just fine.

atlaste, 2022-12-08:
This is the list of dependencies I needed:

```
	FS
	WiFi
	WiFiClientSecure
	SD
	SD_MMC
	SPIFFS
	FFAT
```

evanvlane, 2023-03-19:
I also ran into this issue, and able to replicate it and resolve it.

I get this error when I include ESP32-audioI2S in my platform.ini, but have not yet added it as an include in a .cpp file that imports Arduino.h first. I don't think the compiler knows about Arduino.h when it's compiling Audio.h, so it misses all the WiFi dependencies, etc.

My platform.ini:
```
[env:esp-wrover-kit]
platform = espressif32
board = esp-wrover-kit
framework = arduino
monitor_speed = 115200
lib_deps = 
    esphome/ESP32-audioI2S@^2.0.6
```

With no `#include "Audio.h"` in my `main.cpp`, I get the following compile error:
```
 #include <WiFi.h>
          ^~~~~~~~
compilation terminated.
*** [.pio\build\esp-wrover-kit\libb15\ESP32-audioI2S\Audio.cpp.o] Error 1
```

When I add `#include "Audio.h"` to `main.cpp` it builds without error.

Hope this helps and resolves the issue!

closed 2024-03-16 by github-actions[bot]