schreibfaul1/ESP32-audioI2S #415: ESP32 FLAC radiostations

marcus81zhur, 2022-11-23:
Hello!
I am building kitchen radio project as a gift to my father. It almost done, but now when ESP32 module came to me and i have started to check the code, i found that i cannot hear half of good FLAC radiostations i have choosen. Audioinfo prints that OGG is not supported, or "audio/ogg" not supported, but i know if it's ogg found there and the station is flac that should be like ogg flac, that has to be supported.
I would really appreciate any suggestions on what to do as there are only a couple of weeks left to finish the project and finish the gift already. Thank you.
station examples:
http://stream.radioparadise.com/flac
http://89.223.45.5:8000/geny-flac
http://stream.sing-sing-bis.org:8000/singsingFlac

valera39, 2022-11-24:
I am added to the question, PCM5102, FLAC do not play

marcus81zhur, 2022-11-24:
Spent hours digging for a solution. No any library decodes FLAC from OGG container and that's sad. Fortunately, good radio stations have duplicate 320kb streams.

valera39, 2022-11-24:
on VS1053 Flack works

schreibfaul1, 2022-11-24:
I deleted the old ogg parser, which was not working properly. Since the last update, the FLAC decoder has taken over rudimentary ogg functions. This means that some ogg streams are now also working

closed 2023-05-15 by stale[bot]