schreibfaul1/ESP32-audioI2S #425: Another question - drops.

marcus81zhur, 2022-12-01:
Is there any way to improve the playback quality if the stream breaks? Debug writes - slow stream. But the same stream from a nearby computer plays fine. Would increasing any of the buffers help? What is the limits?

marcus81zhur, 2022-12-01:
Found may be not the best, but shortest way to solve the problem here:
https://github.com/espressif/arduino-esp32/issues/4529#issuecomment-869017503
Will try this evening.

e1z0, 2022-12-21:
> Found may be not the best, but shortest way to solve the problem here:
> [espressif/arduino-esp32#4529 (comment)](https://github.com/espressif/arduino-esp32/issues/4529#issuecomment-869017503)
> Will try this evening.

Did you solved the problem ?

Weltkugel, 2023-01-16:
I used the RC4-lib-files from [RafigRzayev](https://github.com/RafigRzayev) for my built. As ESP32-audioI2S needs RC5 (see  Audio.cpp line 170) I installed RC4 and changed the files from this commit https://github.com/espressif/arduino-esp32/commit/ef99cd7f for my installation.

To make it short: There are already drops, but I imagine it became better. Was it worth it? I don't know. Next time I will try the esp32-eth01 with Ethernet.