schreibfaul1/ESP32-audioI2S #429: Playing multiple audio files one after another in `void setup()`  only plays the last audio file called

timf34, 2022-12-08:
Hi,

I'm trying to play multiple audio files one after another within the `void setup()` - this is so that I can call different functions between the different audio files. 

However, when I try to run this, only the final audio file is played. I have tried using `delay()` function calls but they don't change anything.

Here is some sample code below:

```
void tutorial(){
  audio.connecttoFS(SPIFFS, mp3_files[0].c_str());
  audio.connecttoFS(SPIFFS, mp3_files[1].c_str());
  audio.connecttoFS(SPIFFS, mp3_files[2].c_str());
}

void setup(){
  tutorial()
}
```

This code will only contain the audio file within `mp3_files[2]` skipping over the first two.

schreibfaul1, 2022-12-09:
Does not work like that. Maybe this will help you:
````c++
void tutorial(){
  audio.connecttoFS(SPIFFS, mp3_files[0].c_str());
}
void setup(){
  tutorial()
}
void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);
    static int i = 1;
    if(i == 1) audio.connecttoFS(SPIFFS, mp3_files[1].c_str());
    if(i == 2) audio.connecttoFS(SPIFFS, mp3_files[2].c_str());
    ...
    i++;
}
````

timf34, 2022-12-09:
Hi, that's great, thank you for such a quick response! It solved my problem, consider this closed. Thanks!

closed 2022-12-09 by timf34

xfce, 2022-12-20:
> 

@schreibfaul1 
This worked, thanks.
If I want to play the SD files accord to three buttons, when BUTTON1 pressed, play audiofile1,BUTTON1 released audiofile1 stopped, BUTTON2 pressed, play audiofile2, BUTTON2 released also audiofile2 stop to play, how can I do this?

schreibfaul1, 2022-12-21:
Hi xfce,
mechanical keys tend to bounce. So it's better to add a short delay. 
You can add the example to your existing code
````c++
#define Button1 21
#define Button2 22
    ....
void setup() {
    ....
    pinMode(Button1, INPUT_PULLUP);
    pinMode(Button2, INPUT_PULLUP);
    ....
}

void loop(){
    ....
    static uint32_t btn1 = 0;
    static uint32_t btn2 = 0;
    if(digitalRead(Button1) == false) btn1++; else btn1 = 0; // button 1
    if(digitalRead(Button2) == false) btn2++; else btn2 = 0; // button 2
    if(btn1 == 50 && btn2 == 0) audio.connecttoFS(SD, "/file1.mp3");
    if(btn2 == 50 && btn1 == 0) audio.connecttoFS(SD, "/file2.mp3");
    if(btn1 == 0 && btn2 == 0) if(audio.isRunning()) audio.stopSong();
    ....
}
````

xfce, 2022-12-22:
> Hi xfce, mechanical keys tend to bounce. So it's better to add a short delay. You can add the example to your existing code
>
> ```c++
> #define Button1 21
> #define Button2 22
>     ....
> void setup() { [...]

Thanks, it works.