schreibfaul1/ESP32-audioI2S #432: Mux.com URLs not working

parvez01010, 2022-12-13:
Hello,
https://test-streams.mux.dev/x36xhzz/x36xhzz.m3u8
 I am trying to play this live streaming URL , but it is not working. Is there any issue with signed URLs?
AACDecoder is initialized but there is no audio.

Thanks.

parvez01010, 2022-12-13:
I am also attaching response of URL

info        Connect to new host: "https://stream.mux.com/JYrRSgMNik4Bscdnq02QImNXNpeZT5MWG.m3u8"
info        PSRAM not found, inputBufferSize: 6399 bytes
info        buffers freed, free Heap: 199292 bytes
info        1 default set
info        2 CONNECT TO "stream.mux.com" ON PORT 80, EXTENSION "/JYrRSgMNik4Bscdnq02QImNXNpeZT5MWG.m3u8"
info        SSL has been established in 8826 ms, free Heap: 158624 bytes
info        3 SELECTING EXTERNSION AND EXPECTED PLS FORMAT
info        4 SELECTED /JYrRSgMNik4Bscdnq02QImNXNpeZT5MWG.m3u8 EXTERNSION AND SELECTED PLS FORMAT 4
STARTING AUDIO PLAY
info        * httpResponseHeader: HTTP/1.1 200 OK
info        * HTTP response CODE
info        * httpResponseHeader: Date: Mon, 12 Dec 2022 20:40:16 GMT
info        * httpResponseHeader: Content-Type: application/x-mpegURL
info        * httpResponseHeader: Transfer-Encoding: chunked
info        chunked data transfer
info        * httpResponseHeader: Connection: keep-alive
info        * httpResponseHeader: CF-Ray: 7789462c9ab08af1-DEL
info        * httpResponseHeader: Access-Control-Allow-Origin: *
info        * httpResponseHeader: Cache-Control: no-cache, no-store, must-revalidate
info        * httpResponseHeader: CF-Cache-Status: DYNAMIC
info        * httpResponseHeader: Grpc-Metadata-Content-Type: application/grpc
info        * httpResponseHeader: X-CDN: cloudflare;Xf8LOcN16JiklTvHv6BcQIRe02qcqPQiSsk2BqSwwJ6psH9pwSaTUls7mySC2I5z9iUasezADs2U
info        * httpResponseHeader: X-Cdn-Version: 1669306814-88810-7cccefa104
info        * httpResponseHeader: X-CF-Ray: 7789462c9ab08af1
info        * httpResponseHeader: X-litix-View-Session-Id: Xf8LOcN16JiklTvHv6BcQIRe02qcqPQiSsk2BqSwwJ6psH9pwSaTUls7mySC2I5z9iUasezADs2U
info        * httpResponseHeader: X-Origin-Version: 1670386098-90168-56e2459406
info        * httpResponseHeader: Vary: Accept-Encoding
info        * httpResponseHeader: Server: cloudflare
info        GOING TO AUDIO PLAYLISTINIT
info        5 M3U8 DATASTREAM
info        * INSIDE READ PLAYLIST DATA
info        *PSRAM REQUIRED
info        * INSIDE WHILE LOOP
info        GOING TO AUDIO PLAYLIST DATA
info        new request: "https://manifest-gcp-us-east1-vop1.cfcdn.mux.com/LqIGFhzo2Fi02IIXqGlqolrIer70241wRIiieI02LF6rjVbLTJfe8gaOgHtw6fIqQ6zEkgBcrEmU1ZtFbmRhqhEyS019a6v4Og52p01SMYMiP01K00/rendition.m3u8?cdn=cloudflare&expires=1671483600&resolution=1280x720&skid=default&signature=NjNhMGQwZDBfZTMxNjBkMzlhZTlmNmM4ZDQ2MWMyYzk4MWE1OTU5NWZjMzZhMWY4NWM4NzEyM2EzYWRiNTUzZGQ0MjI5NjZjZA==&vsid=Xf8LOcN16JiklTvHv6BcQIRe02qcqPQiSsk2BqSwwJ6psH9pwSaTUls7mySC2I5z9iUasezADs2U"
info        12 FOUND HOST https://manifest-gcp-us-east1-vop1.cfcdn.mux.com/LqIGFhzo2Fi02IIXqGlqolrIer70241wRIiieI02LF6rjVbLTJfe8gaOgHtw6fIqQ6zEkgBcrEmU1ZtFbmRhqhEyS019a6v4Og52p01SMYMiP01K00/rendition.m3u8?cdn=cloudflare&expires=1671483600&resolution=1280x720&skid=default&signature=NjNhMGQwZDBfZTMxNjBkMzlhZTlmNmM4ZDQ2MWMyYzk4MWE1OTU5NWZjMzZhMWY4NWM4NzEyM2EzYWRiNTUzZGQ0MjI5NjZjZA==&vsid=Xf8LOcN16JiklTvHv6BcQIRe02qcqPQiSsk2BqSwwJ6psH9pwSaTUls7mySC2I5z9iUasezADs2U
info        * httpResponseHeader: HTTP/1.1 200 OK
info        * HTTP response CODE
info        * httpResponseHeader: Date: Mon, 12 Dec 2022 20:40:18 GMT
info        * httpResponseHeader: Content-Type: application/x-mpegURL
info        * httpResponseHeader: Content-Length: 2157
info        * httpResponseHeader: Connection: keep-alive
info        * httpResponseHeader: CF-Ray: 77894630be388af1-DEL
info        * httpResponseHeader: Accept-Ranges: bytes
info        * httpResponseHeader: Access-Control-Allow-Origin: *
info        * httpResponseHeader: Cache-Control: no-cache, no-store, must-revalidate
info        * httpResponseHeader: Last-Modified: Mon, 12 Dec 2022 20:40:17 GMT
info        * httpResponseHeader: Vary: Accept-Encoding
info        * httpResponseHeader: CF-Cache-Status: MISS
info        * httpResponseHeader: Grpc-Metadata-Content-Type: application/grpc
info        * httpResponseHeader: X-CDN: cloudflare;Xf8LOcN16JiklTvHv6BcQIRe02qcqPQiSsk2BqSwwJ6psH9pwSaTUls7mySC2I5z9iUasezADs2U
info        * httpResponseHeader: X-Cdn-Version: 1669306814-88810-7cccefa104
info        * httpResponseHeader: X-CF-Ray: 77894630be388af1
info        * httpResponseHeader: X-litix-View-Session-Id: Xf8LOcN16JiklTvHv6BcQIRe02qcqPQiSsk2BqSwwJ6psH9pwSaTUls7mySC2I5z9iUasezADs2U
info        * httpResponseHeader: X-Origin-Version: 1670386098-90168-56e2459406
info        * httpResponseHeader: Server: cloudflare
info        GOING TO AUDIO PLAYLISTINIT
info        * INSIDE READ PLAYLIST DATA
info        *PSRAM REQUIRED
info        * INSIDE WHILE LOOP
info        GOING TO AUDIO PLAYLIST DATA
info        new request: "https://chunk-gcp-us-east1-vop1.cfcdn.mux.com/v1/chunk/D021oYMWzdCxRRrlw18PMNVRl73y3iFnug7PmDA1zuJLzWyUWfuHqKGx02sKcOS5TcEWEdfRuowgEaZTdVrTJODCqWtZQ7HgQyHumheJa02KQ4/0.ts?resolution=1280x720&skid=default&signature=NjNhMGQwZDBfMzg4M2QzYTI0ZjhiOWEzYjI0ZmU0OGE3NzU4MDgxMjQzMWVkYjQzZmJlM2NiY2MwMTZlYTM3ODc0YzZjOGM3Zg==&vsid=Xf8LOcN16JiklTvHv6BcQIRe02qcqPQiSsk2BqSwwJ6psH9pwSaTUls7mySC2I5z9iUasezADs2U"
info        12 FOUND HOST https://chunk-gcp-us-east1-vop1.cfcdn.mux.com/v1/chunk/D021oYMWzdCxRRrlw18PMNVRl73y3iFnug7PmDA1zuJLzWyUWfuHqKGx02sKcOS5TcEWEdfRuowgEaZTdVrTJODCqWtZQ7HgQyHumheJa02KQ4/0.ts?resolution=1280x720&skid=default&signature=NjNhMGQwZDBfMzg4M2QzYTI0ZjhiOWEzYjI0ZmU0OGE3NzU4MDgxMjQzMWVkYjQzZmJlM2NiY2MwMTZlYTM3ODc0YzZjOGM3Zg==&vsid=Xf8LOcN16JiklTvHv6BcQIRe02qcqPQiSsk2BqSwwJ6psH9pwSaTUl [... cut ...]
info        * httpResponseHeader: Server: cloudflare
info        AACDecoder has been initialized, free Heap: 131068 bytes
info        13 INSIDE AUDIO DATA
info        * PROCESS WEBSTREAMTS
info        * getdatamode is not equal to audio_data
info        * AVAILABLE BYTES
info        14 FOUND NEW HOST GOING TO AUDIO PLAYLISTDATA
info        13 INSIDE AUDIO DATA
info        * PROCESS WEBSTREAMTS
info        * getdatamode is not equal to audio_data
info        * AVAILABLE BYTES
info        14 FOUND NEW HOST GOING TO AUDIO PLAYLISTDATA

parvez01010, 2022-12-13:
Thanks for replying.
This is a sample file generated through mux.com.
same response I am getting for audio live stream. 
if you require an audio stream I can share it with you.

schreibfaul1, 2022-12-13:
The problem is the ESP32 WiFiClient. The throughput is just enough for audio only, if audio and video data are mixed, the throughput is not sufficient.

parvez01010, 2022-12-13:
Thanks for replying.
Below is my live stream audio logs.

info        Connect to new host: "https://stream.mux.com/5nfpN00gMjdLyTFGscKTDoEtzDnEpxuUAIJJiMzo2uSo.m3u8"
info        PSRAM not found, inputBufferSize: 6399 bytes
info        buffers freed, free Heap: 199308 bytes
info        1 default set
info        2 CONNECT TO "stream.mux.com" ON PORT 80, EXTENSION "/5nfpN00gMjdLyTFGscKTDoEtzDnEpxuUAIJJiMzo2uSo.m3u8"
info        SSL has been established in 5776 ms, free Heap: 157700 bytes
info        3 SELECTING EXTERNSION AND EXPECTED PLS FORMAT
info        4 SELECTED /5nfpN00gMjdLyTFGscKTDoEtzDnEpxuUAIJJiMzo2uSo.m3u8 EXTERNSION AND SELECTED PLS FORMAT 4
STARTING AUDIO PLAY
info        * httpResponseHeader: HTTP/1.1 200 OK
info        * HTTP response CODE
info        * httpResponseHeader: Connection: keep-alive
info        * httpResponseHeader: Content-Length: 573
info        * httpResponseHeader: Access-Control-Allow-Origin: *
info        * httpResponseHeader: Content-Type: application/x-mpegURL
info        * httpResponseHeader: Grpc-Metadata-Content-Type: application/grpc
info        * httpResponseHeader: X-Origin-Version: 1670386098-90168-56e2459406
info        * httpResponseHeader: Accept-Ranges: bytes
info        * httpResponseHeader: Date: Tue, 13 Dec 2022 16:31:35 GMT
info        * httpResponseHeader: Via: 1.1 varnish
info        * httpResponseHeader: X-Served-By: cache-del21748-DEL
info        * httpResponseHeader: X-Cache: MISS
info        * httpResponseHeader: X-Cache-Hits: 0
info        * httpResponseHeader: X-Timer: S1670949095.419134,VS0,VE276
info        * httpResponseHeader: X-Cdn-Version: 1669306923-88811-4dad23c3c6
info        * httpResponseHeader: X-Request-ID: 274ef3ad-bf14-4ca6-bd2a-e4d5503ef34f
info        * httpResponseHeader: X-litix-View-Session-Id: Z2MboPlnoUxUr01hx9YDNjan2XdpLsSJCUp2Smu7Vi2BtC7EJf4tKdTTJPXK1gjZHAh6UjrFK34s
info        * httpResponseHeader: X-CDN: fastly;Z2MboPlnoUxUr01hx9YDNjan2XdpLsSJCUp2Smu7Vi2BtC7EJf4tKdTTJPXK1gjZHAh6UjrFK34s
info        * httpResponseHeader: Cache-Control: no-cache, no-store, must-revalidate
info        GOING TO AUDIO PLAYLISTINIT
info        5 M3U8 DATASTREAM
info        * INSIDE READ PLAYLIST DATA
info        *PSRAM REQUIRED
info        * INSIDE WHILE LOOP
info        GOING TO AUDIO PLAYLIST DATA
info        new request: "https://manifest-gcp-us-east4-vop1.fastly.mux.com/RLC00epMkbAHa7M5M2dtWGaTVlRrqiOWqGX521mgzQGHlbvLvPh16lTBetbtx8JNx00tlIO2Hrkmw/rendition.m3u8?cdn=fastly&exclude_pdt=false&expires=1671555600&live=1&resolution=0x0&skid=default&signature=NjNhMWVhMTBfMGNkNDc0NDczZDlkYzkzZTAzODRiNjNhYTRlMGVlZDI0YjJiZTc2YTMyNjEyZDMwMzcwODg3MzcyZjFlYzAyMg==&vsid=Z2MboPlnoUxUr01hx9YDNjan2XdpLsSJCUp2Smu7Vi2BtC7EJf4tKdTTJPXK1gjZHAh6UjrFK34s"
info        12 FOUND HOST https://manifest-gcp-us-east4-vop1.fastly.mux.com/RLC00epMkbAHa7M5M2dtWGaTVlRrqiOWqGX521mgzQGHlbvLvPh16lTBetbtx8JNx00tlIO2Hrkmw/rendition.m3u8?cdn=fastly&exclude_pdt=false&expires=1671555600&live=1&resolution=0x0&skid=default&signature=NjNhMWVhMTBfMGNkNDc0NDczZDlkYzkzZTAzODRiNjNhYTRlMGVlZDI0YjJiZTc2YTMyNjEyZDMwMzcwODg3MzcyZjFlYzAyMg==&vsid=Z2MboPlnoUxUr01hx9YDNjan2XdpLsSJCUp2Smu7Vi2BtC7EJf4tKdTTJPXK1gjZHAh6UjrFK34s
info        * httpResponseHeader: HTTP/1.1 200 OK
info        * HTTP response CODE
info        * httpResponseHeader: Connection: keep-alive
info        * httpResponseHeader: Content-Length: 1875
info        * httpResponseHeader: Access-Control-Allow-Origin: *
info        * httpResponseHeader: Content-Type: application/x-mpegURL
info        * httpResponseHeader: Grpc-Metadata-Content-Type: application/grpc
info        * httpResponseHeader: X-Origin-Version: 1670386098-90168-56e2459406
info        * httpResponseHeader: Via: 1.1 varnish, 1.1 varnish
info        * httpResponseHeader: Date: Tue, 13 Dec 2022 16:31:36 GMT
info        * httpResponseHeader: Age: 0
info        * httpResponseHeader: X-Served-By: cache-iad-kcgs7200079-IAD, cache-del21748-DEL
info        * httpResponseHeader: X-Cache: MISS, MISS
info        * httpResponseHeader: X-Cache-Hits: 0, 0
info        * httpResponseHeader: X-Timer: S1670949096.060815,VS0,VE354
info        * httpResponseHeader: Vary: Accept-Encoding
info        * httpResponseHeader: X-Cdn-Version: 1669306923-88811-4dad23c3c6
info        * httpResponseHeader: X-Request-ID: 09f1328d-6f3c-4c1a-9a80-2010001fd5bf
info        * httpResponseHeader: Accept-Ranges: bytes
info        * httpResponseHeader: X-litix-View-Session-Id: Z2MboPlnoUxUr01hx9YDNjan2XdpLsSJCUp2Smu7Vi2BtC7EJf4tKdTTJPXK1gjZHAh6UjrFK34s
info        * httpResponseHeader: X-CDN: fastly;Z2MboPlnoUxUr01hx9YDNjan2XdpLsSJCUp2Smu7Vi2BtC7EJf4tKdTTJPXK1gjZHAh6UjrFK34s
info        * httpResponseHeader: Cache-Control: no-cache, no-store, must-revalidate
info        GOING TO AUDIO PLAYLISTINIT
info        * INSIDE READ PLAYLIST DATA
info        *PSRAM REQUIRED
info        * INSIDE WHILE LOOP
info        GOING TO AUDIO PLAYLIST DATA
info        new request: "https://chunk-gcp-us-east4-vop1.fastly.mux.com/v1/chunk/g1MkPS5cJ5mClc01w6YVnY3MIeWMoJvmWzcC3Z00lgqeHOjLji54o7C8G02zGDeIhAKQND02skCn7LSl3unoRt02dGQ/113.ts?resolution=0x0&skid=default&signature=NjNhMWVhMTBfNzQxMWRlMjYxNmFiMDA2YjFkNWQ3NjA2Y2Y1MWU5MGUwOGRlNGRkZWFjNDFiYjJiYTU4MTM4YWJiZDk3YTYwMA=="
info        12 FOUND HOST https://chunk-gcp-us-east4-vop1.fastly.mux.com/v1/chunk/g1MkPS5cJ5mClc01w6YVnY3MIeWMoJvmWzcC3Z00lgqeHOjLji54o7C8G02zGDeIhAKQND02skCn7LSl3unoRt02dGQ/113.ts?resolution=0x0&skid=default&signature=NjNhMWVhMTBfNzQxMWRlMjYxNmFiMDA2Yj [... cut ...]
info        * RES is greater then 0
info        * ts pointer is less than packetsize
info        14 FOUND NEW HOST GOING TO AUDIO PLAYLISTDATA
info        13 INSIDE AUDIO DATA
info        * PROCESS WEBSTREAMTS
info        * RES is greater then 0
info        * ts pointer is less than packetsize
info        14 FOUND NEW HOST GOING TO AUDIO PLAYLISTDATA
info        13 INSIDE AUDIO DATA
info        * PROCESS WEBSTREAMTS
info        * RES is greater then 0
info        * ts pointer is less than packets

parvez01010, 2022-12-13:
Hello Sir,
I am able to play this video live stream.
info        Connect to new host: "https://cph-p2p-msl.akamaized.net/hls/live/2000341/test/master.m3u8"
info        PSRAM not found, inputBufferSize: 6399 bytes
info        buffers freed, free Heap: 199396 bytes
info        SSL has been established in 62790 ms, free Heap: 157648 bytes
STARTING AUDIO PLAY
info        new request: "https://cph-p2p-msl.akamaized.net/hls/live/2000341/test/level_0.m3u8"
info        new request: "https://cph-p2p-msl.akamaized.net/hls/live/2000341/test/segment_0_20221213_1670955156.ts"
info        AACDecoder has been initialized, free Heap: 130596 bytes
info        syncword found at pos 0
info        Channels: 2
info        SampleRate: 44100
info        BitsPerSample: 16
info        BitRate: 52024
info        AAC HeaderFormat: ADTS
info        AAC Codec: MPEG-4
info        AAC Profile: LowComplexity
info        VBR recognized, audioFileDuration is estimated
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        new request: "https://cph-p2p-msl.akamaized.net/hls/live/2000341/test/segment_0_20221213_1670955162.ts"
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        new request: "https://cph-p2p-msl.akamaized.net/hls/live/2000341/test/segment_0_20221213_1670955168.ts"
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        new request: "https://cph-p2p-msl.akamaized.net/hls/live/2000341/test/segment_0_20221213_1670955174.ts"
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        new request: "https://cph-p2p-msl.akamaized.net/hls/live/2000341/test/segment_0_20221213_1670955180.ts"
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        new request: "https://cph-p2p-msl.akamaized.net/hls/live/2000341/test/segment_0_20221213_1670955186.ts"
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        new request: "https://cph-p2p-msl.akamaized.net/hls/live/2000341/test/segment_0_20221213_1670955192.ts"
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        new request: "https://cph-p2p-msl.akamaized.net/hls/live/2000341/test/segment_0_20221213_1670955198.ts"
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        new request: "https://cph-p2p-msl.akamaized.net/hls/live/2000341/test/segment_0_20221213_1670955204.ts"
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        new request: "https://cph-p2p-msl.akamaized.net/hls/live/2000341/test/segment_0_20221213_1670955210.ts"
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        new request: "https://cph-p2p-msl.akamaized.net/hls/live/2000341/test/segment_0_20221213_1670955216.ts"
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        new request: "https://cph-p2p-msl.akamaized.net/hls/live/2000341/test/segment_0_20221213_1670955222.ts"
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        new request: "https://cph-p2p-msl.akamaized.net/hls/live/2000341/test/segment_0_20221213_1670955228.ts"
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        new request: "https://cph-p2p-msl.akamaized.net/hls/live/2000341/test/segment_0_20221213_1670955234.ts"
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        new request: "https://cph-p2p-msl.akamaized.net/hls/live/2000341/test/segment_0_20221213_1670955240.ts"
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible
info        slow stream, dropouts are possible

parvez01010, 2022-12-15:
KINDLY LOOK INTO ABOVE LOGS

schreibfaul1, 2022-12-15:
This is not a library for videos,
 only pure audio streams, the data rate is too high for interleaved audio + video streams
[image]