schreibfaul1/ESP32-audioI2S #434: Problems playing a .wav file over I2S

Protonerd, 2022-12-14:
When playing a .wav file from an SD card, only noise is coming over the I2S module .
My setup:
Adafruit Huzzah 32 (no PSRAM!)
MAX98357A
SD breakout

The communication with the SD works fine, the content is dumped and according to the serial monitor the .wav file is read in.
The I2S connection is also proven to be working, as using the internet radio I can receive a clear audio.
I tried different .wav file, now I'm using the one coming with the library. All have the same issue.

I'm using the latest update.

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
/System Volume Information
/test.wav
/320k_test.mp3
/myPlaylist.m3u
/Pink-Panther.wav
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
[I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: csaladWIFI 7490
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
[I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[I][WiFiMulti.cpp:119] run(): [WIFI] 4 networks found
[D][WiFiMulti.cpp:149] run():  --->   0: [11][7C:FF:4D:91:BC:F0] csaladWIFI 7490 (-63) *
[D][WiFiMulti.cpp:149] run():  --->   1: [11][DC:39:6F:A7:5D:9B] csaladWIFI 7490 (-74) *
[D][WiFiMulti.cpp:149] run():  --->   2: [11][DC:39:6F:A7:7E:E9] csaladWIFI 7490 (-75) *
[D][WiFiMulti.cpp:149] run():  --->   3: [11][2C:91:AB:16:8B:EE] csaladWIFI 7490 (-93) *
[I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 7C:FF:4D:91:BC:F0 SSID: csaladWIFI 7490 Channel: 11 (-63)
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.178.40, MASK: 255.255.255.0, GW: 192.168.178.1
[I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
[D][WiFiMulti.cpp:175] run(): [WIFI] SSID: csaladWIFI 7490
[D][WiFiMulti.cpp:176] run(): [WIFI] IP: 192.168.178.40
[D][WiFiMulti.cpp:177] run(): [WIFI] MAC: 7C:FF:4D:91:BC:F0
[D][WiFiMulti.cpp:178] run(): [WIFI] Channel: 11
[D][ESP32_Audio_I2S.ino:61] setup(): Total heap: 344116 
[D][ESP32_Audio_I2S.ino:62] setup(): Free heap: 203996
[D][ESP32_Audio_I2S.ino:63] setup(): Total PSRAM: 0
[D][ESP32_Audio_I2S.ino:64] setup(): Free PSRAM: 0
info        PSRAM not found, inputBufferSize: 6399 bytes
[V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.
info        buffers freed, free Heap: 195780 bytes
info        Reading file: "/test.wav"
info        file has no RIFF tag
info        stream ready
info        syncword found at pos 0
info        Channels: 2
info        SampleRate: 16000
info        BitsPerSample: 16
info        BitRate: N/A
info        Closing audio file
info        End of file "/test.wav"
eof_mp3     /test.wav

Protonerd, 2022-12-14:
I assumed that the wiring was right as using the WiFi to connect to a radio station works OK (I use your example where I can select between internet radio and .wav playback), with clear sound. Both the Huzzah32 and the MAX98357A have a soldered supply and GND connection because I know from experience that generating sounds with higher volume settings can consume power. For the same reason besides the USB connection (for powering and upload) I have a fully loaded Li-Ion battery connected.

Protonerd, 2023-01-02:
I downloaded the newest version assuming that some of the recent updates might have something to do with the problem I face (especially the ones related to no PSRAM). However I still do not get a playback.
I observed that the serial log is slightly different from the one you sent me a while ago.
Reference (yours):
`info        PSRAM not found, inputBufferSize: 6399 bytes
info        buffers freed, free Heap: 162432 bytes
info        Reading file: "/wave_test/Pink-Panther.wav"
info        FormatCode: 1
info        DataRate: 176400
info        DataBlockSize: 4
info        BitsPerSample: 16
info        Audio-Length: 1251992
info        stream ready
info        syncword found at pos 0
info        Channels: 2
info        SampleRate: 44100
info        BitsPerSample: 16
info        BitRate: 1411200
info        Closing audio file
info        End of file "Pink-Panther.wav"
eof_mp3     Pink-Panther.wav`

What I get currently: 
`info        buffers freed, free Heap: 195768 bytes
info        Reading file: "/Pink-Panther.wav"
info        file has no RIFF tag
info        stream ready
info        syncword found at pos 0
info        Channels: 2
info        SampleRate: 16000
info        BitsPerSample: 16
info        BitRate: N/A
info        Closing audio file
info        End of file "/Pink-Panther.wav"
eof_mp3     /Pink-Panther.wav`

I wonder for instance why the BitRate is set to N/A in my case.
Could this be the way how the SD card has been formatted? Are there special requirements there?
Internet streaming still works nicely.

schreibfaul1, 2023-01-02:
it works:
``audio.connecttohost("https://github.com/schreibfaul1/ESP32-audioI2S/raw/master/additional_info/Testfiles/Pink-Panther.wav"); ``
and when you download the file and save it to the sd card it doesn't work?
``audio.connecttoFS(SD, "/Pink-Panther.wav");``
It looks like your wav file is corrupt and the RIFF header is unreadable.

Protonerd, 2023-01-02:
That was a good idea, and playing the same file from the internet works OK.
So we have 2 options left:
- either the file is corrupted
- or there is a problem reading and playing the file from an SD-card

Protonerd, 2023-01-02:
I reformatted the SD card using Windows utility, File system FAT (default), allocation unit size default.
Then I stored the wav file directly from github to my SD-card.
Heureka, now it works. I still do not know if the reformatting did the work or the direct copy, I suspect rather the latter.
Thanks Latte for the help!

closed 2023-01-02 by Protonerd

markbiasotti, 2024-04-09:
@Protonerd   
Can you post your wiring between the Huzzah32, MAX98357A  and SD breakout? I'm using the same three and not getting any audio from my .wav file on the SD card. I believe I have formatted it but I could have my wiring between the three incorrect.  Here's my original post:  https://forums.adafruit.com/viewtopic.php?t=209829

thank you 

Mark

Protonerd, 2024-04-09:
I haven't played with it for some time, so I'm not even sure this was the working wiring, but the way it's wired up on my breadboard is as follows:
A1 -> DIN (MAX98357A)
27 -> BCLK (MAX98357A)
A0 -> LRC (MAX98357A)
MI -> DO (SD breakout)
MO -> CLK (SD breakout)
SDA -> CMD (SD breakout)
SCK -> D3 (SD breakout)

markbiasotti, 2024-04-09:
Thank you for your hookup. I will take a look at mine and confirm.

Mark