schreibfaul1/ESP32-audioI2S #435: Compile error when SDFATFS_USED is defined

Protonerd, 2022-12-14:
If I activate the SDFAT in the Audio.h :
#define SDFATFS_USED  // activate for SdFat
I get following compile errors:

I'm not sure in which case SDFAT is preferred over SD, but since I'm experiencing problems playing .wav file, I wanted to give it a try and by doing so got these errors:

C:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6\libraries\FS\src/FS.h:86:7: error: redefinition of 'class fs::FS'
 class FS
       ^
In file included from ..\sloeber.ino.cpp:10:0:
C:\Users\kunandra\Arduino\libraries_backup\ESP32-audioI2S-master\src/Audio.h:41:11: error: previous definition of 'class fs::FS'
     class FS : public SdFat {
           ^
C:\Users\kunandra\Arduino\libraries_backup\ESP32-audioI2S-master\src/Audio.h:65:12: error: conflicting declaration 'fs::SDFS SD_SDFAT'
 #define SD SD_SDFAT
            ^
C:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6\libraries\SD\src/SD.h:41:17: note: in expansion of macro 'SD'
 extern fs::SDFS SD;
                 ^
C:\Users\kunandra\Arduino\libraries_backup\ESP32-audioI2S-master\src/Audio.h:62:20: note: previous declaration as 'fs::SDFATFS SD_SDFAT'
 extern fs::SDFATFS SD_SDFAT;
                    ^
make[1]: *** [subdir.mk:20: sloeber.ino.cpp.o] Error 1
make: *** [makefile:109: all] Error 2

ggalisky, 2022-12-16:
I get the same errors. I fixed them by renaming the namespace and classes from 

fs to fs_audio

and FS to FS_audio

EXAMPLE:

namespace fs_audio {
    class FS_audio : public SdFat {
    public:
        bool begin(SdCsPin_t csPin = SS, uint32_t maxSck = SD_SCK_MHZ(25)) { return SdFat::begin(csPin, maxSck); }
    };

    class SDFATFS : public fs_audio::FS_audio {
    public:
        // sdcard_type_t cardType();
        uint64_t cardSize() {
            return totalBytes();
        }
        uint64_t usedBytes() {
            // set SdFatConfig MAINTAIN_FREE_CLUSTER_COUNT non-zero. Then only the first call will take time.
            return (uint64_t)(clusterCount() - freeClusterCount()) * (uint64_t)bytesPerCluster();
        }
        uint64_t totalBytes() {
            return (uint64_t)clusterCount() * (uint64_t)bytesPerCluster();
        }
    };
}

extern fs_audio::SDFATFS SD_SDFAT;

Protonerd, 2022-12-16:
Doing these steps I still had no luck.
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"C:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/config" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/app_trace" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/app_update" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/asio" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/bootloader_support" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/bt" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/coap" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/console" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/driver" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/efuse" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp-tls" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp32" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp_adc_cal" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp_event" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp_http_client" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp_http_server" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp_https_ota" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp_https_server" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp_ringbuf" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp_websocket_client" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/espcoredump" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/ethernet" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/expat" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/fatfs" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freemodbus" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/heap" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/idf_test" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/jsmn" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/json" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/libsodium" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/log" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/lwip" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/mbedtls" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/mdns" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/micro-ecc" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/mqtt" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/newlib" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/nghttp" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/nvs_flash" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/openssl" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/protobuf-c" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/protocomm" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/pthread" "-IC:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/ [... cut ...]
C:\Users\kunandra\Arduino\libraries_backup\ESP32-audioI2S-master\src/Audio.h:65:12: error: conflicting declaration 'fs::SDFS SD_SDFAT'
 #define SD SD_SDFAT
            ^
C:\Users\kunandra\Arduino\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6\libraries\SD\src/SD.h:41:17: note: in expansion of macro 'SD'
 extern fs::SDFS SD;
                 ^
C:\Users\kunandra\Arduino\libraries_backup\ESP32-audioI2S-master\src/Audio.h:61:26: note: previous declaration as 'fs_audio::SDFATFS SD_SDFAT'
 extern fs_audio::SDFATFS SD_SDFAT;