schreibfaul1/ESP32-audioI2S #440: Enable/Disable Audio object I2S resources?

beustens, 2022-12-17:
I'm trying to use your library alongside this one: https://github.com/pschatzmann/ESP32-A2DP. Basically, I want to switch between different audio sources (files on SD card, wifi web radio, bluetooth). Also, I already have the second I2S processor in use to record from a microphone.

How can I "disconnect" the Audio object from the I2S amp and later "reconnect" to it? On a long term improvement, it would be also nice to implement these bluetooth functionalities into your project as it fits your approach to have a multi-source audio object.

schreibfaul1, 2022-12-18:
There are two events that can be implemented in your project:
audio_process_extern(int16_t* buff, uint16_t len, bool *continueI2S); passes the output buffer and its length
audio_process_i2s(uint32_t* sample, bool *continueI2S); passes a single sample
With this you can save the samples on an SD card or send them via BT or...
The ESPuino project has such BT functionality and can serve as an example.
The normal ESP32 can support BT classic, newer generations like the S3 no longer support BT classic.

mircose, 2022-12-23:
Hi,
if I want to use audio lib, I configura I2S bus with:
audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT); 

if I want to switch to BT streamer, I configure I2S bus with:

#include "BluetoothA2DPSink.h"
...
          i2s_pin_config_t my_pin_config = {
            .bck_io_num = 27,
            .ws_io_num = 26,
            .data_out_num = 25,
            .data_in_num = -1
        };
        a2dp_sink.set_pin_config(my_pin_config);
        a2dp_sink.start("MyMusic");

but the I2S bus is used by the audio lib and BT don'y work.
There is a method to release I2S bus of audio lib?

beustens, 2022-12-23:
I haven't tried again, but there are 2 approaches:

**1**
In _Audio.h_, there is the Audio class, which has these methods
```
esp_err_t I2Sstart(uint8_t i2s_num);
esp_err_t I2Sstop(uint8_t i2s_num);
```
as private. You could put them under the public methods and try to call `audio.I2Sstop(0);` before starting the bluetooth sink.

**2**
The other approach would be to make use of the example here: https://github.com/pschatzmann/ESP32-A2DP#accessing-the-sink-data-stream-with-callbacks. Here, you start the bluetooth sink with i2s disabled and let Audio output the samples with its (also private) `bool playSample(int16_t sample[2]);` method. 

Unfortunately, I haven't had luck with both ideas yet, the ESP32 is always crashing randomly shortly whenever I try to start the bluetooth sink. Maybe I already have too much code and RAM is full or whatever.

schreibfaul1, 2022-12-23:
As long as the audio object exists, the I2S driver is installed. Make the object dynamic.
````c++
audio = new Audio(false, 3, 0);
audio->(I2S_BCLK, I2S_LRC, I2S_DOUT, -1, I2S_MCLK);
....
delete audio;
````
The destructor uninstalls the I2S driver

mircose, 2022-12-24:
thank you a lot for immediate response, I tried:

1) declare both object dynamic:
Audio* audio = new Audio(false, 3, 0);
BluetoothA2DPSink* a2dp_sink = new BluetoothA2DPSink();

2) setup audio lib:
        audio->setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
        audio->connecttohost(...);    
OK (play song correctly)

2) when I want to switch to BT:
      audio->stopSong();
      delete audio;
OK 

3) setup BT:
      a2dp_sink.set_pin_config(my_pin_config);
      a2dp_sink.start(...);
OK (BT play song from phone correctly)

4) when I want to switch from BT to audio lib:
      a2dp_sink->disconnect(); //Closes the connection, stops A2DP, and release memory
      delete a2dp_sink;  //delete object
debug message: E (15160) I2S: i2s_driver_uninstall(2047): I2S port 0 has not installed
seems I2S correctly released

5) setup audio lib:
      Audio* audio = new Audio(false, 3, 0);  
      audio->setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);  

no error message but when I play the song I get the message
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
and ESP reboot!

schreibfaul1, 2022-12-24:
This runs for me in an endless loop without reboot.
````c++
#include "Arduino.h"
#include "WiFiMulti.h"
#include "Audio.h"

Audio* audio;
WiFiMulti wifiMulti;
String ssid =     "xxxxx";
String password = "xxxxx";

    #define I2S_LRC     26
    #define I2S_DOUT    25
    #define I2S_BCLK    27
    #define I2S_MCLK     0

void makeAudio(){
    audio = new Audio;
    audio->setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, -1, I2S_MCLK);
    audio->setVolume(21);
}

void setup() {
    pinMode(2, INPUT_PULLUP);
    wifiMulti.addAP(ssid.c_str(), password.c_str());
    wifiMulti.run();
    if(WiFi.status() != WL_CONNECTED){
        WiFi.disconnect(true);
        wifiMulti.run();
    }
    makeAudio();
    audio->connecttohost("https://raw.githubusercontent.com/schreibfaul1/ESP32-audioI2S/master/additional_info/Testfiles/Olsen-Banden.mp3");
}

void loop(){
    if(audio) audio->loop();
}

void audio_info(const char * info){
    log_i("info %s", info);
}
void audio_eof_stream(const char *info){
    delete audio;
    makeAudio();
    audio->connecttohost("https://raw.githubusercontent.com/schreibfaul1/ESP32-audioI2S/master/additional_info/Testfiles/Olsen-Banden.mp3");
}
````

mircose, 2022-12-26:
Your sketch work fine, it's possible easly start and stop I2S Audio service.
The problem is BluetoothA2DPSink library.
When use a2dp_sink.end(true) for close BT service and release the memory, is't possible re-start the BT service (not supported after end(true) ).

Thenkyou a lot.

closed 2022-12-27 by beustens