schreibfaul1/ESP32-audioI2S #443: Compiling for ESPHome with ESP32-C3 board fails

krakowski, 2022-12-20:
Hi,

following the ESPHome [I2S Audio Media Player](https://esphome.io/components/media_player/i2s_audio.html) guide with a [Seeed Studio XIAO ESP32C3](https://wiki.seeedstudio.com/XIAO_ESP32C3_Getting_Started/) and the [`toolchain-riscv32-esp`](https://registry.platformio.org/tools/espressif/toolchain-riscv32-esp) Toolchain results in multiple errors during compilation.

## Related

  * #387 
  * #373 

## Compilation Errors

```
.piolibdeps/media-player/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp: In function 'void UnpackQuads(int, int, int*)':
.piolibdeps/media-player/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp:3289:106: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
         nCodeBits = DecodeHuffmanScalar(huffTabSpec, &huffTabSpecInfo[cb - HUFFTAB_SPEC_OFFSET], bitBuf, &val);
                                                                                                          ^~~~

.piolibdeps/media-player/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp: In function 'void UnpackPairsNoEsc(int, int, int*)':
.piolibdeps/media-player/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp:3336:104: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
         nCodeBits = DecodeHuffmanScalar(huffTabSpec, &huffTabSpecInfo[cb-HUFFTAB_SPEC_OFFSET], bitBuf, &val);
                                                                                                        ^~~~

.piolibdeps/media-player/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp: In function 'void UnpackPairsEsc(int, int, int*)':
.piolibdeps/media-player/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp:3378:104: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
         nCodeBits = DecodeHuffmanScalar(huffTabSpec, &huffTabSpecInfo[cb-HUFFTAB_SPEC_OFFSET], bitBuf, &val);
                                                                                                        ^~~~

.piolibdeps/media-player/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp: In function 'int DecodeOneScaleFactor()':
.piolibdeps/media-player/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp:4179:82: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
     nBits = DecodeHuffmanScalar(huffTabScaleFact, &huffTabScaleFactInfo, bitBuf, &val);
                                                                                  ^~~~

.piolibdeps/media-player/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp: In function 'void GenerateNoiseVector(int*, int*, int)':
.piolibdeps/media-player/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp:5313:41: error: invalid conversion from 'uint32_t*' {aka 'long unsigned int*'} to 'unsigned int*' [-fpermissive]
         coef[i] = ((int32_t)Get32BitVal((uint32_t *)last)) >> 16;
                                         ^~~~~~~~~~~~~~~~

.piolibdeps/media-player/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp: In function 'int DecodeOneSymbol(int)':
.piolibdeps/media-player/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp:8549:57: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
     nBits = DecodeHuffmanScalar(huffTabSBR, hi, bitBuf, &val);
                                                         ^~~~

.piolibdeps/media-player/ESP32-audioI2S/src/Audio.cpp: In member function 'void Audio::processWebStream()':
.piolibdeps/media-player/ESP32-audioI2S/src/Audio.cpp:3033:65: error: no matching function for call to 'min(uint32_t&, size_t)'
         availableBytes = min(availableBytes, InBuff.writeSpace());
                                                                 ^

.piolibdeps/media-player/ESP32-audioI2S/src/Audio.cpp: In member function 'void Audio::processWebFile()':
.piolibdeps/media-player/ESP32-audioI2S/src/Audio.cpp:3093:61: error: no matching function for call to 'min(size_t, uint32_t&)'
     availableBytes = min(InBuff.writeSpace(), availableBytes);
                                                             ^
```

## ESPHome Configuration

```yaml
esphome:
  name: media-player

esp32:
  board: seeed_xiao_esp32c3
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  password: "xxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Media-Player Fallback Hotspot"
    password: "xxxxxxxxxxxxxxxxxx"

media_player:
  - platform: i2s_audio
    name: ESPHome I2S Media Player
    dac_type: external
    i2s_lrclk_pin: GPIO21
    i2s_bclk_pin: GPIO7
    i2s_dout_pin: GPIO6
    mode: mono

captive_portal:
```

schreibfaul1, 2022-12-20:
[image]

krakowski, 2022-12-20:
Hi @schreibfaul1,

I'm aware that the ESP32-C3 is a single-core chip. However, this does not explain why the code does not compile using this board's toolchain.

The ESP32-C3 explicitly supports I2S, so I would like to give it a try and see how far I can push it. Looking through the code, I couldn't find any reference to `xTaskCreatePinnedToCore`. Does this project really make use of both cores?

schreibfaul1, 2022-12-20:
Hello,
I don't own an ESP32-C3. But an S2, even with only one core, runs very poorly. Please use an ESP32 or an ESP32-S3.
Unfortunately, I have no experience with ESPHome, I've never used it.
You can take a simple example, without additional SW. This will compile for sure, but the sound output will stutter.

krakowski, 2022-12-22:
Hi @schreibfaul1,

I set up Arduino IDE using [Seeed Studio's Guide](https://wiki.seeedstudio.com/XIAO_ESP32C3_Getting_Started/#software-setup), installed this project as as library using the [instructions in the wiki](https://github.com/schreibfaul1/ESP32-audioI2S/wiki#can-the-arduino-ide-be-used) and tried compiling the [webstream example from the wiki](https://github.com/schreibfaul1/ESP32-audioI2S/wiki#a-simple-project-to-receive-a-webstream) which unfortunately also fails with the same error messages.

[image]

schreibfaul1, 2022-12-22:
This is peculiar.
int and int32_t are identical in size and value range in ESP32. The compiler has nothing to complain about.
Is that different with the C3, is int 16 bit or 64 bit???
Then please force a type conversion either (int32_t)var_int  or  (int)var_int32_t

krakowski, 2022-12-22:
I found out that `uint32_t` and `size_t` are not identical on RISC-V, which the ESP32-C3 is based on (see https://github.com/earlephilhower/ESP8266Audio/pull/454) . Most likely this is the issue I am having compiling the code. I will try to force the type conversion at the specific locations and see if this fixes the problem.

schreibfaul1, 2022-12-22:
I selected the C3 board and compiled with it. Not just two errors came, but a whole list. I adjusted the data types, now compiling should be possible. Please report back if there are still compiler errors.

schreibfaul1, 2022-12-22:
Thanks for the link. Some things are clear to me now.
[image]

krakowski, 2022-12-22:
Thanks for taking the time to fix this! :slightly_smiling_face:

I can confirm that the code now compiles for RISC-V and runs on a ESP32-C3

[image]

I think you can close #387 and #373 as well, since they describe the same problem. Thanks again!

closed 2022-12-22 by krakowski