schreibfaul1/ESP32-audioI2S #448: Error compiling on ESP-C3-13

DavidWicks, 2022-12-27:
```.piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp: In function 'void UnpackQuads(int, int, int*)':
.piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp:3289:106: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
         nCodeBits = DecodeHuffmanScalar(huffTabSpec, &huffTabSpecInfo[cb - HUFFTAB_SPEC_OFFSET], bitBuf, &val);
                                                                                                          ^~~~
In file included from .piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp:9:
.piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.h:490:115: note:   initializing argument 4 of 'int DecodeHuffmanScalar(const short int*, const HuffInfo_t*, unsigned int, int32_t*)'
 int DecodeHuffmanScalar(const signed short *huffTab, const HuffInfo_t *huffTabInfo, unsigned int bitBuf, int32_t *val);
                                                                                                          ~~~~~~~~~^~~
.piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp: In function 'void UnpackPairsNoEsc(int, int, int*)':
.piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp:3336:104: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
         nCodeBits = DecodeHuffmanScalar(huffTabSpec, &huffTabSpecInfo[cb-HUFFTAB_SPEC_OFFSET], bitBuf, &val);
                                                                                                        ^~~~
In file included from .piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp:9:
.piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.h:490:115: note:   initializing argument 4 of 'int DecodeHuffmanScalar(const short int*, const HuffInfo_t*, unsigned int, int32_t*)'
 int DecodeHuffmanScalar(const signed short *huffTab, const HuffInfo_t *huffTabInfo, unsigned int bitBuf, int32_t *val);
                                                                                                          ~~~~~~~~~^~~
.piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp: In function 'void UnpackPairsEsc(int, int, int*)':
.piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp:3378:104: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
         nCodeBits = DecodeHuffmanScalar(huffTabSpec, &huffTabSpecInfo[cb-HUFFTAB_SPEC_OFFSET], bitBuf, &val);
                                                                                                        ^~~~
In file included from .piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp:9:
.piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.h:490:115: note:   initializing argument 4 of 'int DecodeHuffmanScalar(const short int*, const HuffInfo_t*, unsigned int, int32_t*)'
 int DecodeHuffmanScalar(const signed short *huffTab, const HuffInfo_t *huffTabInfo, unsigned int bitBuf, int32_t *val);
                                                                                                          ~~~~~~~~~^~~
.piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp: In function 'int DecodeOneScaleFactor()':
.piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp:4179:82: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
     nBits = DecodeHuffmanScalar(huffTabScaleFact, &huffTabScaleFactInfo, bitBuf, &val);
                                                                                  ^~~~
In file included from .piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp:9:
.piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.h:490:115: note:   initializing argument 4 of 'int DecodeHuffmanScalar(const short int*, const HuffInfo_t*, unsigned int, int32_t*)'
 int DecodeHuffmanScalar(const signed short *huffTab, const HuffInfo_t *huffTabInfo, unsigned int bitBuf, int32_t *val);
                                                                                                          ~~~~~~~~~^~~
.piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp: In function 'void GenerateNoiseVector(int*, int*, int)':
.piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp:5313:41: error: invalid conversion from 'uint32_t*' {aka 'long unsigned int*'} to 'unsigned int*' [-fpermissive]
         coef[i] = ((int32_t)Get32BitVal((uint32_t *)last)) >> 16;
                                         ^~~~~~~~~~~~~~~~
.piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp:5146:40: note:   initializing argument 1 of 'unsigned int Get32BitVal(unsigned int*)'
 unsigned int Get32BitVal(unsigned int *last)
                          ~~~~~~~~~~~~~~^~~~
.piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp: In function 'int DecodeOneSymbol(int)':
.piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp:8549:57: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
     nBits = DecodeHuffmanScalar(huffTabSBR, hi, bitBuf, &val);
                                                         ^~~~
In file included from .piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.cpp:9:
.piolibdeps/test2/ESP32-audioI2S/src/aac_decoder/aac_decoder.h:490:115: note:   initializing argument 4 of 'int DecodeHuffmanScalar(const short int*, const HuffInfo_t*, unsigned int, int32_t*)'
 int DecodeHuffmanScalar(const signed short *huffTab, const HuffInfo_t *huffTabInfo, unsigned int bitBuf, int32_t *val);
                                                                                                          ~~~~~~~~~^~~
*** [.pioenvs\test2\libdae\ESP32-audioI2S\aac_decoder\aac_decoder.cpp.o] Error 1
.piolibdeps/test2/ESP32-audioI2S/src/Audio.cpp: In member function 'void Audio::processWebStream()':
.piolibdeps/test2/ESP32-audioI2S/src/Audio.cpp:3033:65: error: no matching function for call to 'min(uint32_t&, size_t)'
         availableBytes = min(availableBytes, InBuff.writeSpace());
                                                                 ^
In file included from c:\users\david.wicks\.platformio\packages\toolchain-riscv32-esp\riscv32-esp-elf\include\c++\8.4.0\algorithm:62,
                 from C:/Users/David.Wicks/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:162,
                 from .piolibdeps/test2/ESP32-audioI2S/src/Audio.h:15,
                 from .piolibdeps/test2/ESP32-audioI2S/src/Audio.cpp:11:
c:\users\david.wicks\.platformio\packages\toolchain-riscv32-esp\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algo.h:3456:5: note: candidate: 'template<class _Tp, class _Compare> _Tp std::min(std::initializer_list<_Tp>, _Compare)'
     min(initializer_list<_Tp> __l, _Compare __comp)
     ^~~
c:\users\david.wicks\.platformio\packages\toolchain-riscv32-esp\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algo.h:3456:5: note:   template argument deduction/substitution failed:
.piolibdeps/test2/ESP32-audioI2S/src/Audio.cpp:3033:65: note:   mismatched types 'std::initializer_list<_Tp>' and 'long unsigned int'
         availableBytes = min(availableBytes, InBuff.writeSpace());
                                                                 ^
In file included from c:\users\david.wicks\.platformio\packages\toolchain-riscv32-esp\riscv32-esp-elf\include\c++\8.4.0\algorithm:62,
                 from C:/Users/David.Wicks/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:162,
                 from .piolibdeps/test2/ESP32-audioI2S/src/Audio.h:15,
                 from .piolibdeps/test2/ESP32-audioI2S/src/Audio.cpp:11:
c:\users\david.wicks\.platformio\packages\toolchain-riscv32-esp\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algo.h:3450:5: note: candidate: 'template<class _Tp> _Tp std::min(std::initializer_list<_Tp>)'
     min(initializer_list<_Tp> __l)
     ^~~
[... 2 lines cut ...]
         availableBytes = min(availableBytes, InBuff.writeSpace());
                                                                 ^
In file included from c:\users\david.wicks\.platformio\packages\toolchain-riscv32-esp\riscv32-esp-elf\include\c++\8.4.0\vector:60,
                 from .piolibdeps/test2/ESP32-audioI2S/src/Audio.h:14,
                 from .piolibdeps/test2/ESP32-audioI2S/src/Audio.cpp:11:
c:\users\david.wicks\.platformio\packages\toolchain-riscv32-esp\riscv32-esp-elf\include\c++\8.4.0\bits\stl_algobase.h:243:5: note: candidate: 'template<class _Tp, class _Compare> const _Tp& std::min(const _Tp&, const _Tp&, _Compar [... cut ...]
.piolibdeps/test2/ESP32-audioI2S/src/Audio.cpp:3033:65: note:   deduced conflicting types for parameter 'const _Tp' ('long unsigned int' and 'size_t' {aka 'unsigned int'})
         availableBytes = min(availableBytes, InBuff.writeSpace());
                                                                 ^
.piolibdeps/test2/ESP32-audioI2S/src/Audio.cpp: In member function 'void Audio::processWebFile()':
.piolibdeps/test2/ESP32-audioI2S/src/Audio.cpp:3093:61: error: no matching function for call to 'min(size_t, uint32_t&)'
     availableBytes = min(InBuff.writeSpace(), availableBytes);```

schreibfaul1, 2022-12-27:
We recently had the topic #443, the new version eliminates the compiler errors, but the C3 will still not work properly.
Reason: C3 is an 32-bit RISC-V single-core processor. Only the ESP32 and the ESP32-S3 work.

DavidWicks, 2022-12-27:
ok, how about ESP32-S ?

probonopd, 2023-02-18:
> Reason: C3 is an 32-bit RISC-V single-core processor. Only the ESP32 and the ESP32-S3 work.

While I don't have a ESP32-C3 board to test with yet, https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-player/player-url-i2s is said to work well on the ESP32-C3 according to the author, so the question is whether it can also be used with the ESP32-audioI2S library.

probonopd, 2023-05-15:
.

probonopd, 2023-05-29:
> This issue has been automatically marked as stale because it has not had recent activity. It will be closed if no further activity occurs. Thank you for your contributions.

Les-A, 2023-07-07:
> While I don't have a ESP32-C3 board to test with yet, https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-player/player-url-i2s is said to work well on the ESP32-C3 [...]

@probonopd I've tried this with the Seeed XIAO esp32-c3 board with HTTP input (internet streaming radio at 22khz sample rate) and though it works it is very choppy. I've seen various comments about the TCP stream not being processed fast enough. I haven't yet tried it with the SD Card input but it might work with that.

probonopd, 2024-03-15:
Please do not close valid tickets just because there has not been recent activity on them. Thanks!

schreibfaul1, 2024-04-27:
But the readme states at the top that single core processors do not work.
[image]

closed 2024-04-27 by schreibfaul1

probonopd, 2024-04-27:
Ah, ok, that's a valid reason of course.
(That bot is still not doing any good imho, though.)