schreibfaul1/ESP32-audioI2S #449: Pause doesn't work for ST_WEBFILE sources

darkskiez, 2022-12-27:
I added it to this list and it seems to work for me, i'm not sure why this type is excluded:

https://github.com/schreibfaul1/ESP32-audioI2S/blob/bec84fe845643a98ab12cff3844d82d69dfca486/src/Audio.cpp#L2075

schreibfaul1, 2022-12-27:
Hello @darkskiez,
that could be doable. So far no one has asked about it. I don't know how the servers rate a pause, maybe it times out or the connection stays fine. I will test this and if the test is successful I will add the feature.

darkskiez, 2022-12-27:
Ah, I had some assumptions that the access was chunked anyway and the connection would just use a range request to reconnect/resume the stream at the right place. I guess thats another feature request, thanks for the library and consideration!

closed 2023-05-15 by stale[bot]