schreibfaul1/ESP32-audioI2S #451: if "no psram" is requested in setBufSize(), it may still get used

seife, 2022-12-27:
The `AudioBuffer::setBufsize(int ram, int psram)` allows to specifically request that buffers are *not* in psram, e.g. by doing it like this:

```
   audio.setBufsize(buf_sz + 1600, 0);
```

But if the `buf_sz` in the above example is big enough, it will be allocated in PSRAM (if available) anyway. I could not find the exact size again, I seem to remember that it was "above 4k, PSRAM is used if available for plain malloc()".

In order to avoid that (I have a board that gets really flaky WiFi performance if PSRAM is enabled in ESP32-audioI2S), we should explicitly request to allocate from internal RAM if psram has been disabled by setting the psram parameter of setBufSize() to zero.

closed 2022-12-27 by schreibfaul1

codegrue, 2024-02-13:
How do you call setBufsize before instantiating the Audio object which sets up the buffers on init()?

seife, 2024-02-13:
you don't. This is actually AudioBuffer::init() which is not called on Audio object instantiation, but from Audio::SetDefaults() which in turn is called in various places, usually when starting to play some URL / file /...

codegrue, 2024-02-13:
Hmm. When I create the Audio object (and don't make any calls to it), my available heap drops from 84064 to 64668. This prevents SSL network calls from working, unfortunately. Is there any way to reduce that footprint?

PR #452 (2022-12-27): AudioBuffer: if "no PSRAM" is requested, really avoid it
This fixes #451