schreibfaul1/ESP32-audioI2S #460: ESP crashes during playing local MP3 file with ID3 Tag v2.2

anp59, 2023-01-10:
Hi, 
during a test I noticed that local MP3 files with ID3 tags v2.2 cause the programme to crash. With common Windows players these files are played correctly.
This error was tested with the standard example, both with and without SdFat support. The test file is attached.
[test1.zip](https://github.com/schreibfaul1/ESP32-audioI2S/files/10387550/test1.zip)

Perhaps you can investigate this error in more detail when you get the chance. 
Many thanks in advance.

**LOG**
_info        PSRAM not found, inputBufferSize: 6399 bytes
info        buffers freed, free Heap: 238520 bytes
info        Reading file: "/test1.mp3"
info        MP3Decoder has been initialized, free Heap: 210704 bytes
info        Content-Length: 4209541
info        ID3 framesSize: 112444
info        ID3 version: 2.2
id3data     Title/Songname/Content description: La mamma morta
id3data     Lead artist(s)/Lead performer(s)/Soloist(s)/Performing group: Callas
id3data     Track number/Position in set: 1
id3data     Content type: (103)P
id3data     Attached picture: PNG
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled._

schreibfaul1, 2023-01-11:
Hello @anp59,
strange, no one noticed the error before. MP3 files with ID3V2.2 and embedded image are rare. I hadn't considered the APIC and so the MP3 decoder was served the image data. After the image data, there is more metadata that is now read out correctly. Only then does the MP3 decoder come into play.
Many greetings
Wolle

schreibfaul1, 2023-01-11:
The error message comes at the end of the file. The file has a constant bit rate throughout. This means that all mp3 frames must have the same length (can vary by 1, depending on the padding bit). Everything is fine in test1.mp3. The last mp3 frame of the test2.mp3 file is too short. You can calculate how long the frame has to be: 144*bitrate/samplerate. 144*112000/44100 ~ 365 Bytes. Of these, 23 are in the frame header, leaving 342 for the audio data. I watched that in a hex exitor.
I marked the frame header. The red is the syncword and the yellow is the heder information. The last frame should also be 342 bytes long, but it isn't. I think the easiest will be to edit the file with an audio tool.
[image]

anp59, 2023-01-12:
Ok. Now I know the cause and can edit the file if necessary. Without changes, it also works, you don't hear the error message :-). 
The original problem is closed with this and can be closed.  
Thank you very much and best regards!

schreibfaul1, 2023-01-12:
Most commercial audio players like VLC or Celluloid are fault tolerant. You can use a hex editor to erase the last incomplete frame with no problem and everything works fine.
If you discover further problems, you are welcome to let us know, there are certainly still some errors or deficiencies undiscovered.

khvolk, 2023-02-28:
I have a slightly different problem: I have two local mp3 files which do not play at all. When the playback starts you see only the meta-data in monitor. No sound. These files have the CD-cover as local stream embedded. 

ffmpeg: 
...
  Duration: 00:06:14.39, start: 0.000000, bitrate: 128 kb/s
  Stream #0:0: Audio: mp3, 44100 Hz, stereo, fltp, 128 kb/s
  Stream #0:1: Video: mjpeg (Baseline), yuvj420p(pc, bt470bg/unknown/unknown), 300x298 [SAR 1:1 DAR 150:149], 90k tbr, 90k tbn (attached pic)

closed 2023-06-18 by stale[bot]