schreibfaul1/ESP32-audioI2S #463: Problem with BBC .m3u8 stream

OGB1952, 2023-01-12:
Firstly, many thanks for a great bit of code, been using it with a very simple streaming radio built in to our kitchen for a year or so now.  The BBC here in the UK is changing its streams to HLS from mid-2023 and is broadcasting a message on all older streams that they will end soon.  The old BBC Radio 4 stream at http://stream.live.vc.bbcmedia.co.uk/bbc_radio_fourfm is working fine for now but is going to stop working mid-2023.  The new stream URL is http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_med/ak/bbc_radio_fourfm.m3u8

I edited the code to change to this new stream URL, using the latest version of this library to get .m3u8 support, and am getting a constant 503 error in the log files, after a re-direct from the BBC.  This is the latest log file:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4
info        Connect to new host: "http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_med/ak/bbc_radio_fourfm.m3u8"
info        PSRAM not found, inputBufferSize: 6399 bytes
info        buffers freed, free Heap: 199284 bytes
info        Connection has been established in 34 ms, free Heap: 198296 bytes
info        new request: "http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d128000.norewind.m3u8"
streamtitle HTTP/1.1 503 Service Unavailable
station     
icyurl  

Any clues as to what the problem may be with this new BBC stream would be most welcome.  The BBC .m3u8 stream plays OK in VLC Player on a PC.  From VLC Player the stream is using the ADTS Codec, 48k samples/sec, 32 bits per sample.

The problem seems to be with whatever is causing the 503, and I have a feeling this may not be related to the audio library.  I'm not sufficiently knowledgeable to be sure if this is the case or not, but I wonder if this may be related to an HTTP 1.1 related error within the wifi.h library?

The sketch for my really simple cut-down version based on your web radio example is below (minus WiFi credentials). :

#include "Arduino.h"
#include "WiFi.h"
#include "Audio.h"

// Digital I/O used
#define I2S_DOUT      25
#define I2S_BCLK      27
#define I2S_LRC       26

Audio audio;

String ssid =     "*******";
String password = "*******";

void setup() {
    Serial.begin(115200);
    WiFi.disconnect();
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid.c_str(), password.c_str());
    while (WiFi.status() != WL_CONNECTED) delay(1500);
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(6); // 0...21

//    audio.connecttohost("http://stream.live.vc.bbcmedia.co.uk/bbc_radio_fourfm");  //old BBC stream due to end mid-2023
    audio.connecttohost("http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_med/ak/bbc_radio_fourfm.m3u8");
}

void loop()
{
    audio.loop();
}

// optional
void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
void audio_id3data(const char *info){  //id3 metadata
    Serial.print("id3data     ");Serial.println(info);
}
void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);
}
void audio_showstation(const char *info){
    Serial.print("station     ");Serial.println(info);
}
void audio_showstreamtitle(const char *info){
    Serial.print("streamtitle ");Serial.println(info);
}
void audio_bitrate(const char *info){
    Serial.print("bitrate     ");Serial.println(info);
}
void audio_commercial(const char *info){  //duration in sec
    Serial.print("commercial  ");Serial.println(info);
}
void audio_icyurl(const char *info){  //homepage
    Serial.print("icyurl      ");Serial.println(info);
}
void audio_lasthost(const char *info){  //stream URL played
    Serial.print("lasthost    ");Serial.println(info);
}
void audio_eof_speech(const char *info){
    Serial.print("eof_speech  ");Serial.println(info);
}

OGB1952, 2023-01-12:
Thanks for the reply, but I don't think this is the problem.  I'm in the UK and my IP address does geolocate correctly to close to where we live.  I have tried both the UK-only URL (this one, http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_med/ak/bbc_radio_fourfm.m3u8 ) and the non-UK URL (this one: http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/nonuk/sbr_low/ak/bbc_radio_fourfm.m3u8 ) and both return the 503 error message.

I've spent the past hour or two trying to see if this could be an issue with the way that the ESP32 handles HTTP, with no luck so far.  I have a feeling that it must be, perhaps related to HTTP 1.0 versus HTTP 1.1.  Infuriatingly I have a vague memory of reading something about HTTP 1.0 versus HTTP 1.1 and the ESP32 some time ago, and a need to be explicit with GET calls, but try as I might I can't find it.

It must be specific to the ESP32, as my Windows PC will play either of the above streams in VLC Player with no problem.

I'll keep digging and see if I can find a reason for the problem and report back here.

schreibfaul1, 2023-01-12:
I can reach the URL http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_med/ak/bbc_radio_fourfm.m3u8 from Germany
Two things are unusual:
1) HTTP err 503 for http://as-hls-uk-live.akamaized.net/pool_904/live/uk/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d128000.norewind.m3u8
2) the TS packets are not interpreted correctly
I'll look at that in the next few days

OGB1952, 2023-01-13:
Many thanks. 

I've had a look at whether the 503 might be related to an HTTP/1.0 versus HTTP/1.1 issue but I don't think that's it.  My thoughts were that HTTP/1.1 assumes a persistent connection, HTTP/1.0 doesn't.  There were some issues with this with the ESP32 (specifically the HTTPClient code) some time ago, but they've been resolved.

schreibfaul1, 2023-01-13:
I changed two things
1) in the ts package: According to the MPEG2 specification, an 'adaptation field' is permitted in every package. I hadn't previously provided this for the PAT and PMT tables, this has now been corrected.
2) the BBC server is smart, HTTP 503 occurs when there is more than one connection from client. In the case of a redirection, there were two, so I have to cut the connection and then rebuild it

I updated the master, BBC4 will work now.

OGB1952, 2023-01-13:
Thank you very much indeed!  

Amazing bit of detective work, I had been scratching my head about this for hours.

closed 2023-01-13 by OGB1952