schreibfaul1/ESP32-audioI2S #468: Support for OPUS codec

aidam38, 2023-01-23:
Hello, are there any plans to support the OPUS codec (would allow playing audio from YouTube).

schreibfaul1, 2023-01-23:
I adapted the decoder of the Xiph.Org Foundation to the ESP32 and removed all ballast. This works quite well in a standalone project (see https://github.com/schreibfaul1/ESP32-Opus-Player.git). However, this consumes so many resources that it cannot be used in the audioI2S library.

aidam38, 2023-01-23:
Hmm, how hard would it be to adapt the networking code from ESP32-audioI2S to work with ESP32-Opus-Player, so that opus audio can be played from a URL?

aidam38, 2023-01-24:
^made this into an issue here https://github.com/schreibfaul1/ESP32-Opus-Player/issues/1

closed 2023-05-15 by stale[bot]