schreibfaul1/ESP32-audioI2S #469: audiofile.size() is not an uint32_t

podaen, 2023-01-28:
https://github.com/schreibfaul1/ESP32-audioI2S/blob/0ef0ca3542329722606442e57b864b8b342b11a5/src/Audio.cpp#L2718

I have no matching function for call to 'min(uint32_t&, long long unsigned int)

podaen, 2023-01-28:
Pull has made #470

podaen, 2023-02-04:
My pull didn't go through because I changed below back to his original state before making the push, because I don't need the core lib SD.

https://github.com/schreibfaul1/ESP32-audioI2S/blob/ea287871228a6773390aa2fe6567ca3d79de793e/src/Audio.h#L11

Now If I look closer to attention to this

https://github.com/schreibfaul1/ESP32-audioI2S/blob/ea287871228a6773390aa2fe6567ca3d79de793e/src/Audio.cpp#L2717-L2718

The first is for webstream, the second for an SD card if I not mistaken. I seems strange to set availableBytes two time after each other... To compile I need to cover on the second line  `audiofile.size()` to an `uint32_t`. I have two possibilities that file could be...

- The audiofile in an FS system for SdFat is FsBaseFile, uint64_t. Is the output uint32_t or  uint64_t? 
- The audiofile in an File32 is always uint32_t.
- For the core FS lib as I sad before I don't know.

schreibfaul1, 2023-02-04:
Previously, the file system could only process ASCII characters. That was the reason for SdFat. In the meantime you can use V2.0.6 e.g. use Chinese or Cyrillic letters in the file name.
This means that there is no longer any advantage in using SdFat, especially since the FS is incompatible with Espressif.

podaen, 2023-02-04:
I did the test and don't have building errors or warnings anymore using the core SD. So that solved it for me.

closed 2023-02-04 by podaen